raspberrypi/linux #1496: Logitech RumblePad 2 USB stopped working in kernel 4.4

diogomsantos, 2016-05-27:
After the latest upgrade on Raspbian, to kernel 4.4 with `apt-get dist-upgrade` my Logitech RumblePad 2 USB stopped working.

`[  830.395979] input: Logitech Logitech RumblePad 2 USB as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5:1.0/0003:046D:C218.0008/input/input8
[  830.421915] logitech 0003:046D:C218.0008: input,hidraw3: USB HID v1.10 Joystick [Logitech Logitech RumblePad 2 USB] on usb-3f980000.usb-1.5/input0
[  830.446984] logitech: probe of 0003:046D:C218.0008 failed with error -1`

There are other gamepads from the same brand with the same problem. 
This was already reported and fixed in OSMC.
Report:
https://discourse.osmc.tv/t/rumblepad-support/11404/1

Fix:
https://github.com/osmc/osmc/commit/cdbd1354d18c2d5a5719606e411df276aa9f51f0

Please include/fix this.
Thanks

popcornmix, 2016-05-31:
The options should be added. Please run rpi-update and test.

diogomsantos, 2016-05-31:
It works again. Thank you. Closed.

closed 2016-05-31 by diogomsantos