raspberrypi/linux #1497: new stuttering on IQAudio Dac

rbnpi, 2016-05-27:
Since upgrading to the latest Raspbian 2016-05-10 I have noticed a stuttering on the audio output of my IQAudio DacPlus. I use this with Sonic Pi 2.10 On previous versions of Raspbian it is fine. I gather that the same effect has been noted on teh HiFiBerry card as well.

Sam Aaaron (developer of Sonic Pi) also tells me that he has noticed the effect as well. The card plays .wav files OK directly with aplay, so the problem may be connected to the jackd connection that Sonic Pi uses.

multivac61, 2016-05-29:
Can confirm that I have silmilar issues with specific audio output from my HiFiBerry card after upgrading to the lastest Raspbian. I get crystal clear audio when running `aplay` but when I run `jackd` commands and play audio that way the issue becomes apparent.

stuzog, 2016-05-30:
Same issue with Sonic Pi on a RPi2 with a HiFiBerry DAC+. Sonic Pi uses 'jackd' but a `qjackctrl` window shows Jack not running and many instances of "skipped 13" and "skipped 14" in the log window (the distorted sound and clicking). It also shows system audio connected, but not Sonic Pi. The green light on the DAC+ comes on, however, so it is running.

Comments on [this Google group](https://groups.google.com/forum/#!msg/sonic-pi/BvkSTW0Cqnk/nRtzYn4aAwAJ) suggest that it's a Linux problem.

clivem, 2016-05-30:
The long and the short of it, from reading that link......

MMAP is now disabled by default with brcm2835 I2S driver. You need to e-enable it with an overlay, "dtoverlay=i2s-mmap".

stuzog, 2016-05-30:
`dtoverly=i2s-mmap` was already enabled to get Sonic Pi running.

rbnpi, 2016-05-30:
That’s not the problem it is enabled with that line in the config.txt file.

clivem, 2016-05-30:
Sorry, not really clear to me what is going on. That Google thread starts with "jack doesn't work without MMAP" and then halfway through it now seems to be about "ticks" in the audio stream.

rbnpi, 2016-05-31:
Yes. initially mmap was the problem. This was then sorted (a long time ago) and things have been fine until the latest release, when a clicking sound has been introduced using HiFiBerry or IQAudio cards. They work fine with aplay. The problem is in the jack connection necessary for Sonic Pi

stuzog, 2016-06-13:
STUTTERING AUDIO SOLVED -- Updating both distribution (apt-get dist-update) and kernel (using rpi-update) results in clean audio, no stuttering on RPi2 with HiFiBerry DAC+. Thanks to Martin Sperl for this, info via Clive Messer.

Stuart

popcornmix, 2016-06-13:
@rbnpi Please run rpi-update and close if issue is resolved.

rbnpi, 2016-06-14:
Thanks guys for all your hard work. Works perfectly now :-)
Closed.

closed 2016-06-14 by rbnpi