raspberrypi/linux #1503: VC4: different colors with kernel 4.4.11

anyc, 2016-05-30:
After the upgrade to (a self-built) 4.4.11, I noticed that the colors on my HDMI TV look brighter and light colors become white. I guess this has something to do with the VC4 update and "drm/vc4: Add support for gamma ramps". Hence, I wanted to ask if this is intended? I will check if I can change these values somehow as I don't run a DE but just a fullscreen slideshow.

anyc, 2016-05-31:
Okay, although xrandr shows me that the brightness is set to 1, setting it again to 1 seems to fix the colors. I noticed though that the gamma also gets changed from 0.46:0.46:0.46 to 1:1:1 although I only set the brightness.

popcornmix, 2016-05-31:
Are you using the experimental graphics driver?

anyc, 2016-05-31:
Yes, I enabled the overlay and the VC4 module gets loaded. I use Xorg and qmlscene to show a QML-based slideshow (that is not released yet). Everything else looks good. No errors in the kernel log.

popcornmix, 2016-05-31:
Can you report it here:
https://github.com/anholt/linux/issues
We don't directly support the experimental graphics driver, so I don't know if the behaviour is expected.
Eric Anholt who develops it will know.

closed 2016-05-31 by popcornmix