raspberrypi/linux #1507: installing libboost1.55-dev causes kernel panic

JamesCarscadden, 2016-06-01:
Hi there..,

Just got a new RPI3 the other day, trying to install RetroPie (current from github) on top of stock Raspbian image (downloaded on May 31, 2016). While attempting to install libboost1.55-dev I get a kernel panic:

```
Preparing to unpack .../libboost1.55-dev_1.55.0+dfsg-3_armhf.deb ...
Unpacking libboost1.55-dev:armhf (1.55.0+dfsg-3) ...

Message from syslogd@frodo at Jun  1 19:53:05 ...
 kernel:[  283.746318] Internal error: Oops - BUG: 0 [#1] PREEMPT SMP ARM

Message from syslogd@frodo at Jun  1 19:53:05 ...
 kernel:[  283.850366] Process mmcqd/0 (pid: 57, stack limit = 0xad0e0210)

Message from syslogd@frodo at Jun  1 19:53:05 ...
 kernel:[  283.856363] Stack: (0xad0e1d88 to 0xad0e2000)
```

Cleared out all the RetroPie stuff, and tried just installing the package on its own, seems to result in the same problem. Something appears to be wrong with this package.

popcornmix, 2016-06-01:
Do you get any warning symbols on display? (under-voltage/over-temperature)
Are you overclocking? Any other custom config.txt settings?
Can you confirm:
```
uname -a
vcgencmd version
```

JamesCarscadden, 2016-06-01:
Upon further testing, this doesn't appear to be an issue with raspbian. From a new stock image on my card, libboost installed without issue. The kernel panic was on an image that was using Adafruit's PiTFT (and so was using a different kernel). Looks like the issue is there.

JamesH65, 2017-05-18:
Closing as this issue has been deemed unnecessary.

closed 2017-05-18 by JamesH65