raspberrypi/linux #1508: Kernel panic on Pi Zero just after raspistill fails to find the sensor and has exited.

6by9, 2016-06-02:
From https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=150137#p987029
- Pi Zero 1.3
- enable i2c_vc in config.txt
- run raspistill. It errors as it can't find the sensor, but then shortly after quitting the kernel panics.

Logging here to investigate rather than polluting a forum thread that is only vaguely related.

P33M, 2016-06-02:
This historically has never played well, as it's not really a supported configuration. i2c_vc exposes i2c_0 to the ARM on Pi zero so any VPU usage will lead to spurious interrupts prodding the ARM.

We've hardened the GPU against crashes when (ab)use of the GPU-reserved I2C bus is experienced but presumably the ARM-side driver hasn't had the treatment.

A stacktrace would be useful.

6by9, 2016-06-02:
Indeed, it's not a great plan. However I had the recollection that it was all happy though as long as they weren't both actively trying to use the peripheral at the same time - perhaps that's my memory playing tricks on me.

I was going to have an experiment tonight with it to try and get more details - it's only a proxy report from the forums at the moment as I haven't reproduced it, and it may be that we ignore it anyway as "unsupported".

campag, 2016-06-02:
Hi - i just grabbed the kernel log from a serial terminal
[kern.log.txt](https://github.com/raspberrypi/linux/files/295940/kern.log.txt)
It shows references to my sdio wifi driver, esp8089

On another zero without that add-on, no freeze, no crash dump on running raspistill with dtparam=i2c_vc=on.

gregeric.

6by9, 2016-06-02:
Thanks @campag.
I think that actually puts the I2C driver in the clear and I'll close this. 
It's sort of reassuring that I'm not going totally senile over i2c_vc working before, just as long as the two processors don't access the device at the same time.

closed 2016-06-02 by 6by9