raspberrypi/linux #1509: V4L2 stalls after streaming for a while

shahriman, 2016-06-02:
Hi

I am using the bcm2835-v4l2 driver with this camera: https://www.adafruit.com/product/1937. (The camera seems to be pin compatible with the official Raspberry Pi camera.) For a while now, I have been seeing this issue where the V4L2 driver, when asked to stream H.264, will stall after streaming for a while. What I do is, I ask the V4L2 device to stream at a resolution of 320x224@10fps. But then we perform some operation on the frame which can take a while some time. So even though I am asking the camera to stream at 10 fps, I cannot always read at 10 fps.

This does not cause any problem when I am streaming raw BGR24 frames. The problem only arise when I am streaming H.264. The camera will start off fine, stream for a while, and then suddenly the V4L2 driver will start returning EAGAIN. The only way out seems to be to close / reopen the device.

I am attaching my entire kern.log here. But I'd like to highlight some part that may be relevant. The camera seems to be streaming fine until Jun  2 14:36:19:
```
.
.
.
Jun  2 14:36:19 placemetersensor kernel: [ 1202.819303] vb2: vb2_internal_dqbuf: dqbuf of buffer 25, with state 0
Jun  2 14:36:19 placemetersensor kernel: [ 1202.846076] bcm2835-v4l2: buffer_prepare: dev:d65fb000
Jun  2 14:36:19 placemetersensor kernel: [ 1202.846127] bcm2835-v4l2: buffer_queue: dev:d65fb000 buf:cbc24400
Jun  2 14:36:19 placemetersensor kernel: [ 1202.846221] vb2: vb2_internal_qbuf: qbuf of buffer 25 succeeded
Jun  2 14:36:19 placemetersensor kernel: [ 1202.846659] bcm2835-v4l2: buffer_cb: status:0, buf:cbc24400, length:9861, flags 4, pts 1050838340
Jun  2 14:36:19 placemetersensor kernel: [ 1202.846705] bcm2835-v4l2: Convert start time 786.160509 and 788304007 with offset 1050838340 to 1047.694842
Jun  2 14:36:19 placemetersensor kernel: [ 1203.027548] vb2: vb2_internal_dqbuf: dqbuf of buffer 26, with state 0
Jun  2 14:36:19 placemetersensor kernel: [ 1203.055685] bcm2835-v4l2: buffer_prepare: dev:d65fb000
Jun  2 14:36:19 placemetersensor kernel: [ 1203.055740] bcm2835-v4l2: buffer_queue: dev:d65fb000 buf:cbc24800
Jun  2 14:36:19 placemetersensor kernel: [ 1203.055832] vb2: vb2_internal_qbuf: qbuf of buffer 26 succeeded
Jun  2 14:36:19 placemetersensor kernel: [ 1203.056251] bcm2835-v4l2: buffer_cb: status:0, buf:cbc24800, length:3833, flags 4, pts 1051138321
Jun  2 14:36:19 placemetersensor kernel: [ 1203.056291] bcm2835-v4l2: Convert start time 786.160509 and 788304007 with offset 1051138321 to 1047.994823
Jun  2 14:36:19 placemetersensor kernel: [ 1203.254280] vb2: vb2_internal_dqbuf: dqbuf of buffer 31, with state 0
Jun  2 14:36:19 placemetersensor kernel: [ 1203.281013] bcm2835-v4l2: buffer_prepare: dev:d65fb000
Jun  2 14:36:19 placemetersensor kernel: [ 1203.281064] bcm2835-v4l2: buffer_queue: dev:d65fb000 buf:cbc25c00
Jun  2 14:36:19 placemetersensor kernel: [ 1203.281574] bcm2835-v4l2: buffer_cb: status:0, buf:cbc25c00, length:3171, flags 4, pts 1051338309
Jun  2 14:36:19 placemetersensor kernel: [ 1203.281615] bcm2835-v4l2: Convert start time 786.160509 and 788304007 with offset 1051338309 to 1048.194811
```

Then it seems that the V4L2 driver has stopped receiving buffer_cb callback:
```
Jun  2 14:36:19 placemetersensor kernel: [ 1203.281676] vb2: vb2_internal_qbuf: qbuf of buffer 31 succeeded
Jun  2 14:36:22 placemetersensor kernel: [ 1205.851781] vb2: vb2_internal_dqbuf: dqbuf of buffer 27, with state 0
Jun  2 14:36:22 placemetersensor kernel: [ 1205.908001] bcm2835-v4l2: buffer_prepare: dev:d65fb000
Jun  2 14:36:22 placemetersensor kernel: [ 1205.908054] bcm2835-v4l2: buffer_queue: dev:d65fb000 buf:cbc24c00
Jun  2 14:36:22 placemetersensor kernel: [ 1205.908082] vb2: vb2_internal_qbuf: qbuf of buffer 27 succeeded
Jun  2 14:36:22 placemetersensor kernel: [ 1206.273099] vb2: vb2_internal_dqbuf: dqbuf of buffer 28, with state 0
Jun  2 14:36:22 placemetersensor kernel: [ 1206.317200] bcm2835-v4l2: buffer_prepare: dev:d65fb000
Jun  2 14:36:22 placemetersensor kernel: [ 1206.317253] bcm2835-v4l2: buffer_queue: dev:d65fb000 buf:cbc25000
Jun  2 14:36:22 placemetersensor kernel: [ 1206.317278] vb2: vb2_internal_qbuf: qbuf of buffer 28 succeeded
Jun  2 14:36:23 placemetersensor kernel: [ 1206.535489] vb2: vb2_internal_dqbuf: dqbuf of buffer 29, with state 0
Jun  2 14:36:23 placemetersensor kernel: [ 1206.588758] bcm2835-v4l2: buffer_prepare: dev:d65fb000
Jun  2 14:36:23 placemetersensor kernel: [ 1206.588811] bcm2835-v4l2: buffer_queue: dev:d65fb000 buf:cbc25400
Jun  2 14:36:23 placemetersensor kernel: [ 1206.588841] vb2: vb2_internal_qbuf: qbuf of buffer 29 succeeded
Jun  2 14:36:23 placemetersensor kernel: [ 1206.798182] vb2: vb2_internal_dqbuf: dqbuf of buffer 30, with state 0
Jun  2 14:36:23 placemetersensor kernel: [ 1206.826464] bcm2835-v4l2: buffer_prepare: dev:d65fb000
Jun  2 14:36:23 placemetersensor kernel: [ 1206.826518] bcm2835-v4l2: buffer_queue: dev:d65fb000 buf:cbc25800
Jun  2 14:36:23 placemetersensor kernel: [ 1206.826545] vb2: vb2_internal_qbuf: qbuf of buffer 30 succeeded
Jun  2 14:36:24 placemetersensor kernel: [ 1207.661739] vb2: vb2_internal_dqbuf: dqbuf of buffer 1, with state 0
Jun  2 14:36:24 placemetersensor kernel: [ 1207.689786] bcm2835-v4l2: buffer_prepare: dev:d65fb000
Jun  2 14:36:24 placemetersensor kernel: [ 1207.689843] bcm2835-v4l2: buffer_queue: dev:d65fb000 buf:d73dd400
Jun  2 14:36:24 placemetersensor kernel: [ 1207.689874] vb2: vb2_internal_qbuf: qbuf of buffer 1 succeeded
Jun  2 14:36:24 placemetersensor kernel: [ 1207.913602] vb2: vb2_internal_dqbuf: dqbuf of buffer 2, with state 0
Jun  2 14:36:24 placemetersensor kernel: [ 1207.938773] bcm2835-v4l2: buffer_prepare: dev:d65fb000
Jun  2 14:36:24 placemetersensor kernel: [ 1207.938824] bcm2835-v4l2: buffer_queue: dev:d65fb000 buf:d73dc800
Jun  2 14:36:24 placemetersensor kernel: [ 1207.938851] vb2: vb2_internal_qbuf: qbuf of buffer 2 succeeded
Jun  2 14:36:24 placemetersensor kernel: [ 1208.121116] vb2: vb2_internal_dqbuf: dqbuf of buffer 3, with state 0
Jun  2 14:36:24 placemetersensor kernel: [ 1208.161817] bcm2835-v4l2: buffer_prepare: dev:d65fb000
Jun  2 14:36:24 placemetersensor kernel: [ 1208.161868] bcm2835-v4l2: buffer_queue: dev:d65fb000 buf:d56ddc00
Jun  2 14:36:24 placemetersensor kernel: [ 1208.165516] vb2: vb2_internal_qbuf: qbuf of buffer 3 succeeded
Jun  2 14:36:24 placemetersensor kernel: [ 1208.359109] vb2: vb2_internal_dqbuf: dqbuf of buffer 4, with state 0
Jun  2 14:36:24 placemetersensor kernel: [ 1208.392912] bcm2835-v4l2: buffer_prepare: dev:d65fb000
Jun  2 14:36:24 placemetersensor kernel: [ 1208.392965] bcm2835-v4l2: buffer_queue: dev:d65fb000 buf:d56dd000
Jun  2 14:36:24 placemetersensor kernel: [ 1208.393338] vb2: vb2_internal_qbuf: qbuf of buffer 4 succeeded
Jun  2 14:36:25 placemetersensor kernel: [ 1209.323683] vb2: vb2_internal_dqbuf: dqbuf of buffer 5, with state 0
Jun  2 14:36:25 placemetersensor kernel: [ 1209.359398] bcm2835-v4l2: buffer_prepare: dev:d65fb000
Jun  2 14:36:25 placemetersensor kernel: [ 1209.359449] bcm2835-v4l2: buffer_queue: dev:d65fb000 buf:d56dd800
Jun  2 14:36:25 placemetersensor kernel: [ 1209.359482] vb2: vb2_internal_qbuf: qbuf of buffer 5 succeeded
Jun  2 14:36:26 placemetersensor kernel: [ 1209.564441] vb2: vb2_internal_dqbuf: dqbuf of buffer 6, with state 0
[... 93 lines cut ...]
```
pi@placemetersensor ~ $ uname -a
Linux placemetersensor 4.1.6+ #810 PREEMPT Tue Aug 18 15:19:58 BST 2015 armv6l GNU/Linux
pi@placemetersensor ~ $ modinfo bcm2835
bcm2835-rng   bcm2835-v4l2  bcm2835_wdt   
pi@placemetersensor ~ $ modinfo bcm2835-v4l2 
filename:       /lib/modules/4.1.6+/kernel/drivers/media/platform/bcm2835/bcm2835-v4l2.ko
version:        0.0.2
license:        GPL
author:         Vincent Sanders
description:    Broadcom 2835 MMAL video capture
srcversion:     48211A5D67D3C76719949A5
depends:        videobuf2-core,videodev,videobuf2-vmalloc,v4l2-common
intree:         Y
vermagic:       4.1.6+ preempt mod_unload modversions ARMv6 
parm:           debug:int
parm:           bcm2835_v4l2_debug:Debug level 0-2
parm:           max_video_width:Threshold for video mode (int)
parm:           max_video_height:Threshold for video mode (int)
parm:           gst_v4l2src_is_broken:If non-zero, enable workaround for Gstreamer (int)
```

I can share more information as required.

[kern_v4l2_hang.log.gz](https://github.com/raspberrypi/linux/files/295959/kern_v4l2_hang.log.gz)

shahriman, 2016-06-02:
@6by9 It will be great if you can take a look. Any help in understanding the problem, if not fixing it, will be greatly appreciated.

6by9, 2016-06-02:
When all the buffers are returned, note that the length of each is 0, as well as the timestamps being MMAL_TIME_UNKNOWN (https://github.com/raspberrypi/userland/blob/master/interface/mmal/mmal_types.h#L88). The codec is just being flushed of all output buffers.

I'm afraid that not being able to handle the data rate with H264 is always going to end badly. There is a ~2MB FIFO in the output side of the codec, so that will take a while to fill, and then it will start applying back pressure to the camera system. The camera system should deal with that, at least up to some point. There is an error callback if the system is stalled for a significant period, although I can't remember how long that is, nor what it does at that point. 
The V4L2 driver doesn't connect to the control port, so it won't see the error(s) that would be thrown there anyway.

BGR has significantly less buffering (it takes the current frame), so you're less likely to stall for that long.
(Do NOT try it with MJPEG - the codec is almost guaranteed to lock up!).

Only a couple of thoughts:
- add ```start_debug=1``` to /boot/config.txt, and run ```sudo vcdbg log assert``` if/when it fails.
- you're on linux 4.1.6 on a build that is over 9 months old. I must admit that I can't think of anything that has changed since then that might affect this, but an update wouldn't go amiss.

shahriman, 2016-06-02:
This is what I get with the existing kernel:
```
pi@placemetersensor ~ $ sudo vcdbg log assert
001618.919: assert( card not initialised ) failed; ../../../../../filesystem/media/sdcard/sdcard.c::sdhost_close line 1045
014058.564: assert( camera_subsystem_state.power_enable_ref_count[port] > 0 ) failed; ../../../../../vcfw/drivers/device/camera_subsystem/camera_subsystem.c::camera_disable_ctrl line 1116
```
I will upgrade the kernel and report again. Thanks.

6by9, 2016-06-02:
Yes, those are two lurking asserts that always get logged. I'm surprised there's nothing else when it goes wrong.

shahriman, 2016-06-02:
Updated the firmware to latest and the problem seems to have disappeared. Here's the version I have now:
```
pi@placemetersensor ~ $ uname -a
Linux placemetersensor 4.1.19+ #858 Tue Mar 15 15:52:03 GMT 2016 armv6l GNU/Linux
pi@placemetersensor ~ $ modinfo bcm2835-v4l2
filename:       /lib/modules/4.1.19+/kernel/drivers/media/platform/bcm2835/bcm2835-v4l2.ko
version:        0.0.2
license:        GPL
author:         Vincent Sanders
description:    Broadcom 2835 MMAL video capture
srcversion:     872FEF5DC420FE51D9A79D1
depends:        videobuf2-core,videodev,videobuf2-vmalloc,v4l2-common
intree:         Y
vermagic:       4.1.19+ mod_unload modversions ARMv6 
parm:           debug:int
parm:           bcm2835_v4l2_debug:Debug level 0-2
parm:           max_video_width:Threshold for video mode (int)
parm:           max_video_height:Threshold for video mode (int)
parm:           gst_v4l2src_is_broken:If non-zero, enable workaround for Gstreamer (int)
```
With the old firmware mentioned above, I could reproduce the issue consistently within 5-15 mins every time. After the updated, I have been testing for a couple of hours now - and the issue did not resurface. May be the upgrade helped after all. I will keep this issue open until tomorrow and close if the issue does not appear again with the latest driver.

Thanks for looking into this.