raspberrypi/linux #1513: Maintaining old kernel branches

betafive, 2016-06-04:
The upstream 4.1.y branch is planned to be maintained until around Sept 2017 (https://www.kernel.org/category/releases.html). The 4.1.y branch here hasn't been updated in around 2 months so I'm guessing all focus has moved to 4.4.y and 4.1.y won't recieve any further updates. Is that correct?

If there aren't any "official" plans to maintain the Raspberry Pi 4.1.y branch any further then do you know of any unofficial efforts to do this?

My plan is to maintain my own 4.1.y branch until its upstream EOL, basically starting with the last commit to 4.1.y here and merging in each new linux-stable release. I've done some initial testing with 4.1.25 merged in and it looks good to me. Just wanted to check before I put too much effort into this whether I'm duplicating anything done elsewhere.

popcornmix, 2016-06-04:
We won't be updating kernels older than 4.4 and I'm not aware of anyone else doing it, so feel free to maintain a 4.1 branch.

betafive, 2016-06-04:
Thanks for the quick reply. The repository is at https://gitlab.com/oryx/linux-oryx-rpi if it's useful to anyone else.

closed 2016-06-04 by betafive