raspberrypi/linux #1516: settled doesnt work with scrolllock

n0rt0nthec4t, 2016-06-05:
The latest raspbain release using setleds -L +scroll no longer sets the scroll lock led on. Using +caps or +num still works

This was working on a build of RetroPie 3.6/3.7 but the latest 3.8.1 no longer works. I tried a clean installed on Jessie (2016/5/27) and the problem still persists

popcornmix, 2016-06-07:
Are you getting an error message, or does it appear to work but not affect the LEDs?

n0rt0nthec4t, 2016-06-07:
No error message and setleds says the led should be on but it does not light.  Some more testing I have been doing using ioctl directly to set keyboard LEDs and again, nun and caps work but not scroll

popcornmix, 2016-06-07:
What is the kernel version on working and non-working systems? (``uname -a``)

n0rt0nthec4t, 2016-06-07:
non-working is
Linux AmigaPI 4.4.11-v7+ #888 SMP Mon May 23 20:10:33 BST 2016 armv7l GNU/Linux

I'll post the work tonight, but from memory the kernel was 4.1.x

popcornmix, 2016-06-07:
Can you identify the exact update which caused this. See:
https://github.com/Hexxeh/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to revert back to that version. Report the first version with this error.

n0rt0nthec4t, 2016-06-08:
The kernel commit on 12/4/2016 to update to 4.4.6 breaks this. Prior to this, all works fine

date	commit hash	name -a	Status
7 April 2016	dea25fa62132365c11087e51e416df656db28bf3	Linux retropie 4.1.21-v7+ #872 SMP Wed Apr 6 17:34:14 BST 2016 armv7l GNU/Linux	OK
9 April 2016	4bc6b67905e3206f8cf4d9a316f3343aaaf14d62	Linux retropie 4.1.21-v7+ #872 SMP Wed Apr 6 17:34:14 BST 2016 armv7l GNU/Linux	OK
11 April 2016	826a75550fc731f2a2dacf5d776d09d008ec37b5	Linux retropie 4.1.21-v7+ #872 SMP Wed Apr 6 17:34:14 BST 2016 armv7l GNU/Linux	OK
12 April 2016	771a503cfc2a1130e2df2a4ddfc45ffa0f592b3f	Linux retropie 4.1.21-v7+ #873 SMP Mon Apr 11 18:22:10 BST 2016 armv7l GNU/Linux	OK
12 April 2016	c05b9e6fba9cc2ce9e47abf41b8d413b2a69d952	Linux retropie 4.4.6-v7+ #874 SMP Mon Apr 11 19:20:24 BST 2016 armv7l GNU/Linux	Failed

popcornmix, 2016-06-08:
Okay, this will be an upstream change that likely affects all linux platforms running 4.4.x kernels.
What is the make/model of keyboard?

n0rt0nthec4t, 2016-06-08:
Generic Dell 101 keyboard
Microsoft Natural Keyboard
Keyrah v2 connected to Amiga 600 keyboard

n0rt0nthec4t, 2016-06-13:
an updated.. creating a udev rule with the following enables the scrollock led to function as per normal. Wonder if there is typo somewhere up stream give the line below references "scrolllock" with 3 Ls vs the trigger name of "scrollock" with just 2 Ls

ACTION=="add", SUBSYSTEM=="leds", ENV{DEVPATH}=="*/input*::scrolllock", ATTR{trigger}="kbd-scrollock"

prefim, 2016-08-29:
I've put that udev line into a file call 00-scrollled.rules into the etc/udev/rules.d folder but its not seeming to make any difference. any thoughts?

n0rt0nthec4t, 2016-08-29:
Appears I have a typo in my rules above

create a file in /etc/udev/rules.d called 50-leds.rules

It contains the following

ACTION=="add", SUBSYSTEM=="leds", ENV{DEVPATH}=="*/input*::scrolllock", ATTR{tri
gger}="kbd-scrollock"

pelwell, 2016-08-29:
Try with three "l"s in "kbd-scrolllock".

n0rt0nthec4t, 2016-08-29:
and some wildcards that seem to have gotten remove in my initial report.. fixed above

prefim, 2016-08-29:
I wonder if the same fix can be applied for caps lock?

pelwell, 2016-08-29:
But the source (http://lxr.free-electrons.com/source/drivers/input/input-leds.c?v=4.4#L30) says:
```
[LED_SCROLLL]   = { "scrolllock", VT_TRIGGER("kbd-scrolllock") },
```

JamesH65, 2017-05-18:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.

closed 2017-05-18 by JamesH65

markusobi, 2019-05-14:
This issue is still relevant. It is still broken on latest raspbian:
Tested on on Raspberry Pi 3B+
OS:
Raspbian Stretch with desktop
Image with desktop based on Debian Stretch
Version: April 2019
Release date: 2019-04-08
Kernel version: 4.14

Scroll lock still doesn't work

JamesH65, 2019-05-14:
Just tried `setleds -L +scroll` and a number of other combinations on a pi3B+ with 4.19.42 and it seemed to work as expected.