raspberrypi/linux #1517: I2S (RPI as a slave) audio stuttering on RPI 3 since 4.4.6

gtrainavicius, 2016-06-05:
Hello,

We are developing an audio card for RPI with one stereo input and one stereo output, and we've hit a stuttering issue with playing audio while at the same time recording. If audio is just being played without recording at the same time it is played alright.

On our board we use the following configuration: the ADC is the I2S master, DAC is a slave and RPI is a slave.

Furthermore, the same card with the same module code works just fine on RPI 2, even on the 4.4.x kernel, and RPI 3 itself works fine on 4.1.x kernel, with the same code apart from .dtb instead of .dtbo for the overlay and one argument less for GPIOD family of functions.

Please see the screenshot below, it was produced by loopbacking the input and output of the card with a cable, generating a 440Hz 30 second sine wave using audacity, then recording it using different sampling rates. As you can see, the 96000 sampling rate doesn't even record anything.

[image]

Thank you,
Giedrius.

clivem, 2016-06-05:
I'm clutching at straws, but ISTR that the fix for I2S channel swap went in around 4.4.6 or 4.4.7. 
https://github.com/raspberrypi/linux/commit/3de232825c9ba5989522b8691eb6ac5df6619458

pelwell, 2016-06-05:
Was it the switch from 4.1 to 4.4 that caused the problem, or a change within 4.4? You can use rpi-update to choose a specific firmware release using the commit IDs listed [here](https://github.com/Hexxeh/rpi-firmware/commits/master).

gtrainavicius, 2016-06-05:
I've narrowed it down to be between 4.1.21 (still working) and 4.4.6 (broken) releases.

Keep in mind that RPI 2 using 4.4.6 and newer works well, so it must be caused by some difference between the RPI 2 and RPI 3.

Let me know what you'd like me to try out, I've already tried disabling bluetooth as suggested in other I2S issue, and disabling wireless lan, but the issue still persists.

pelwell, 2016-06-05:
Have you tried more recent commits? There was a change just before the jump to 4.4.10 that squashed  a lot of unwanted UART interrupts.

gtrainavicius, 2016-06-05:
The same issue reproduces with all the 4.4.x releases I've tried, including the latest 4.4.12.

gtrainavicius, 2016-06-05:
> I'm clutching at straws, but ISTR that the fix for I2S channel swap went in around 4.4.6 or 4.4.7.

I have tried reverting the mentioned commit and rebuilding snd-soc-bcm2835-i2s module, but the issue still reproduces.

clivem, 2016-06-05:
> I have tried reverting the mentioned commit...

Yes, that really was clutching at straws, rather than any real expectation that reverting that commit would solve your issue. Just out of interest, do you have MMAP enabled, either by hacking support back into the bcm2835-i2s driver or using 'dtoverlay=i2s-mmap'?

gtrainavicius, 2016-06-05:
The issue still occurs with i2s-mmap enabled, but the behavior is slightly different - when trying to record at 96kHz, the recording is stuck, instead of recording silence. The other sampling rates behave the same as shown in the previous screenshot.

clivem, 2016-06-05:
@msperl Martin and @HiassofT Matthias, do you have any thoughts?

HiassofT, 2016-06-05:
Just tested simultaneous playback and recording with the Cirrus Logic Audio Card, using kernel 4.4.12 with my cirrus driver and everything worked as expected.

This was also on RPi3, 96kHz stereo S16_LE. But I used arecord and aplay on the command line, not audacity.

@gtrainavicius can you check if this also fails with arecord/aplay?

One difference between my cirrus tree and the RPi tree is that it includes fixes for the broken period/frame calculation and allow periods up to 64k-4 bytes (instead of 32k). But the period size code is the same in 4.1 and 4.4 so I don't think that could explain your issues.

Here's my tree, the period fixes are the first 4 commits on top of the rpi tree (the ones with bcm2835-i2s/dma in the commit message)
https://github.com/HiassofT/rpi-linux/tree/cirrus-4.4.y

gtrainavicius, 2016-06-05:
I can't use 16 bit sample formats - the card's ADC and DAC run on 24 bits.

Using arecord & aplay on 96kHz seems to be recording, but trying to record on 192kHz still results in silence gaps every ~128 milliseconds.

Furthermore, I've seen occasionally getting short bursts of random or old data in the similar period of time, hinting that this might be some wrong or mismatched buffer size calculations somewhere.

[image]

Still I can't figure out why the RPI 2 works just fine with 4.4.12 kernel while RPI 3 has these issues on the same kernel version (and no problems on 4.1.x kernels)

One thing that's different within the same kernel release is the device tree between RPI 2 and RPI 3. Are there more such areas that could make a difference in the same release?

Would it be possible for me to do a 'binary search' in between the commits of 4.4.6 and 4.1.21 releases to find the exact moment the issue I'm seeing appeared?

HiassofT, 2016-06-05:
Have you checked with aplay/arecord on the command line? Especially look out for overrun/underrun errors, you might be running into an IO bandwidth issue with the SD card (or did you record to an USB hdd?).

With the original 8GB (black Samsung) NOOBS card I had dropouts/underruns when playing back 192kHz 24/32 bit WAVs on RPi2 and RPi3 when they weren't cached in RAM, using a 32GB EVO / EVO+ ifixed that. That was only playback, haven't checked simultaneous playback/recording yet.

msperl, 2016-06-06:
I am currently on a business travel, so got minimal access to rpi hw, but some things come to my mind:
Does it happen:
* with the upstream dma patch as well?
* does it show up writing to a RAM disk? (My estimate indicates that 60 seconds should consume about 64mb, so I guess it should be possible to reproduce it in such a window)
* does it show when using the same mmc driver (mmc or Sochi - can't remember which is used where) on the rpi3 as on the rpi2 - there is a difference here, so that the rpi3 supports wifi!
* does it happen when SRAM calibration is disabled in the firmware - I heard there was a means to do this.

The reason why I ask the last point is because the one thing that may be different between a rpi2 and a rpi3 - assuming you run identical kernels - could be different behavior in the firmware. Maybe the firmware is clocking down the arm/Vpu because of heat or similar when those glitches happen, which would mean changing the Vpu frequency, which would mean changing the axi clock, which would mean shortly stopping Vpu/axi clock, but then it should impact playback as well and not only recording...
So maybe:
* run the test inside a fridge/freezer to see if this is impacting us.

Also: are there any intervals at which these gaps/glitches happen.

HiassofT, 2016-06-06:
A simple test would be to run "arecord | aplay" on the command line, that'll bypass mass-storage IO.

Or try "dtoverlay=sdhost" on the RPi3, then the RPi3 uses the same SD/MMC driver as the RPi2.

clivem, 2016-06-06:
@gtrainavicius Just to follow-up on one of Martins points about trying the new dma code.... 

Suggest you use the https://github.com/msperl/linux-rpi/tree/rpi-next-merged-patches-not-accepted-upstream if you are going to try it. 

@msperl Have you re-based against latest rpi-4.4.y? Is he going to run into issues with conflicts because of those last clock patches from Eric if he tries to sync it to latest rpi-4.4.y?

popcornmix, 2016-06-06:
> Maybe the firmware is clocking down the arm/Vpu because of heat or similar when those glitches happen, which would mean changing the Vpu frequency

run ``vcgencmd get_throttled`` after the glitch.
If that is zero then no throttling has occurred since boot. If it is non-zero you have had throttling and that may be suspicious.

rbnpi, 2016-06-06:
I think this may be related to this issue I reported #1497 about problems with IQAudio card.

gtrainavicius, 2016-06-06:
I have tried recording and playing using `arecord | aplay` with 48k, 96k and 192k rates, it was fine, no stuttering. Also there was no stuttering if recording and playing back a .wav file at the same time instead of piping arecord and aplay, showing that sdcard speed isn't causing this issue. Additionally, when I was looking for the release on which this issue first appeared, I was always using the same sd card for the tests.

The issue seems to reliably occur thus far when playing a sound and recording at the same time with Audacity. If Audacity is recording and not playing anything, it records fine.

I have tried dtoverlay=sdhost and the issue is still the same.

vcgencm get_throttled shows 0 after reproducing the issue. The raspberry pi board itself doesn't feel like it's getting too hot.

gtrainavicius, 2016-06-06:
> Also: are there any intervals at which these gaps/glitches happen.

Yes, there is a repeating pattern - the glitch occurs every 128 millisecods, with the correct audio lasting for ~85ms and the glitch lasting for the remaining ~43 milliseconds. See the selection time start and end at the bottom of this screenshot:

[image]

> but then it should impact playback as well and not only recording...

Actually it does impact both - even without anything connected to the input the same glitchy pattern occurs on both playback and recording if Audacity is doing both at the same time.

The repeating glitch itself seems to be some 'old' audio data which was played or recorded at some point previously, it seems like some memory copying operation doesn't fill the buffer fully, or some sizes are mismatched.

I don't know what Audacity is doing differently than aplay and arecord which don't produce this glitch, but I don't blame it - the same Audacity with older kernel releases work fine.

gtrainavicius, 2016-06-06:
@clivem @msperl I have tried building and running the kernel at https://github.com/msperl/linux-rpi/tree/rpi-next-merged-patches-not-accepted-upstream and the glitches are gone! Kernel version reported by uname is 4.4.8-v7

When are you planning to merge the fixes to the main branch? I could run more builds to help you pick out just the right commits to fix this issue.

clivem, 2016-06-06:
> When are you planning to merge the fixes to the main branch?

Well, that's up to @msperl and @pelwell. It isn't so much a case of merging a patch or two. Although the dma code patch set can be merged in isolation, independently of the i2s backport and switch to clockmgr.

I wonder if one of the SonicPi people with issues could build and test from that @msperl branch? 
@rbnpi I wonder if you want to try building from https://github.com/msperl/linux-rpi/tree/rpi-next-merged-patches-not-accepted-upstream and report back.

msperl, 2016-06-07:
I guess it is only dma related, so just merging that dma-patch set  could solve that issue.

As mentioned above: I am on a business trip and do not have the hw available to test these things - too much focused on Java performance issues... Also some portions are very slow moving upstream - especially the sdram-related patches are still "pending" and Eric is producing an alternative patch for these... So this will take some longer.

msperl, 2016-06-07:
@gtrainavicius : there is definitely a pattern when the "issue is occurring" - the pattern look actually identical at those times of glitches.
As said: I guess it is the old dma code that is triggering it - not knowing the exact settings that the kernel is using for the cyclic "dma" transfers in your case, I would assume that you could reach such a situation where you could trigger one big dma transfer plus one small one - each producing an interrupt and then there could be some latencies on top that could trigger some delays. Or that the accounting of which buffer is filled is badly accounted for/badly calculated because of more interrupts than expected...
If I remember correctly there was one (one-line) patch that went in sometime that is related to correctly accounting the offsets, which might make the situation worse under the right circumstances!

Please try the dma-patch #1433 - I know that it does not apply any longer, but I can not figure out using github alone on my iPad what commit is producing the conflict...

HiassofT, 2016-06-07:
I can confirm that it's a DMA issue, most certainly the broken period/fragment code (or something in that code triggering the issue).

With my Cirrus tree and my simple DMA period/fragment fixes audacity works fine, but if I revert them (bcm2835-i2s/dma as in current rpi-4.4.y) I see the same pattern.

I think it's best to get the DMA fixes by @msperl into the 4.4 tree as they are already upstreamed.

I did a quick test (reverting downstream commits and picking from 4.7 & msperl trees) but it didn't quite work out (playback worked fine but recording was stuck). Either I messed up somewhere or I missed some important commit - diverging trees are fun.

I'm a little bit short on time as well so if someone wants to pick up here I wouldn't mind that at all.

clivem, 2016-06-07:
Just need to take the first 16 patches from @msperl rpi-next-merged-patches-not-accepted-upstream, and tweak number 16, the one that adds dma to the vc4-kms-v3d-overlay.

@gtrainavicius Can you build a kernel from this branch... https://github.com/DigitalDreamtimeLtd/linux/commits/rpi-4.4.y-dma
....and confirm that your problem is solved, in which case we are 100% sure it is the dma code and we have a patchset that will apply cleanly to current rpi-4.4.y.

gtrainavicius, 2016-06-07:
@clivem Just tried building from above link, the original issue does not reproduce with that code. uname reports 4.4.12-v7 version.

HiassofT, 2016-06-07:
@clivem @gtrainavicius could you test if playback and recording works at 192/24 or 192/32 with periods larger than 65532 bytes? It doesn't seem to work here...

According to /proc/asound/card0/pcm0p/sub0/hw_params Audacity uses a period size of 4800 audio frames / 38400 bytes when playing (works fine) and 8192 audio frames / 65536 bytes when recording (nothing recorded). 

I can also reproduce that with arecord/aplay. This works:
```
arecord -v --period-size 8191 --buffer-size 32764 -r 192000 -f S32_LE -c 2 /tmp/rec.wav
```

But this doesn't (lots of overruns):
```
arecord -v --period-size 8192 --buffer-size 32768 -r 192000 -f S32_LE -c 2 /tmp/rec.wav
```

aplay using mmap also gives lots of underruns at period-size 8192 (fine at 8191):
```
aplay -v --mmap --period-size 8192 --buffer-size 32768 audiotest/192-32.wav
```

It looks like the period splitting code is still not correct (I thought we had that working in some earlier incarnation of the DMA patches).

I added some tracing to the control-block setup:
```
--- a/drivers/dma/bcm2835-dma.c
+++ b/drivers/dma/bcm2835-dma.c
@@ -354,6 +354,7 @@ static struct bcm2835_desc *bcm2835_dma_create_cb_chain(
                                len, period_len, &total_len,
                                cyclic ? finalextrainfo : 0);
 
+                       printk(KERN_INFO "frame %d: length %d\n", frame, control_block->length);
                        /* calculate new remaining length */
                        len -= control_block->length;
                }
```

With period-size 8191 audio frames (65528 bytes, within limit of lite channels) everything's fine:
```
[ 1481.478840] frame 0: length 65528
[ 1481.478857] frame 1: length 65528
[ 1481.478866] frame 2: length 65528
[ 1481.478875] frame 3: length 65528
```

But with period-size 8192 (65536 bytes, needs splitting for lite channels) it doesn't look good:
```
[ 1485.312058] frame 0: length 65532
[ 1485.312075] frame 1: length 65532
[ 1485.312086] frame 2: length 65532
[ 1485.312096] frame 3: length 65532
[ 1485.312105] frame 4: length 16
[ 1485.312113] frame 5: length 0
[ 1485.312122] frame 6: length 0
[ 1485.312131] frame 7: length 0
```

gtrainavicius, 2016-06-07:
@HiassofT Looks like I can reproduce the same overruns. Audacity with longer 'recording' buffer size (in terms of milliseconds) also fails to work, but works with smaller ones.

clivem, 2016-06-07:
> could you test if playback and recording works at 192/24 or 192/32 with periods larger than 65532 bytes? It doesn't seem to work here...

Hmmm. Although I have 20 or so I2S boards/HATs, I don't really have any interest in capture so don't have one that does. ;) From a playback perspective, the max I normally use is buffer size=16384, period size=4096.... Unfortunately I'm not in a position to test any other playback settings until this evening.

HiassofT, 2016-06-07:
OK, found the bug: total_len was never increased so the period splitting code couldn't work as it should.

Here's a tree with the fix: https://github.com/HiassofT/rpi-linux/tree/rpi-4.4-dmafix

I've also fixed the code so that it won't generate very short control blocks. The original implementation would split a 64k bytes period into a 64k-4 bytes frame plus a 4 bytes frame which could be problematic. The new code ensures that split frames are larger than max_len/2 and generates two 32k frames instead.

@clivem @gtrainavicius could you please test with that tree?

@msperl if you find some time please review if the fix is correct

msperl, 2016-06-07:
@HiassofT : I wonder why this would make a difference - we only trigger the dma on the last one, so that portion should not be a problem.
Writing a 64k as:
* 64k-4 bytes
* 4 bytes + trigger interrupt
Should not make a difference, as only the interrupt is important and as the i2s has some buffer of 64 entries for rx and rx. Also transfers for rx would trigger Dreq on one sample in buffer, so reading these bytes should never be an issue. (But that is with the default register values for rxthr=0).

So I am unconvinced that this is really relevant.

msperl, 2016-06-07:
Also looking at the patch you are mixing 2 different issues you mention:
* fix period splitting (which I can not identify)
* splitting at half the size (which is - as far as I can tell - all that this patch does, and which I believe should not be necessary)

Please (as per the upstream mantra): one patch per issue.

gtrainavicius, 2016-06-07:
@HiassofT I have just built your latest branch and the overruns are gone, and Audacity now works with higher buffer sizes.

HiassofT, 2016-06-07:
@msperl I'm not sure about the internals of the DMA controller but my understanding is that at the end of each control block the DMA controller has to fetch the next one from RAM.

So when we have very short (1 word) transfers this'll result in quite some overhead (fetch control block, do 1 word DMA transfer, fetch next control block before next transfer can be done). That increases pressure on the RAM controller and might potentially lead to issues.

I'll split the patch, the period splitting fix is the *ottal_len += control_block->length; line BTW - I also mentioned that in the commit message.

msperl, 2016-06-07:
If there are such issues then these should be intermittent and not happening every time.

Still: any kind of rx triggers Dreq when a single sample is in the buffer - and we got 64 x32 bit samples as rx and tx buffer.

Even at 192k @32bit/channel this would only fill in 0.000166 seconds (1/6000 s) and that should be sufficient time for reading a single cache line of 32 bytes to read the control block - even the tiniest microcontroller should be able to read that much data from its ram in that time. (Again: that assumes that rxthr=0, but I am not sure what the i2s driver really sets...

So if there is an issue then I would tend to think that this total length thing could be the triggering issue, but I am unsure how your patch really looks to give a review.

HiassofT, 2016-06-08:
These are really 2 independent things. Just setting total_len fixed the reported issue in the tests here (also in the case period_len=65536) and the control block layout was as expected (respecting period_len).

Grep the original code for total_len and you'll see that it can't work that way, total_len is only set to 0 and never assigned a different value, a clever compiler might just have optimized that away.

Adjusting the last-but-one and last block sizes is basically a precaution and a rather easy change. I'm not sure if this is really an issue but having dealt a lot with timing critical stuff I can tell you that it's best to avoid possible high-traffic/congestion/latency situations in the first place because debugging them when they surface is next to impossible.

msperl, 2016-06-08:
So the way I understand it, the first portion fixes the issue.

The second portion is just a cleanup and avoids potential issues (or in other words it could hide some other issues down the road with some drivers setting up Dreq in the wrong way...)

Please share the separate patches - I would be glad to add an ack to it.

One thing: to avoid unnecessary side effects it would help if someone would do long transfers via spi - say to a tftfb device using a block size > 64 k to see if that still works properly as well... 
A real device for testing would help, as it would show display issues when showing a movie or similar.

HiassofT, 2016-06-08:
@msperl I've split the patch and added more detailed explanations to the commit messages. I hope it's clearer now what the patches are doing.

BTW: these changes should have no impact on sg or memcpy DMA. In these cases period_len is 0 and the function bails out early before hitting the new code path.

clivem, 2016-06-08:
@HiassofT Just about to build with your latest patches now and do some testing.....

@msperl I don't know if you realise this, and not that it really matters, (well, not to me anyway), but with the i2s/clockmgr driver backport, max ALSA buffer_size=65536 and period_size=32768. Not sure where those values are coming from now, since snd_pcm_hardware/snd_dmaengine_pcm_config was removed from bcm2835-i2s.c and not used in the call to snd_dmaengine_pcm_register.

msperl, 2016-06-08:
@HiassofT : I am still surprised that this piece of code does not detect the situation:
https://github.com/HiassofT/rpi-linux/blob/3a3015a3c1c2ae410856e84eed6b4b8e637d9e0d/drivers/dma/bcm2835-dma.c#L382
(that is what it was supposed to detect)

Ah no - I guess I understand:
with 64K transfer and 32K block size we get:
* first Transfer is 64K-4bytes + interrupt
* second Transfer is 4bytes + interrupt

but what it should be is (and what the first patch fixes):
* first Transfer is 32Kbytes + interrupt
* second Transfer is 32KBytes + interrupt 

Besides that please send the patches out to upstream.
You can add my sign-off when sending it:
```Signed-off-by: Martin Sperl <kernel@martin.sperl.org>```

Thanks, Martin

clivem, 2016-06-08:
OK, to move forward..... @pelwell How do you feel about me submitting a pull request for this 18 patch dma patch series and taking it straight into rpi-4.4.y? As this is a subset of what was the complete dma/i2s/clockmgr backport, do we still need to go via a "next" branch for the dma part of the complete patch set?
I've added @msperl sign-off and my Tested-by to the 2 patches from @HiassofT. I've also re-synced to latest rpi-4.4.y. https://github.com/DigitalDreamtimeLtd/linux/commits/rpi-4.4.y-dma

msperl, 2016-06-08:
I have not seen the 2 patches in the mailinglists of upstream rpi or dmaengine - please post them there!

clivem, 2016-06-08:
@msperl I don't understand. (It seems I never do!) You are against downstream accepting code, until it has been accepted upstream? I am assuming @HiassofT will submit his patches upstream, but I don't see why that should stop us moving forward downstream.

pelwell, 2016-06-08:
I'm OK with it in principle, it's just a question of getting enough test coverage.

Just a thought: if you created a PR against 4.6 it could get into a LibreElec build with a slightly lower bar.  What do you think, @popcornmix?

pelwell, 2016-06-08:
Correction - LE are on 4.4 at the moment, but the idea of extra testing still stands.

popcornmix, 2016-06-08:
Milhouse nightly builds are on 4.6 and they have a couple of hundred testers, a number using I2S audio, so that is a possibility.

I'm also okay with this in principle. If @clivem makes a PR and @msperl and @HiassofT say it's good then merging it shouldn't be a problem. We'd like to know that upstream is getting the same commits or there will be grief down the line, but that doesn't need to block this which includes a real bug fix. If commits get tweaked during the upstream process then it would be good to keep downstream in line with the changes.

clivem, 2016-06-08:
@pellwell For the record, this code has been "in use" by me since the 6th January. ie. for the last six months, patched into every rpi-4.{1,4} kernel I have built since then. 
NB. It has already solved one issue which wouldn't have needed solving had the code made it into downstream git repo earlier.... the 16 bit channel swap which was never reproducible with the new dma code. ;)

msperl, 2016-06-08:
@clivem : I just was asking to send the patches to upstream immediately - not necessarily waiting for the merge to happen upstream, but maybe at least waiting for some feedback from some others to see if there is any concern. The biggest concern - as far as I could guess - would have been 2 patches in a single patch, but that we have fixed

HiassofT, 2016-06-09:
@msperl I just sent out the patches to upstream lists/maintainers. I'll keep you posted about the status.

clivem, 2016-06-09:
@pelwell @popcornmix @msperl @HiassofT 
Guys, I have applied the 18 patch series to rpi-4.6.y.... if you want to get this tested as part of a Millhouse build.
https://github.com/DigitalDreamtimeLtd/linux/commits/rpi-4.6.y

@pelwell Do you want me to create a pull request for this?

@pelwell @popcornmix  In the interests of disclosure, and so there aren't any misunderstandings of my motives... My primary interest is with getting this into rpi-4.4.y, before any other tree. I will not be moving my current 17x machine (Zeros, CM, Bs, B+s, 2Bs, 3Bs) I2S test farm to a 4.6 based kernel or running a 4.6 kernel on any of the 4 setups I alternate between for DAC board/HAT listening purposes during the working day. I can only vouch for my testing of this patch series on top of rpi-4.4.y.

pelwell, 2016-06-09:
Yes - give us a PR. @popcornmix willing, we can get it into Milhouse's build tonight and then merge to 4.4 subsequently, perhaps tomorrow.

gtrainavicius, 2016-06-10:
Great work guys, let me know if there's any build you would like me to test!

popcornmix, 2016-06-13:
@gtrainavicius Please run rpi-update and close if issue is resolved.

gtrainavicius, 2016-06-15:
Sure, I will give an update on this later today!

gtrainavicius, 2016-06-15:
Just tried the rpi-update, and the issue is fixed. Thank you!

closed 2016-06-15 by gtrainavicius