raspberrypi/linux #1518: Raspbian 2016-05-27 kernel crash

jobermayr, 2016-06-05:
On fresh Raspbian 2016-05-27 image Kernel crashes. The system is not usable.

[image]

jobermayr, 2016-06-05:
Forgot to mention it happens on a RASPBERRY **PI 3** MODEL B.

pelwell, 2016-06-05:
1) How did you do the installation?

2) Have you performed any customisation?

3) Which images, if any, have you used before?

4) Which SD card are you using? Have you used it before?

5) How are you powering the Pi3? You should use a 2.5A supply.

jobermayr, 2016-06-05:
1)
`$ wget https://downloads.raspberrypi.org/raspbian/images/raspbian-2016-05-31/2016-05-27-raspbian-jessie.zip`
`$ unzip 2016-05-27-raspbian-jessie.zip`
`# dd bs=4M if=2016-05-27-raspbian-jessie.img of=/dev/sde`

2)
No

3)
NOOBS 1.9.1; had post-install issues while `apt-get upgrade`

4)
KINGSTON SDCA10/UHS-I 10, 16 GB (http://www.kingston.com/en/flash/microsd_cards/sdca10)
Yes, see 3)

5)
5A supply

jobermayr, 2016-06-06:
3) https://downloads.raspberrypi.org/raspbian/images/raspbian-2016-05-13/2016-05-10-raspbian-jessie.zip works

So the error is anywhere between 2016-05-10 and 2016-05-27

pelwell, 2016-06-06:
We've had some trouble with one new variety of Kingston cards, but the most recent Raspbian release includes a "quirk" to work-around the problem. If you are prepared to investigate a bit - I know it's time consuming - could you try the following (you can stop when you have success):

1) Using the 2016-05-10 image, run this and post the output:
```
grep . /sys/class/mmc_host/mmc0/mmc0:*/*
```
2) Flash the 2016-05-27 image, then before booting it edit cmdline.txt and add `mmc_block.card_quirks=0x80000000`.

3) Flash the 2016-05-27 image, but this time add `mmc_block.card_quirks=0`.

Then report back your findings.

stanke-sk, 2016-06-23:
Hello pelwell. I had the exact same problem . I have the 16GB kingston and the latest jessie now. I was not able even to perform apt-get upgrade it failed after few second and the fs was read only. no errors. so I set up the remote syslog and the issue were 
May 27 12:05:45 rpi kernel: [  940.921956] error 0 requesting status 0x4080900
May 27 12:05:45 rpi kernel: [  940.982802] mmc0: tried to reset card
May 27 12:05:45 rpi kernel: [  940.983056] error 0 requesting status 0x80900
May 27 12:05:45 rpi kernel: [  940.983065] blk_update_request: I/O error, dev mmcblk0, sector 7081384
May 27 12:05:45 rpi kernel: [  940.984567] mmcblk0: mmc_blk_err_check: general error sending stop command, stop cmd response 0x8                                               0d00
May 27 12:05:45 rpi kernel: [  940.984597] mmcblk0: retrying write for general error
May 27 12:05:45 rpi kernel: [  940.986026] mmcblk0: mmc_blk_err_check: general error sending stop command, stop cmd response 0x8             

hovewer looks like after adding mmc_block.card_quirks=0x80000000 to cmdline.txt the issue is fixed. So Thank you very much.

pelwell, 2016-06-23:
Can you run this command and post the output?:
```
grep . /sys/class/mmc_host/mmc0/mmc0:*/*  2>/dev/null
```
I would like to add your card to the list of those that require the quirk to be set.

stanke-sk, 2016-06-23:
Sure 

grep . /sys/class/mmc_host/mmc0/mmc0:*/*  2>/dev/null
/sys/class/mmc_host/mmc0/mmc0:0007/cid:4134325344313647300023412901026f
/sys/class/mmc_host/mmc0/mmc0:0007/csd:400e00325b59000073ff7f800a4000bb
/sys/class/mmc_host/mmc0/mmc0:0007/date:02/2016
/sys/class/mmc_host/mmc0/mmc0:0007/erase_size:512
/sys/class/mmc_host/mmc0/mmc0:0007/fwrev:0x0
/sys/class/mmc_host/mmc0/mmc0:0007/hwrev:0x3
/sys/class/mmc_host/mmc0/mmc0:0007/manfid:0x000041
/sys/class/mmc_host/mmc0/mmc0:0007/name:SD16G
/sys/class/mmc_host/mmc0/mmc0:0007/oemid:0x3432
/sys/class/mmc_host/mmc0/mmc0:0007/preferred_erase_size:4194304
/sys/class/mmc_host/mmc0/mmc0:0007/scr:0235800201000000
/sys/class/mmc_host/mmc0/mmc0:0007/serial:0x00234129
/sys/class/mmc_host/mmc0/mmc0:0007/type:SD
/sys/class/mmc_host/mmc0/mmc0:0007/uevent:DRIVER=mmcblk
/sys/class/mmc_host/mmc0/mmc0:0007/uevent:MMC_TYPE=SD
/sys/class/mmc_host/mmc0/mmc0:0007/uevent:MMC_NAME=SD16G
/sys/class/mmc_host/mmc0/mmc0:0007/uevent:MODALIAS=mmc:block

pelwell, 2016-06-23:
Thanks - that shows that the quirk wasn't being applied correctly - it will be fixed in the next release. Once you've updated you should no longer need the mmc_block.card_quirks parameter.

popcornmix, 2016-06-29:
Latest rpi-update firmware should apply the quirks correctly.
@jobermayr @stanke-sk Is this still an issue?

Ko-, 2016-07-17:
Hi. Exactly the same thing happens to me on a Raspberry Pi 2 model B. Yesterday, I first installed Arch from http://os.archlinuxarm.org/os/ArchLinuxARM-rpi-2-latest.tar.gz and then upgraded the kernel and firmware to the latest versions. syslog reports the same error messages as with stanke-sk. After setting card_quirks in cmdline.txt, the issue is resolved. Here's my card info:
```
/sys/class/mmc_host/mmc0/mmc0:0007/cid:275048534431364730dad55e61010101
/sys/class/mmc_host/mmc0/mmc0:0007/csd:400e00325b59000073a77f800a400001
/sys/class/mmc_host/mmc0/mmc0:0007/date:01/2016
/sys/class/mmc_host/mmc0/mmc0:0007/erase_size:512
/sys/class/mmc_host/mmc0/mmc0:0007/fwrev:0x0
/sys/class/mmc_host/mmc0/mmc0:0007/hwrev:0x3
/sys/class/mmc_host/mmc0/mmc0:0007/manfid:0x000027
/sys/class/mmc_host/mmc0/mmc0:0007/name:SD16G
/sys/class/mmc_host/mmc0/mmc0:0007/oemid:0x5048
/sys/class/mmc_host/mmc0/mmc0:0007/preferred_erase_size:4194304
/sys/class/mmc_host/mmc0/mmc0:0007/scr:0235800201000000
/sys/class/mmc_host/mmc0/mmc0:0007/serial:0xdad55e61
/sys/class/mmc_host/mmc0/mmc0:0007/type:SD
/sys/class/mmc_host/mmc0/mmc0:0007/uevent:DRIVER=mmcblk
/sys/class/mmc_host/mmc0/mmc0:0007/uevent:MMC_TYPE=SD
/sys/class/mmc_host/mmc0/mmc0:0007/uevent:MMC_NAME=SD16G
/sys/class/mmc_host/mmc0/mmc0:0007/uevent:MODALIAS=mmc:block
```

Ruffio, 2016-07-28:
@jobermayr do you still experience this issue, else please close this issue. Thanks.

peteheist, 2016-07-31:
I'm seeing the same thing, but neither appending mmc_block.card_quirks=0x80000000 nor mmc_block.card_quirks=0 to cmdline.txt fixes it for me. Here's the output for my card:

pi@raspberrypi:~ $ grep . /sys/class/mmc_host/mmc0/mmc0:*/*  2>/dev/null
/sys/class/mmc_host/mmc0/mmc0:0007/cid:4134325344313647300029214a010497
/sys/class/mmc_host/mmc0/mmc0:0007/csd:400e00325b59000073ff7f800a4000bb
/sys/class/mmc_host/mmc0/mmc0:0007/date:04/2016
/sys/class/mmc_host/mmc0/mmc0:0007/erase_size:512
/sys/class/mmc_host/mmc0/mmc0:0007/fwrev:0x0
/sys/class/mmc_host/mmc0/mmc0:0007/hwrev:0x3
/sys/class/mmc_host/mmc0/mmc0:0007/manfid:0x000041
/sys/class/mmc_host/mmc0/mmc0:0007/name:SD16G
/sys/class/mmc_host/mmc0/mmc0:0007/oemid:0x3432
/sys/class/mmc_host/mmc0/mmc0:0007/preferred_erase_size:4194304
/sys/class/mmc_host/mmc0/mmc0:0007/scr:0235800201000000
/sys/class/mmc_host/mmc0/mmc0:0007/serial:0x0029214a
/sys/class/mmc_host/mmc0/mmc0:0007/type:SD
/sys/class/mmc_host/mmc0/mmc0:0007/uevent:DRIVER=mmcblk
/sys/class/mmc_host/mmc0/mmc0:0007/uevent:MMC_TYPE=SD
/sys/class/mmc_host/mmc0/mmc0:0007/uevent:MMC_NAME=SD16G
/sys/class/mmc_host/mmc0/mmc0:0007/uevent:MODALIAS=mmc:block

marc40000, 2016-08-18:
I just got 5 rpi 3 b models and try the following 3 times:

- get jessie lite image from 2016-05-10 
- sudo apt-get update
- sudo apt-get upgrade

and there I get some errors about it requiring some older packages as those that are already installed. Afterwards, if I reboot it just gets a kernel panic during boot.

I'm not sure if it's the same issue but in one of the earlier posts jobemayr wrote "had post-install issues while apt-get upgrade"

I have 2.5 A power supplies, intenso class 10 8 gb micro sd cards. I also have a rpi 1 b+ model running the same image without issues.

mihalop, 2016-11-22:
Having the similar issues here. Is there a downside/benefit for setting mmc_block.card_quirks=0x80000000, if not much writing is expected?

pelwell, 2016-11-22:
It will disable the SD ERASE command, which could theoretically reduce the write performance, but I doubt you'll notice anything.

Can you run `grep . /sys/class/mmc_host/mmc0/mmc0:/ 2>/dev/null` and post the output?

mihalop, 2016-11-22:
We have 7-8 devices remotely and after a software update, the corruption surfaced and we lost access. Will do as soon as they get recovered. We are using an older firmware in any case, so you might already got it in the list.

With disabling the ERASE command significantly affect the card's lifespan?

pelwell, 2016-11-22:
The /sys/class output identifies the card, not the firmware. If that type of card does show corruption problems that forcing card_quirks fixes then we'd probably enable that quirk for all cards of that type.

I cannot state with any authority that not using ERASE will not negatively affect the lifetime of the card, but I've not seen any suggestion that ERASE is anything other than a performance enhancement. A card has to erase a sector before it can be written to, and erasing takes some time, so if you can erase as part of a delete you may speed up a subsequent write to those sectors.

Many cards don't support the ERASE command, so I don't think you need to worry about lifetime, but you would have to ask the card manufacturer if you want a definitive answer. [This thread](https://github.com/raspberrypi/linux/issues/431) gives a bit more information that may be reassuring.

Ruffio, 2017-02-05:
@jobermayr has this issue been resolved?

closed 2017-05-18 by JamesH65