raspberrypi/linux #1526: KMS/DRM: Are we losing hardware-accelerated blitting?

vanfanel, 2016-06-11:
Hi,

I have been porting published and popular programs (RetroArch, Scummvm, and a long etc) to use the dispmanx API for years now.
However, dispmanx should, sooner or later, be superseeded by KMS/DRM, so I am porting RetroArch to it.

With dispmanx, things were easy for blitting. Let's say I want to copy a 256x256 rect for a 298x256 buffer to a dismanx "resource" (buffer). Well, I have this GREAT dispmanx function for that:

`
vc_dispmanx_resource_write_data( DISPMANX_RESOURCE_HANDLE_T res, VC_IMAGE_TYPE_T src_type, int src_pitch, void * src_address, const VC_RECT_T * rect );`

You see, I can pass a pitch, let's say 256*4 if I am blitting a pixel array with 4 bytes per pixel, and then it will be uploaded to the GPU without using the CPU for the transfer. Very fast and good solution!

But in KMS, using a dumb buffer, without an specific function to do it, I would have to copy a pitch of pixels each line, so in 256 lines I would be calling memcpy() 256 times to archieve the same. And that's for small rect...

So, I have seen that /usr/include/drm contains some hardware-specific implementations of blitting functions. Is there something similar to the Pi? I don't mind using IOCTLs or whatever is needed...

I feel porting RetroArch to plain KMS, now that the Pi is following the standard, is very important, and I already have it running except for the slow blitting, so if it's not the right place to ask, please point me to the right place.

popcornmix, 2016-06-11:
I'd suggest posting here: https://github.com/anholt/linux
Eric owns the vc4_kms_drm driver and is more likely to know what the options are.

closed 2016-06-11 by popcornmix

vanfanel, 2016-06-11:
Thanks about the tip, popcornmix.