raspberrypi/linux #1527: Wrong Y axis for HID: dragonrise

sec, 2016-06-12:
Hi,
I think that Y axis for this USB gamepad is wrong. Currently it reports Y values on axis 3.
According to current driver source, axis are in order X, Ry, Z, Y, Ry(?).
Shouldn't this be X,Y,Z,Ry, Rx(y) ?
This makes unable to play games/emulators which use new evdev to access the joystick (which don't provide option to rebind axis) - under old joydev this can be rebinded, but it's also in wrong order IMO.

Simply swaping:
0x09, 0x33,        /*     Usage (Ry)			*/
with
0x09, 0x31,        /*     Usage (Y)			*/
fixes the issue - axis Y is reported on axis 1.

P33M, 2016-06-13:
This is an upstream driver - which means other distributions will be affected.

Post the bug report to the upstream HID device mailing list:
linux-input@vger.kernel.org

An FAQ on submitting bug reports to the mailing list can be found here:
www.tux.org/lkml/#s2

If an upstream patch is accepted, we can backport it if necessary.

closed 2016-06-13 by P33M

netpipe, 2021-01-30:
in jstest-gtk it has analog values for x axis from 0 to 255 but y axis behaves like dpad.. i think it should be on axis 4
[image]
[image]

the difference between half way there

netpipe, 2021-01-30:
https://github.com/twinone/n64-arduino