raspberrypi/linux #1529: Unable to start RT5370 in SoftAP mode in 4.4.10

mevatron, 2016-06-14:
Today I attempted to upgrade from 4.1.12 to 4.4.10, and connman 1.21 can no longer enable SoftAP mode on my Ralink RT5370 base USB WiFi dongle.

I'm currently using Raspberry Pi 2 model B for this testing. If there is further information I can provide, I would be happy to do so.

I've perused the changes involving nl80211, mac80211, and rt2800usb modules, but I didn't see anything that popped out at me that could be causing the issue.

Any help would be greatly appreciated!

LoyalNine1487, 2016-06-19:
What does it say in the console when you try to turn it on, and dose it work normally when you just try to connect to a AP not try to host a AP?

mevatron, 2016-06-20:
@LoyalNine1487 Thanks for the reply. The client mode does appear to work just fine. I did confirm my suspicion about this being a kernel problem. I'm using [Hypriot](blog.hypriot.com) v0.8.0, which is build using `4.4.10`, so I rebuilt the same v0.8.0 image using `4.1.20` and everything started working again. I may need to bump this issue up to the mainline guys to see what I can find out.

LoyalNine1487, 2016-06-24:
The only thing I can think of is that the driver to make APs is not in the new kernel, have you tried to compile the kernel?

mevatron, 2016-06-25:
Would that be a special driver setting? As it does detect the RT5370 chip
correctly, and loads the rt2800usb driver when attached.

Maybe I'll also try base Raspbian to see if that works as well.

Will

LoyalNine1487, 2016-06-27:
Do sudo iw list and repley what the output is.

Ruffio, 2016-08-16:
@mevatron have you tried @LoyalNine1487 suggestion? What did it output?

JamesH65, 2017-05-18:
Closing due to lack of activity. Reopen if you feel this issue is still relevant, and updating to the latest kernel (4.9) doesn't fix the issue.

closed 2017-05-18 by JamesH65