raspberrypi/linux #1532: Programatically change filtering method for dispmanx scaling

vanfanel, 2016-06-15:
Hi there,

We have the vcgencmd command to change the scaling method applied by the GPU on scaling, so something like this command will change it to nearest-neighbour scaling :

> vcgencmd scaling_kernel 0 0 0 0 0 0 0 0 1 1 1 1 255 255 255 255 255 255 255 255 1 1 1 1 0 0 0 0 0 0 0 0   1

What vc_* or similar functions or even ioctls do we have to do the same thing from a C program?
If this command does it, then it must be possible to do it manually from code.

Thanks

popcornmix, 2016-06-15:
Header files here: https://github.com/raspberrypi/userland/blob/master/interface/vmcs_host/vcgencmd.h
Example code: https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/gencmd/gencmd.c

vanfanel, 2016-06-16:
@popcornmix : I should have looked for those in the first place, sorry. I didn't know we had the vc commands sourcecode available!
Mistery solved and solution applied, now RetroArch can at least honor the "HW Bilinear Filtering" setting when using my dispmanx driver! Thanks, as always!

closed 2016-06-16 by vanfanel