raspberrypi/linux #1534: Warning with backtrace after DMA changes in rpi-4.6.y tree

asavah, 2016-06-15:
Kernel cross-compiled from branch rpi-4.6.y showa the following warning with backtrace in dmesg:

```
[    1.982142] ------------[ cut here ]------------
[    1.982168] WARNING: CPU: 0 PID: 87 at kernel/workqueue.c:2443 check_flush_dependency+0x164/0x168
[    1.982186] workqueue: PF_MEMALLOC task 87(mmcqd/0) is flushing !WQ_MEM_RECLAIM events:lru_add_drain_per_cpu
[    1.982196] Modules linked in:
 
[    1.982217] CPU: 0 PID: 87 Comm: mmcqd/0 Not tainted 4.6.2-v7+ #1
[    1.982225] Hardware name: BCM2709
[    1.982257] [<80110a54>] (unwind_backtrace) from [<8010c3c4>] (show_stack+0x20/0x24)
[    1.982277] [<8010c3c4>] (show_stack) from [<80430648>] (dump_stack+0xd4/0x118)
[    1.982296] [<80430648>] (dump_stack) from [<8011dcd4>] (__warn+0xf8/0x110)
[    1.982313] [<8011dcd4>] (__warn) from [<8011dd34>] (warn_slowpath_fmt+0x48/0x50)
[    1.982332] [<8011dd34>] (warn_slowpath_fmt) from [<801331e4>] (check_flush_dependency+0x164/0x168)
[    1.982355] [<801331e4>] (check_flush_dependency) from [<8013495c>] (flush_work+0x6c/0x174)
[    1.982375] [<8013495c>] (flush_work) from [<8020d884>] (lru_add_drain_all+0x150/0x190)
[    1.982396] [<8020d884>] (lru_add_drain_all) from [<8024e8dc>] (migrate_prep+0x18/0x20)
[    1.982416] [<8024e8dc>] (migrate_prep) from [<80206160>] (alloc_contig_range+0xe8/0x318)
[    1.982435] [<80206160>] (alloc_contig_range) from [<80258b54>] (cma_alloc+0xc4/0x25c)
[    1.982456] [<80258b54>] (cma_alloc) from [<804c11e4>] (dma_alloc_from_contiguous+0x48/0x4c)
[    1.982480] [<804c11e4>] (dma_alloc_from_contiguous) from [<801158bc>] (__alloc_from_contiguous+0x3c/0xa8)
[    1.982497] [<801158bc>] (__alloc_from_contiguous) from [<80115964>] (cma_allocator_alloc+0x3c/0x44)
[    1.982513] [<80115964>] (cma_allocator_alloc) from [<80115cac>] (__dma_alloc+0x12c/0x2b4)
[    1.982529] [<80115cac>] (__dma_alloc) from [<80115ee0>] (arm_dma_alloc+0x64/0x6c)
[    1.982549] [<80115ee0>] (arm_dma_alloc) from [<80246700>] (dma_pool_alloc+0x110/0x234)
[    1.982571] [<80246700>] (dma_pool_alloc) from [<8047d400>] (bcm2835_dma_create_cb_chain+0x140/0x22c)
[    1.982594] [<8047d400>] (bcm2835_dma_create_cb_chain) from [<8047dd70>] (bcm2835_dma_prep_slave_sg+0x128/0x2a8)
[    1.982616] [<8047dd70>] (bcm2835_dma_prep_slave_sg) from [<8059b668>] (bcm2835_sdhost_request+0x410/0x604)
[    1.982641] [<8059b668>] (bcm2835_sdhost_request) from [<805805b8>] (__mmc_start_request+0x50/0xe0)
[    1.982663] [<805805b8>] (__mmc_start_request) from [<80580748>] (mmc_start_request+0x100/0x130)
[    1.982683] [<80580748>] (mmc_start_request) from [<805814f8>] (mmc_start_req+0x2b8/0x3dc)
[    1.982703] [<805814f8>] (mmc_start_req) from [<80590d94>] (mmc_blk_issue_rw_rq+0xd0/0xb4c)
[    1.982719] [<80590d94>] (mmc_blk_issue_rw_rq) from [<8059191c>] (mmc_blk_issue_rq+0x10c/0x4e0)
[    1.982736] [<8059191c>] (mmc_blk_issue_rq) from [<805934c4>] (mmc_queue_thread+0xd4/0x1a4)
[    1.982754] [<805934c4>] (mmc_queue_thread) from [<8013bf8c>] (kthread+0xec/0x104)
[    1.982776] [<8013bf8c>] (kthread) from [<80107f48>] (ret_from_fork+0x14/0x2c)
[    1.982786] ---[ end trace 273cd310faa7dacd ]---
```

The board is rpi3.
Power supply is fine, no colored squares ever.

To make this as clean as possible kernel was built with
`make ARCH=arm CROSS_COMPILE=blah bcm2709_defconfig`

I'm positive this started after all the DMA improvements were merged.
It's just a warning and everything runs just fine after backtrace.

config.txt;
```
start_file=start_x.elf
fixup_file=fixup_x.elf
kernel=kernel7.img
framebuffer_depth=32
framebuffer_ignore_alpha=1
disable_overscan=1
disable_splash=1
gpu_mem=320
dtoverlay=lirc-rpi
dtparam=watchdog=on
dtparam=random=on
dtparam=audio=on
decode_MPG2=0xSNIP
decode_WVC1=0xSNIP
```

cmdline.txt:
```
smsc95xx.turbo_mode=N console=tty1 root=/dev/mmcblk0p2 rootfstype=f2fs elevator=noop rootwait quiet
```

rpi firmware version:
```
Jun 13 2016 13:25:19
Copyright (c) 2012 Broadcom
version 625ddc17506de68faffed80e656109b206905fc0 (clean) (release)
```

The sdcard is Kingston SDC10G2/16GBSP.

Full dmesg:
http://pastebin.com/ACQLvVhd

popcornmix, 2016-06-15:
Yes, we've seen this with 4.7 kernel - I hadn't noticed with 4.6 but I guess the upstream DMA commits may have provoked this.
It didn't seem to cause a problem for me - sdcard still worked. Any issue for you?

@msperl @clivem any thoughts?

clivem, 2016-06-15:
Slowpath warning...... Martin might have thoughts...... I've never seen it on 4.1, through 4.4 kernels. I haven't personally tested anything on 4.6 or 4.7. I've got reasons I need to hand back on 4.4/5 until after the release of Fedora 24, but that's getting seriously off-topic. ;)

asavah, 2016-06-15:
@popcornmix 
>It didn't seem to cause a problem for me - sdcard still worked. Any issue for you?

Nope, everything seems to work fine.
However the pi is used for Kodi only, media is on nas via nfs fstab mount, Kodi dbs are on the nas mysql server too.
Almost no sdcard usage.

The interesting thing is that dd shows that read is slower than write:
```
dd if=/dev/zero of=test.img bs=1M count=1000 oflag=direct
1000+0 records in
1000+0 records out
1048576000 bytes (1.0 GB, 1000 MiB) copied, 56.2505 s, 18.6 MB/s

 dd if=test.img of=/dev/null bs=1M iflag=direct
1000+0 records in
1000+0 records out
1048576000 bytes (1.0 GB, 1000 MiB) copied, 75.9063 s, 13.8 MB/s
```

lategoodbye, 2016-06-19:
@asavah 

Just a guess, could you test this patch

https://git.kernel.org/cgit/linux/kernel/git/next/linux-next.git/commit/mm/swap.c?id=f3a932baa7f65072434f1c04c02c8a4d2746fcfc

pelwell, 2016-06-19:
Thanks, @lategoodbye - that removes the warning. I've pulled it into rpi-4.6.y.

asavah, 2016-06-19:
@lategoodbye @pelwell Thanks!
dmesg is nice and clean now.

closed 2016-06-19 by asavah

pritambaral, 2017-03-14:
Ths exact issue has reappeared in rpi-4.11.y

pelwell, 2017-03-14:
The 4.11 kernel hasn't even been released yet, and rpi-4.11.y is still very much a work-in-progress. Hold off on your bug reports for a while, and create new issues when you do (but you can link to closed ones).