raspberrypi/linux #1536: Raspberry Pi Compute Module: sdhost on 22-27

benpye, 2016-06-16:
I am attempting to configure the compute module to expose the sdhost interface interface on GPIO22-27 on the compute module. My currently device tree overlay is given below though this hangs the boot, as it seems neither the internal eMMC or the external SD card I have attatched work after applying the overlay, with `dtoverlay=sd1,overclock_50=5`

The overlay is
```
/dts-v1/;
/plugin/;

/{
    compatible = "brcm,bcm2708";

    fragment@0 {
        target = <&sdhost>;
        frag0: __overlay__ {
            pinctrl-names = "default";
            pinctrl-0 = <&sdhost_pins>;
            non-removable;
            bus-width = <4>;
            bcrm,overclock-50 = <0>;
            bcrm,pio-limit = <1>;
            bcrm,debug-flags = <0>;
            status = "okay";
        };
    };

    fragment@1 {
        target = <&gpio>;
        __overlay__ {
            sdhost_pins: sdhost_pins {
                brcm,pins = <22 23 24 25 26 27>;
                brcm,function = <7 7 7 7 7 7>;    /* alt3: SD1*/
                brcm,pull = <0 2 2 2 2 2>;    /* pull up all except clk */
            };
        };
    };

    __overrides__ {
        overclock_50     = <&frag0>,"brcm,overclock-50:0";
        force_pio        = <&frag0>,"brcm,force-pio?";
        pio_limit        = <&frag0>,"brcm,pio-limit:0";
        debug            = <&frag0>,"brcm,debug?";
        poll_once        = <&frag0>,"non-removable?";
        bus_width        = <&frag0>,"bus-width:0";
    };
};
```

cc: @pelwell

pelwell, 2016-06-16:
1) You probably don't want `non-removable` on an interface with removable cards. I would remove both the property and the parameter that targets it (`poll_once`).

2) It's less confusing if you spell "bcrm" correctly ("brcm"), although the end result should be the same since your provided values are the defaults. N.B. This is only true with recent firmware - older firmware will not create or extend properties that were targets of integer/cell parameters (e.g. brcm,overclock-50 for overclock_50).

3) Neither `bcrm,debug-flags` nor `brcm,debug-flags` is a valid property. The `debug` parameter targets a boolean property (present = true, absent = false, value ignored) called `brcm,debug`, and since the default is off you shouldn't create the property.

4) The alt3 function on pins 22-27 is SD1 a.k.a. MMC. You want SD0 a.k.a. SDHOST which is alt0 (value = 4).

5) Attempting to underclock an interface by setting the `brcm,overclock-50` property to 5 won't work. Only values greater than 50MHz are allowed.

Note that parameters/overrides only do something if they are applied. In the example above, the only way to remove the non-removable property would be using an explicit `poll_once=off`.

benpye, 2016-06-16:
Ah yes, I realise that is a typo now, sorry, I was basing this somewhat on https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=88781 and there it is brcm, but I didn't realise I ended up with bcrm.

Is there anywhere that 22-27 is documented with ALT0 = sdhost? The peripheral guide gives it as reserved. I do recall finding some GitHub repo with complete docs for every alt function, but I cannot find it again.

With regard to `overclock_50`, is there no way to lower the clock speed for an SD card on the sdhost interface? As currently the SD card is attached with flying wires I found I had to lower the clock with the mmc interface to get a reliable interface, obviously this can be improved on a final design but for debugging it would be very helpful. 

```
/dts-v1/;
/plugin/;

/{
    compatible = "brcm,bcm2708";

    fragment@0 {
        target = <&sdhost>;
        frag0: __overlay__ {
            pinctrl-names = "default";
            pinctrl-0 = <&sdhost_pins>;
            bus-width = <4>;
            brcm,overclock-50 = <0>;
            brcm,pio-limit = <1>;
            status = "okay";
        };
    };

    fragment@1 {
        target = <&gpio>;
        __overlay__ {
            sdhost_pins: sdhost_pins {
                brcm,pins = <22 23 24 25 26 27>;
                brcm,function = <4 4 4 4 4 4>;    /* alt0: SD0*/
                brcm,pull = <0 2 2 2 2 2>;    /* pull up all except clk */
            };
        };
    };

    __overrides__ {
        overclock_50     = <&frag0>,"brcm,overclock-50:0";
        force_pio        = <&frag0>,"brcm,force-pio?";
        pio_limit        = <&frag0>,"brcm,pio-limit:0";
        debug            = <&frag0>,"brcm,debug?";
        poll_once        = <&frag0>,"non-removable?";
        bus_width        = <&frag0>,"bus-width:0";
    };
};
```

The above seems to get further, no longer breaks eMMC but I am getting mmc0 errors, so the SD card doesn't show, as I say this may be due to bad signal integrity, I can try adding some series resistors but I don't really have any way to shorten the wires, I do have a decoupling capacitor on the SD card end already though. The log just repeats

```
[ 1294.427721] mmc0:>cmd op 55 arg 0x12340000 flags 0x95 - resp 00000000 00000000 00000000 00000000, err 0
[ 1294.427737] mmc0: =========== REGISTER DUMP ===========
[ 1294.427751] mmc0: SDCMD  0x00004037
[ 1294.427763] mmc0: SDARG  0x12340000
[ 1294.427775] mmc0: SDTOUT 0x017d7840
[ 1294.427786] mmc0: SDCDIV 0x00000003
[ 1294.427798] mmc0: SDRSP0 0x00000920
[ 1294.427810] mmc0: SDRSP1 0x00003f1d
[ 1294.427821] mmc0: SDRSP2 0x7fffffff
[ 1294.427833] mmc0: SDRSP3 0x00024037
[ 1294.427844] mmc0: SDHSTS 0x00000010
[ 1294.427856] mmc0: SDVDD  0x00000001
[ 1294.427867] mmc0: SDEDM  0x00010801
[ 1294.427878] mmc0: SDHCFG 0x0000040a
[ 1294.427890] mmc0: SDHBCT 0x00000040
[ 1294.427902] mmc0: SDHBLC 0x00000001
[ 1294.427913] mmc0: ===========================================
[ 1294.438697] mmc0: unexpected command 55 error
[ 1294.438745] mmc0:>cmd op 55 arg 0x12340000 flags 0x95 - resp 00000000 00000000 00000000 00000000, err 0
[ 1294.438760] mmc0: =========== REGISTER DUMP ===========
[ 1294.438773] mmc0: SDCMD  0x00000037
[ 1294.438786] mmc0: SDARG  0x12340000
[ 1294.438798] mmc0: SDTOUT 0x017d7840
[ 1294.438809] mmc0: SDCDIV 0x00000003
[ 1294.438821] mmc0: SDRSP0 0x00001240
[ 1294.438833] mmc0: SDRSP1 0x00007e3b
[ 1294.438844] mmc0: SDRSP2 0xffffffff
[ 1294.438855] mmc0: SDRSP3 0x0002480e
[ 1294.438866] mmc0: SDHSTS 0x00000000
[ 1294.438878] mmc0: SDVDD  0x00000001
[ 1294.438889] mmc0: SDEDM  0x00010801
[ 1294.438901] mmc0: SDHCFG 0x0000040a
[ 1294.438912] mmc0: SDHBCT 0x00000040
[ 1294.438923] mmc0: SDHBLC 0x00000001
[ 1294.438934] mmc0: ===========================================
[ 1294.439027] mmc0: command never completed.
[ 1294.439058] mmc0:>cmd op 55 arg 0x12340000 flags 0x95 - resp 00000000 00000000 00000000 00000000, err 0
[ 1294.439071] mmc0: =========== REGISTER DUMP ===========
[ 1294.439085] mmc0: SDCMD  0x00004037
[ 1294.439097] mmc0: SDARG  0x12340000
[ 1294.439108] mmc0: SDTOUT 0x017d7840
[ 1294.439119] mmc0: SDCDIV 0x00000003
[ 1294.439131] mmc0: SDRSP0 0x00001240
[ 1294.439142] mmc0: SDRSP1 0x00007f3b
[ 1294.439153] mmc0: SDRSP2 0xffffffff
[ 1294.439164] mmc0: SDRSP3 0x0004901d
[ 1294.439175] mmc0: SDHSTS 0x00000010
[ 1294.439186] mmc0: SDVDD  0x00000001
[ 1294.439198] mmc0: SDEDM  0x00010801
[ 1294.439209] mmc0: SDHCFG 0x0000040a
[ 1294.439220] mmc0: SDHBCT 0x00000040
[ 1294.439232] mmc0: SDHBLC 0x00000001
[ 1294.439242] mmc0: ===========================================
[ 1294.439306] mmc0: command never completed.
[ 1294.439335] mmc0:>cmd op 55 arg 0x12340000 flags 0x95 - resp 00000000 00000000 00000000 00000000, err 0
[ 1294.439349] mmc0: =========== REGISTER DUMP ===========
[ 1294.439361] mmc0: SDCMD  0x00004037
[ 1294.439373] mmc0: SDARG  0x12340000
[ 1294.439385] mmc0: SDTOUT 0x017d7840
[ 1294.439396] mmc0: SDCDIV 0x00000003
[ 1294.439408] mmc0: SDRSP0 0x00001240
[ 1294.439419] mmc0: SDRSP1 0x00007f3b
[ 1294.439430] mmc0: SDRSP2 0xffffffff
[ 1294.439442] mmc0: SDRSP3 0x0004901d
[ 1294.439453] mmc0: SDHSTS 0x00000010
[ 1294.439464] mmc0: SDVDD  0x00000001
[ 1294.439475] mmc0: SDEDM  0x00010801
[ 1294.439486] mmc0: SDHCFG 0x0000040a
[ 1294.439498] mmc0: SDHBCT 0x00000040
[ 1294.439513] mmc0: SDHBLC 0x00000001
[ 1294.439525] mmc0: ===========================================
[ 1294.439595] mmc0: error -5 whilst initialising SD card
[ 1294.537779] mmc0: host does not support reading read-only switch, assuming write-enable
[ 1294.540543] mmc0: command never completed.
```

pelwell, 2016-06-16:
> Is there anywhere that 22-27 is documented with ALT0 = sdhost? The peripheral guide gives it as reserved. I do recall finding some GitHub repo with complete docs for every alt function, but I cannot find it again.

Apart from the DTS files and overlays you mean? Possibly not.

> With regard to overclock_50, is there no way to lower the clock speed for an SD card on the sdhost interface?

Try adding "max-frequency = <5000000>;" (completely untested) - that seems to be a standard property supported by the MMC framework.

benpye, 2016-06-16:
I have constructed a different SD card breakout for the compute module, and can now get the SD card running with the overlay given above. Thanks @pelwell for your assistance.

tybach, 2016-06-17:
Were you able to lower the clock frequency below 41.7MHz?

benpye, 2016-06-17:
I have actually managed to get the external interfacing working just fine
at the default frequency, I did try the `max-frequency` parameter
suggested, but this appeared to have no effect (didn't make the bad adapter
work anyway), though I have verified this as I lack any way to measure high
enough clock speeds unfortunately. Really need to buy a scope some day...

tybach, 2016-06-18:
I was able to lower the clock frequency using the sdio_overclock override in the sdio device tree overlay.  For some reason it does not work with the overclock_50 override.

ivomilev, 2016-06-18:
Hi All,

I was able to get working SD1 interface with above overlay ( thanks to **benpye** and **pelwell** ) but there is one major issue that I noticed:
After reboot, the compute module won't boot from eMMC if there is inserted SD Card in the SD1 interface. If I remove the card, the CM boots properly and then if I re-insert the card, the CM recognize it properly. Do you have similar issue and do you have an idea what could be the root issue?

Best Regards,
Ivo Milev

tybach, 2016-06-18:
Does it do the same thing if you use the built in overlay 'sdio'?  I'm not using an sdcard on the SD1 interface and it is booting just fine.

ivomilev, 2016-06-18:
It is even worst - it cannot boot from eMMC either with card inserted or not inserted in SD1

Best Regards,
Ivo Milev

pelwell, 2016-06-18:
My guess would be that the SD interface is coming up as mmc0 instead of eMMC, which must be relegated to mmc1. I'll have to look into what dictates the order, but it could be the order of nodes within the device tree.

ivomilev, 2016-06-18:
**Pelwell**, I was suspecting exactly the same :) I made a simple experiment by cloning the eMMC flash to the SD card and then inserted the card into SD1. And it boot successfully .... from the SD card. So now I need to figure out how to instruct the firmware to always boot from eMMC flash ( detect first )

Best Regards,
Ivo Milev

brentpicasso, 2016-07-01:
Interested in this thread. @ivomilev, were you able to detect eMMC first?

ivomilev, 2016-07-01:
No, but in my case I am using the SD card as non-removable device and I changed the root partition location in cmdline.txt and in fstab to boot from eMMC flash (mmcblk1). But if I remove the SD card, the system will hang because the eMMC will become mmcblk0 again...

xdsopl, 2016-07-05:
Sorry for coming late to the party ..
I am able to reliably boot from the on module flash (sdhost at p48-p53) und reliably use the extra controller (mmc at p22-p27) for some time now:
- You need this patch: https://github.com/raspberrypi/linux/pull/1492
- Add p22-p27 as function "emmc" to dt-blob.bin (just copy paste and adapt from the present "sdcard" lines)
- Adapt newly introduced lines in arch/arm/boot/dts/bcm2708-rpi-cm.dtsi regarding mmc and sdhost.

Ahmet

realizator, 2016-07-14:
Hi all!
Have success with boot from eMMC and access to SD inserted after boot - thanx @benpye  and @pelwell ! And I also have a problem with boot if SD is inserted (detected as mmc0).
I need to boot from eMMC with inserted SD, so I try to use dynamic device tree load after boot, but with no success.
What I did:
1. Comment dtoverlay=sd1,overclock_50=5 in config.txt
2. After boot from eMMC (SD is not inserted) i run:
`sudo dtoverlay sd1 overclock_50=5`
3. Check if it is loaded:
`dtoverlay -l`
and obtain this:
```
Overlays (in load order):
0:    sd1   overclock_50=5
```
that is overlay is loaded.
4. After I insert SD I can see mmcblk0 only with lsblk,and mmcblk1 not appears (as it was using config.txt)
Have anybody try to load device tree dynamically after boot?
Regards,
Eugene

benpye, 2016-07-14:
I am probably going to modify the kernel for my application and do
something like http://www.spinics.net/lists/linux-mmc/msg25690.html

realizator, 2016-07-14:
@benpye thank you for the link!
It's interesting approach. A problem with name ID assignation already mentioned in a lot of forums for a years (Arp 2014 in your link). Looks strange this was not yet implemented in a modern kernels.
I'm try to avoid kernel modification and also not touch system dt's like bcm2708-rpi-cm.dtsi (as proposed by @xdsopl ) to keep compatibility with future Raspbian updates.

closed 2017-05-18 by JamesH65