raspberrypi/linux #1540: Xbox 360 controller connectivity issues with xboxdrv after kernel update

HerbFargus, 2016-06-22:
I'm running RetroPie but to reproduce, if you connect two xbox 360 wireless controllers with the [xboxdrv](https://github.com/xboxdrv/xboxdrv) driver they will both connect to the receiver but only one will show up in jstest/receive input and it's intermittent. I apologise if this is the wrong place for this but it only broke after a kernel update. I'm currently on 
```
Linux retropie 4.4.13-v7+ #894 SMP Mon Jun 13 13:13:27 BST 2016 armv7l GNU/Linux
```
If I roll back to (https://github.com/Hexxeh/rpi-firmware/commit/771a503cfc2a1130e2df2a4ddfc45ffa0f592b3f) it works again
```
Linux retropie 4.1.21-v7+ #873 SMP Mon Apr 11 18:22:10 BST 2016 armv7l GNU/Linux
```

This is the output of xboxdrv:

http://pastebin.com/hbV2pTHd

more information here:

https://retropie.org.uk/forum/topic/1640/xbox360-wireless-controllers-receiver-connectivity-issues/15

Let me know if this issue needs to be logged somewhere else or if you need more information on the matter.

popcornmix, 2016-06-22:
The bump to 4.4.6 kernel is purely a change to upstream kernel code. As such it is quite possible that the same problem is happening on other platforms with 4.4 kernel. Finding referenced to bug reports or solutions would be useful.

Is the controller connected via USB? What is the output of the following:
```
lsusb
lsmod
```
on both the working and non-working kernel?

HerbFargus, 2016-06-23:
I have a wireless receiver plugged in that connects up to 4 wireless xbox controllers (brand is zettaguard) 

### Old Kernel (4.1.21-v7+):

**lsusb**
```
pi@retropie:~ $ lsusb
Bus 001 Device 004: ID 045e:0291 Microsoft Corp. Xbox 360 Wireless Receiver for Windows
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
**lsmod**
```
pi@retropie:~ $ lsmod
Module                  Size  Used by
bnep                   10336  2 
hci_uart               13533  1 
btbcm                   4196  1 hci_uart
bluetooth             317981  22 bnep,btbcm,hci_uart
brcmfmac              177635  0 
brcmutil                5501  1 brcmfmac
evdev                  10278  4 
joydev                  9024  0 
cfg80211              407532  1 brcmfmac
xpad                   12312  0 
rfkill                 16036  4 cfg80211,bluetooth
ff_memless              4646  1 xpad
snd_bcm2835            19802  0 
snd_pcm                73474  1 snd_bcm2835
bcm2835_rng             1763  0 
bcm2835_gpiomem         2860  0 
snd_timer              18848  1 snd_pcm
snd                    50779  3 snd_bcm2835,snd_timer,snd_pcm
uio_pdrv_genirq         2944  0 
uio                     7753  1 uio_pdrv_genirq
uinput                  7246  2 
fuse                   80630  1 
ipv6                  340825  38
```

### Latest Kernel (4.4.13-v7+): 

**lsusb**
```
pi@retropie:~ $ lsusb
Bus 001 Device 004: ID 045e:0291 Microsoft Corp. Xbox 360 Wireless Receiver for Windows
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
**lsmod**
```
pi@retropie:~ $ lsmod
Module                  Size  Used by
bnep                   10340  2 
hci_uart               17943  1 
btbcm                   5929  1 hci_uart
bluetooth             326105  22 bnep,btbcm,hci_uart
brcmfmac              186343  0 
brcmutil                5661  1 brcmfmac
cfg80211              427855  1 brcmfmac
rfkill                 16037  4 cfg80211,bluetooth
evdev                  11396  4 
joydev                  9024  0 
snd_bcm2835            20511  1 
snd_pcm                75698  2 snd_bcm2835
snd_timer              19160  1 snd_pcm
snd                    51844  4 snd_bcm2835,snd_timer,snd_pcm
bcm2835_wdt             3225  0 
bcm2835_gpiomem         3040  0 
xpad                   12208  0 
ff_memless              4750  1 xpad
uio_pdrv_genirq         3164  0 
uio                     8000  1 uio_pdrv_genirq
uinput                  7454  2 
fuse                   83461  1 
ipv6                  347530  32 
```
```

joolswills, 2016-06-24:
I believe another user had some errors logged in dmesg when using xboxdrv which might be of use. If so, it might be worth searching on them and seeing if there is any known upstream kernel problem or fix.

joolswills, 2016-06-24:
Does it make any difference if you blacklist the xpad driver ?

HerbFargus, 2016-06-25:
#### dmesg 

```
http://pastebin.com/cL0CG6F2
```

the relevant code I'm guessing:
```
[   14.340037] input: Xbox Gamepad (userspace driver) as /devices/virtual/input/input6
[   14.342252] input: Xbox Gamepad (userspace driver) #2 as /devices/virtual/input/input7
[132580.381949] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
```

and blacklisting made it go all sorts of wonky. it initially received input for like a second then the lights went off and it received no more input at all. 

When I ran `sudo rmmod xpad` and ran xboxdrv manually both connected as player one but only one worked with input.

a7mag3ddon, 2017-01-31:
Does anyone know if this issue ist still being worked on? Theres been a few releases in the last 8 months or so since it broke.

Ruffio, 2017-01-31:
@a7mag3ddon when silence/no one posts, that usually means that it doesn't get any attention. Maybe @HerbFargus found a solution?

a7mag3ddon, 2017-01-31:
no solution i am aware of, and still an issue in latest build. I am just testing if i can downgrade the kernal using 

sudo su -
curl 'http://pastebin.com/raw/e2ZKSMr1' -o 'patched_rpi-update.sh' -L -B
tr -d '\015' < patched_rpi-update.sh > /usr/bin/rpi-update
chmod +x /usr/bin/rpi-update
/usr/bin/rpi-update 771a503cfc2a1130e2df2a4ddfc45ffa0f592b3f
rm ~/patched_rpi-update.sh
exit

which i found but its only a work around at best ?

HerbFargus, 2017-01-31:
As far as our usage for retropie we defaulted to a patched xpad driver. As you say xboxdrv is still broken with the latest kernel.

Ruffio, 2017-01-31:
According to https://github.com/xboxdrv/xboxdrv/commits/stable last commit was merged  22 November 2015. It looks like it is not maintaned anymore:-( The driver should be fixed there. Not here. This should be closed...

a7mag3ddon, 2017-01-31:
Xpad driver doesn't work for me properly either.

Ruffio, 2017-01-31:
That driver is maintaned here: https://github.com/paroj/xpad I think.
If I mere you, I would TRT there asap the developers there would be able to help you...

a7mag3ddon, 2017-01-31:
Honestly, don't see the point, all the latest problems are with Xbox one pad and all the 360 issue are over a year old in there too. Guess no one cares enough for the aging hardware, maybe people order Bluetooth now pi3 has it built in.

closed 2017-05-18 by JamesH65