raspberrypi/linux #1541: Driver for Cirrus/Wolfson Audio Card

clivem, 2016-06-22:
@HiassofT Matthias, good job on maintaining your cirrus-rpi-4.4.y tree.... But is there a reason not to want this merged into RPi tree? (Well, aside from not wanting @pelwell to start foaming at the mouth at the extravagant waste of memory, enabling the regulator framework. ;)
It keeps coming up from time to time, from people who own that card, about the lack of "semi-official" support. The code looks cleaner than it did 18 months ago, most likely as a result of your maintaining it I guess, but coming back to the original question, is there a reason why we wouldn't want to see this included in the default Pi kernel?

NB. I don't own the Cirrus card, so I don't really have an interest one way or the other, but I get asked about it from time to time, and I got asked about it again today, hence my raising it here.

HiassofT, 2016-06-22:
Main reason for not sending out a PR is because we are not quite there yet.

I'm not 100% satisfied with the code quality and recently a new bug popped up.

My plan is to do a major cleanup/rewrite of the driver code at some time (lack of time is currently a bit of a problem) to iron out all these issues.

Also there are some nasty changes to upstream code, tests showed that I can remove some of them but I have to go through quite some out-of-tree commits touching upstream code and check if they are really needed or not.

I'd also like to check some things with upstream, for that a working upstream I2S driver would be really helpful (i.e. Martin's clock changes merged).

So it could take a bit until everything's cleared up and for the time being I'll maintain my out-of-tree code.

clivem, 2016-06-22:
> I'd also like to check some things with upstream, for that a working upstream I2S driver would be really helpful (i.e. Martin's clock changes merged).

Yes, that really does need to happen. I'm waiting for Martin to pop his head back up, to say he's survived whatever he was up to with "Java performance optimization"......

> So it could take a bit until everything's cleared up and for the time being I'll maintain my out-of-tree code.

Thanks for the info. Makes sense.

closed 2016-06-22 by clivem