raspberrypi/linux #1546: gcc error

icode, 2016-06-28:
```
drivers/pinctrl/bcm/pinctrl-bcm2835.c:902:10: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .free = bcm2835_pmx_free,
          ^
drivers/pinctrl/bcm/pinctrl-bcm2835.c:902:10: note: (near initialization for ‘bcm2835_pmx_ops.free’)
```

gcc 5

popcornmix, 2016-06-28:
Are you building the default (rpi-4.4.y) tree?
Are you following the instructions?
https://www.raspberrypi.org/documentation/linux/kernel/building.md

pelwell, 2016-06-28:
There is a type mismatch - for some reason the patch to add the free method in the 4.4 tree define it as returning void, whereas in the other tree is it returns an int.

popcornmix, 2016-06-28:
Okay, pushed a fix. @icode is it okay now?

icode, 2016-06-29:
@popcornmix i download release `raspberrypi-kernel_1.20160620-1`

popcornmix, 2016-06-29:
We can only fix the latest tree. Try using git clone on:
https://github.com/raspberrypi/linux/

closed 2016-06-29 by icode