raspberrypi/linux #1547: SPI hardware (MFRC522) not working with Latest Jessie release (2016-05-10-raspbian-jessie)

cscuilla, 2016-06-28:
The rfid reader (MFRC522) does not work with the latest release of Jessie(2016-05-10).  It works fine on the previous version (2016-03-18).  I believe this is due to the new version of SPI.

On previous Jessie version (2016-03-18-raspbian-jessie) if I type lsmod after installing spi I get the following..
spi_bcm2708

On recent Jessie version (2016-05-10-raspbian-jessie) if I type lsmod after installing spi I get the following..
spi_bcm2835

Others, are having the same issue as described on the raspberry pi forum at this link..
https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=148890

pelwell, 2016-06-28:
Can you describe the configuration of your system - any relevant entries in config.txt, cmdline,txt, /etc/modules or /etc/modprobe.d/*.conf?

Also, can you explain the failure mode - what do you expect to happen, and what actually happens?

cscuilla, 2016-06-28:
To get it working on previous Jessie Release (2016-03-18) I have to add the following to /boot/config.txt
> device_tree_param=spi=on
> dtoverlay=spi-bcm2708
This does NOT work on latest Jessie release.  I've also tried
> dtoverlay=spi-bcm7835

The failure mode is as follows..  Once I run the script it appears to work as it is waiting for a card to be scanned, however when I scan a card, nothing is output.

On previous Jessie(where card reader works), when I run the script and scan a card, the UID of the card is printed.

pelwell, 2016-06-28:
Can you confirm that with just `dtparam=spi=on` in config.txt (by which I mean no other spi-related settings) and `spi-dev` in /etc/modules that /dev/spi0.0 and /dev/spi0.1 appear but your script fails?

pelwell, 2016-06-28:
I've just ordered one of these devices, but it probably won't arrive before next week and I can't promise much time to look at this.

cscuilla, 2016-06-28:
what do you mean by this part
> and spi-dev in /etc/modules that /dev/spi0.0 and /dev/spi0.1 appear but your script fails?
Currently, if I open /etc/modules all I have is
> i2c-dev

pelwell, 2016-06-28:
Sorry, that's correct. SPI and I2C are strangely different in that respect the layered, user-space interface to I2C (i2c-dev) isn't loaded by Device Tree because it isn't hardware-specific, but the equivalent module for SPI (spidev) is.

So, to be more accurate: can you confirm that with just dtparam=spi=on in config.txt (by which I mean no other spi-related settings) that /dev/spi0.0 and /dev/spi0.1 appear but your script fails?

cscuilla, 2016-06-28:
OK, thanks, I am setting up a new memory card now with the latest release so I can confirm this for you.

cscuilla, 2016-06-28:
I can confirm this.. In /boot/config.txt I have the following
> dtparam=spi=on

If I go into the /dev folder and type "ls"
I see spidev0.0 and spidev0.1

When I start my script, I receive no output once I scan a card.

cscuilla, 2016-06-28:
Thank you for ordering the hardware.  I understand your time is limited but do appreciate the help. Here are the full steps to get the reader working on previous debian OS in case it helps you to troubleshoot..

1. add to the bottom of /boot/config.txt
   - device_tree_param=spi=on
   - dtoverlay=spi-bcm2708
2. Install python-dev
  - sudo apt-get install python-dev
3. Install SPI
  - git clone https://github.com/lthiery/SPI-Py.git
  - cd ~/SPI-Py
  - sudo python setup.py install
4. Install Sample RFID Reader python library and sample script.
  - git clone https://github.com/rasplay/MFRC522-python.git
4. Run sample script.
  - sudo python ~/MFRC522-python/read.py

EECSdude, 2016-06-28:
I am running Raspberry Pi 2  B on Linux 4.4.11-v7+ #888 SMP Mon May 23 20:10:33 BST 2016 armv7l GNU/Linux with exact same behavior.  I am online the next hour and happy to run any tests, provide any files. 
 
Scope traces show the following.
GPIO 25, the RST line is always high, I don’t see it move.
GPIO 8 is always low, I don’t see it move, although I do see digital noise similar to SCLK at the millivolt level.
SCLK toggles the whole time Read.py is running, as expected.
As an experiment, I rebooted then ran a few commands to toggle GPIO25 up and down to check my scope setup.  When I ran the same script with 25 set to 8, I got a runtime warning in response to GPIO.setup(8, GPIO.OUT)  “this channel is already in use”.  

I would expect to get that message if the GPIO was in CE0 mode, not a GPIO output as gpio readall says it is.

pelwell, 2016-06-28:
You say that GPIO 25 is the reset line, but the code appears to use GPIO 22 - is this a non-standard numbering issue, or have you modified your version of the code?

As mentioned in the Forum thread, the spi-bcm2835 driver uses software-controlled CE lines rather than ones driven by the hardware block, to work around a hardware glitch, which means that the driver is claiming the pin for GPIO - this explains the contention.

I see there is an spidev Python module that uses the spidev interface - does the spi module also do this, or is it a pure user-space software implementation?

EECSdude, 2016-06-28:
I have not modifed Read.py.  Interesting, I'll move the wire and see what happens.

https://github.com/mxgxw/MFRC522-python shows:

Name	Pin #	Pin name
SDA	24	GPIO8
SCK	23	GPIO11
MOSI	19	GPIO10
MISO	21	GPIO9
IRQ	None	None
GND	Any	Any Ground
RST	22	GPIO25
3.3V	1	3V3

pelwell, 2016-06-28:
I can answer the second question myself now - the SPI-Py (spi) module drives the spidev interface from C code.

EECSdude, 2016-06-28:
PIN 22 is GPIO25, what pin configuration is the code using?  Moving the wire now.

cscuilla, 2016-06-28:
For me, The python script uses "board" as the mode for rip.gpio. So the reset is on pin 22 but GPIO number 25.

I'm not sure what u mean in regards to the spidev Python module.

EECSdude, 2016-06-28:
Moving the RFID modules RST pin from Raspberry Pi pin 22 / GPIO25 to pin 15/GPIO 22 doesn't make it work.

pelwell, 2016-06-28:
OK - I see that RPi.GPIO is configured in BOARD mode, so 22 is the header pin number but it carries GPIO 25, so that isn't the problem.

EECSdude, 2016-06-28:
LONG - Here's all the pertinent information in my system that has been asked for in the thread above:

I did one more thing than the previous post, suggested by a website I have forgotten:

Install latest version of bcm2835, C library for Broadcom BCM 2835 as used in Raspberry Pi; version 1.46 currently.
(http://www.airspayce.com/mikem/bcm2835/)
# download the latest version of the library, say bcm2835-1.xx.tar.gz, then:
$ tar zxvf bcm2835-1.xx.tar.gz
$ cd bcm2835-1.xx
$ ./configure
$ make
$ sudo make check
$ sudo make install

The above didn’t cause MFRC522-python\Read.py to work

It felt like the problem is a conflict between 2835 and 2708, so I have added and commented out things with no success and no error indication.

Here's a snapshot of a system that runs Read.py with no error and no card read:

$ ls /dev/spi*
/dev/spidev0.0  /dev/spidev0.1

$ lsmod
Module                  Size  Used by
xt_multiport            1756  1
iptable_filter          1409  1
ip_tables              11637  1 iptable_filter
x_tables               12507  3 ip_tables,xt_multiport,iptable_filter
cfg80211              427855  0
rfkill                 16037  1 cfg80211
evdev                  11396  4
bcm2835_wdt             3225  0
bcm2835_gpiomem         3040  0
uio_pdrv_genirq         3164  0
uio                     8000  1 uio_pdrv_genirq
spi_bcm2835             7286  0
snd_bcm2835            20511  0
snd_pcm                75698  1 snd_bcm2835
snd_timer              19160  1 snd_pcm
snd                    51844  3 snd_bcm2835,snd_timer,snd_pcm
i2c_bcm2708             4770  0
i2c_dev                 5859  0
fuse                   83461  1
ipv6                  347530  45

$ cat /boot/config.txt (clipped only bottom)
# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on
# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
dtparam=spi=on
dtparam=i2c_arm=on
start_x=1
gpu_mem=128

dtoverlay=spi-bcm2835

#dtoverlay=spi-bcm2708

$ cat /etc/modules
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.

i2c-dev
i2c-bcm2708
snd-bcm2835
#i2c-bcm2835
#rtc-ds1307
#spi-dev
spi-bcm2835
#spi-bcm2708

The files “etc/modprobe.d/*.conf” are only empty blacklist files.

pelwell, 2016-06-28:
Thanks for all of that information. A few comments:
1) Don't use `dtoverlay=spi-bcm2835` - that overlay isn't needed and doesn't exist anymore.
2) You don't need to load i2c-bcm2708, snd-bcm2835 or spi-bcm2835 - they are all loaded by Device Tree nowadays.

I don't think either of these will cause your problem, I'm just trying to encourage best practises.

EECSdude, 2016-06-28:
Just noticed the original post and I are using different libraries with same result.
Original post is: https://github.com/rasplay/MFRC522-python
I'm using: https://github.com/mxgxw/MFRC522-python.

Both are 2 years since last update.
Both use SPI-Py.  Common failure point?

EECSdude, 2016-06-28:
Thank you pelwell, followed best practices and still no joy, but I feel so much cleaner!  Just FYI, I have moved this particular RFID hardware to an Arduino and it works.

pelwell, 2016-06-29:
There was an issue around the time of the Pi2 launch caused by a change to the SPI ioctl interface - a field in a structure that was an output became an input and output, and many libraries had to be changed to zero that field before calling the ioctl in question. I'll find the details later.

EECSdude, 2016-06-29:
Anything like this: http://stackoverflow.com/questions/10828194/issue-with-spi-serial-port-comm-stuck-on-ioctl

or this:

 http://www.raspberry-projects.com/pi/programming-in-c/spi/using-the-spi-interface

Issues:
"Error – Problem transmitting spi data..ioctl: Invalid argument

Due to changes in the underlying library the spi_ioc_transfer struct now needs to be initialised to NULL, and a hacky fix is to add this to the beginning of the for loop (this has been done in the code example above)

memset(&spi[i], 0, sizeof (spi[i]));

pelwell, 2016-06-29:
That second link is the issue I was referring to. You could try adding this to spi.c in the SPI-Py library at line 190:
```diff
                .cs_change = 1,
+               .pad = 0,
	};
```
A more future-proof fix would be to memset the structure to zero before setting the required fields, but that would need a bit more restructuring.

If a non-zero pad field is the cause of the problem then I'm slightly suprised that you don't see the "can't send spi message" error message.

cscuilla, 2016-06-29:
I tried adding the change to spi.c, but the issue is the same.  No error message when running script, but no output when trying to read a card.

EECSdude, 2016-06-30:
Any more help I can provide?  We have some great momentum on this.

pelwell, 2016-06-30:
I won't do any more on this until the hardware arrives, which should be next week.

Jbithell, 2016-06-30:
I'm having issues with this as well - I've had it working quite well for some time - a quick SD flash and new raspian install _(as I change project)_ and nothing with SPI seems to be working.

I've tried everything I can think of, on 2 different Pi's (B+ and 3) - and nothing working on either of them. Currently resorting to banging my head against a wall

Jbithell, 2016-06-30:
I'm going to try a rollback using [http://downloads.raspberrypi.org/raspbian/images/]( http://downloads.raspberrypi.org/raspbian/images/)

danbuntu, 2016-06-30:
@Jbithell I 'solved' my issues with this by using an old Wheezy image and not upgrading it for now. Although that means the lastest firmware isn't installed which means the wifi on rPI3 is very flaky.

Jbithell, 2016-06-30:
@danbuntu I'll try that - thanks. Let's hope the Foundation manage to solve this issue!

pelwell, 2016-07-01:
Good new - the reader arrived, and after hooking up the logic analyser and hacking around a bit I got it to work.

The difference between the working setup (spi-bcm2708 driver) and the broken one (spi-bcm2835 driver) is the handling of the CE lines. The spi-bcm2708 driver uses hardware CE, which drives the CE line low 11us before CLK line starts to oscillate, then raises it again 17us afterwards (note that these timings could correspond to operations at the software level rather than being properties of the hardware). spi-bcm2835 lowers the CE line very early - 5ms before the CLK starts - and the reader drives non-zero data onto MISO even during the first transfer. It also keeps CE low through subsequent transfers.

GOOD: MOSI 02, MISO 00 | MOSI 0F, MISO 00 | MOSI 54, MISO 0F | MOSI 8D, MISO 6A
BAD: MOSI 02, MISO 19 | MOSI 0F, MISO 00 | MOSI 54, MISO 0F | MOSI 8D, MISO 54

Hacking the spi-bcm2835 driver and DTBs to use hardware CE gets it working again.

I've run out of time to do any more today, but perhaps @msperl has some comments? Is this device playing fast-and-loose with the spec? Is it reasonable to keep CE low so long?

EECSdude, 2016-07-01:
I remember comments about there being a problem in the Broadcom SPI hardware w/r to CE, and that's why it is software managed.  Did that issue go away with RPi 2?

pelwell, 2016-07-01:
No it didn't, but the circumstances that lead to the fault don't arise in many applications.
I think the driver should support hardware CE when requested using an overlay (this will need a driver change). It may also be possible to make the software CE work with the Mifare device, but since I don't understand the failure that may be wishful thinking.

EECSdude, 2016-07-01:
Excuse the naiveté, but what happens next?  Do we see a fix/patch quickly, and something more permanent later?  Since the GPIO8 CE is managed by software, is there something in the MFRC code that can be modified?  I'm wondering if the hardware just saw an upedge at the end of a transfer if it would be happier.

pelwell, 2016-07-01:
The quick fix is to push a downstream patch that restores the ability to use hardware CE, followed by an overlay that makes use of that ability. Later we could try to understand why the software CE doesn't work and fix it if possible.

pelwell, 2016-07-01:
I've pushed the quick fix and overlay (spi0-hw-cs). They'll be picked up by the next build.

EECSdude, 2016-07-02:
ah, er, um.  Can someone layout the steps for implementing this fix right now, or do I have to wait for the next build?  Yes, extreme newbie.

pelwell, 2016-07-02:
All you'll need is an updated spi-bcm2835.ko (or whatever the filename is) and the overlay. When I get a moment I'll upload them somewhere for you.

pelwell, 2016-07-02:
You can update to a version that should work by running the following:
```
sudo rpi-update
wget -O hw-cs.tgz "https://docs.google.com/uc?authuser=0&id=0B8VsfKAD4-NOUzUtWkVDZ05MLTQ&export=download"
sudo tar zxvfP hw-cs.tgz
```
The rpi-update is to make sure you are running a 4.14 kernel, otherwise the modules won't be compatible.

popcornmix, 2016-07-02:
I've just updated rpi-update kernel, so just running ``sudo rpi-update`` should be sufficient to get this fix.

EECSdude, 2016-07-02:
After following the above steps and rebooting, I'm seeing no change in behavior.  Read.py runs with no errors but won't read a card.  The CE0 line stays low all the time.  gpio readall shows that the pin is in CE0 mode however, when it used to show it as GPIO8 / output.  Running a script, I can toggle GPIO8 low and high, so it is connected.

Linux  4.4.14-v7+ #896 SMP Sat Jul 2 15:09:43 BST 2016 armv7l GNU/Linux

Module                  Size  Used by
bnep                   10340  2
bluetooth             326105  5 bnep
xt_multiport            1756  1
iptable_filter          1409  1
ip_tables              11445  1 iptable_filter
x_tables               13165  3 ip_tables,xt_multiport,iptable_filter
cfg80211              427855  0
rfkill                 16037  3 cfg80211,bluetooth
evdev                  11396  4
snd_bcm2835            20511  1
snd_pcm                75698  1 snd_bcm2835
snd_timer              19160  1 snd_pcm
snd                    51844  5 snd_bcm2835,snd_timer,snd_pcm
spi_bcm2835             6868  0
i2c_bcm2708             4770  0
bcm2835_gpiomem         3040  2
bcm2835_wdt             3225  0
uio_pdrv_genirq         3164  0
uio                     8000  1 uio_pdrv_genirq
i2c_dev                 5859  0
fuse                   83461  3
ipv6                  347466  58

# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.

i2c-dev
#i2c-bcm2708
#snd-bcm2835
#i2c-bcm2835
#rtc-ds1307
#spi-dev
#spi-bcm2835
#spi-bcm2708

Jbithell, 2016-07-02:
Same here I'm afraid
##lsmod
```
Module                  Size  Used by
cfg80211              499234  0
rfkill                 21397  1 cfg80211
joydev                  9194  0
evdev                  11650  3
snd_bcm2835            23163  0
snd_pcm                95441  1 snd_bcm2835
snd_timer              22396  1 snd_pcm
snd                    68368  3 snd_bcm2835,snd_timer,snd_pcm
spi_bcm2835             7424  0
bcm2835_gpiomem         3823  0
bcm2835_wdt             4133  0
uio_pdrv_genirq         3718  0
uio                    10230  1 uio_pdrv_genirq
i2c_dev                 6578  0
fuse                   89800  1
ipv6                  367543  26
```
##/etc/modules
```
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.

i2c-dev

```
##dmesg | grep spi
**null**

pelwell, 2016-07-02:
You have added `dtoverlay=spi0-hw-cs` to config.txt, haven't you?

Jbithell, 2016-07-02:
:tada: It works! Yaay!

EECSdude, 2016-07-02:
It's so easy to spot the un-initiated.  The software now reads the card, CE0 happily toggling with every read.  Thank you for your dedication and your patience.  I really didn't want to give Grandma a key to the whole house.  Thanks again.

pelwell, 2016-07-02:
If we can work out why the software-CE/CS doesn't work then perhaps the overlay won't be necessary in the future, but until then you'll have to keep using it. And because there are some devices and applications that require the software CE in order to work, we can't make hardware CE the default.

pelwell, 2016-07-03:
A remark on Fig. 25 (Timing Diagram for SPI) of the MFRC522 datasheet says:

> The signal NSS must be LOW to be able to send several bytes in one data stream.
To send more than one data stream NSS must be set HIGH between the data streams.

(where NSS is what they call the CS/CE pin).

So spi-bcm2835 holding CS low (active) is causing the malfunction. But why is it doing that? The function spi_transfer_one_message in the SPI subsystem implements the CS logic, particularly the semantics of the cs_change field in the SPI_IOC_MESSAGE ioctl - between messages it forces CS to go inactive briefly, but on the final message it causes CS to stay asserted. The documentation says:

> All SPI transfers start with the relevant chipselect active. Normally it stays selected until after the last transfer in a message. Drivers can affect the chipselect signal using cs_change.
>
> (i) If the transfer isn't the last one in the message, this flag is used to make the chipselect briefly go inactive in the middle of the message. Toggling chipselect in this way may be needed to terminate a chip command, letting a single spi_message perform all of group of chip transactions together.
>
> (ii) When the transfer is the last one in the message, the chip may stay selected until the next transfer. On multi-device SPI busses with nothing blocking messages going to other devices, this is just a performance hint; starting a message to another device deselects this one. But in other cases, this can be used to ensure correctness. Some devices need protocol transactions to be built from a series of spi_message submissions, where the content of one message is determined by the results of previous messages and where the whole transaction ends when the chipselect goes intactive.

The transfer() function in the spi.c file of SPI-Py always sets cs_change to 1. Since it only submits a single message at a time, the behaviour is covered by (ii) above, i.e. it has the effect of keeping CS asserted between transfers, which is effectively until the application is closed. Changing that 1 to a 0 fixes the problem without the need to force HW CS. (You will need to rebuild the library - `sudo python setup.py install`)

I've reported the issue [here](https://github.com/lthiery/SPI-Py/issues/17).

mab5vot9us9a, 2016-07-04:
I can confirm that changing `.cs_change = 1` to `.cs_change = 0` (line 189 in spi.c) and then running the mentioned `sudo python setup.py install` as well as removing `dtoverlay=spi0-hw-cs` from /boot/config.txt fixes the experienced issues with the MFRC522.

hasenkamp, 2016-07-09:
cannot make it work... I've done what "mab5vot9us9a" suggested without any luck.

pelwell, 2016-07-09:
Does the overlay work for you?

hasenkamp, 2016-07-09:
I've tried with and without overlay... and nothing works...
Is there a way to verify if SPI at least respond something?

pelwell, 2016-07-09:
It sounds like you have a different problem - probably hardware. You should try asking on the [Interfacing forum](https://www.raspberrypi.org/forums/viewforum.php?f=44), being sure to describe your wiring and what your symptoms are.

hasenkamp, 2016-07-09:
Do you know if I can dump any other kind of mifare cards, or just those generic ones that come with the mfrc522 kit? I would like to read other mifare cards to check presence.

pelwell, 2016-07-11:
Since the 4.4 SPI driver is working as it should - the fault is in the SPI-Py module (I've raised an issue with the author [here](https://github.com/lthiery/SPI-Py/issues/17)) - can we close this?

hasenkamp, 2016-07-11:
Ok.

cscuilla, 2016-07-11:
Pelwell, thanks so much for investigating and solving this. Do I need to close this sense I posted it, or do you close it. Please advise. Thanks again for your solutions provided

cscuilla, 2016-07-11:
Nevermind, I closed it.

closed 2016-07-11 by cscuilla

lhh8023, 2016-07-16:
hi,with these nodejs script [https://github.com/ocsacesar/rc522](url) I finally read my cards,thanks to the writer,but now I wounder how I can write something to them. hope someone may figure it out...

msperl, 2016-07-20:
Sorry to comment this late (and on an already closed) ticket:

The reason for the "unexpected" behaviour is that the "native" CS implementation (compared to generic GPIO CS) has quite a few shortcomings and is not compliant with the API of "spi_transfer.cs_change".

See http://lxr.free-electrons.com/source/include/linux/spi/spi.h#L700 for a longer description of cs_change.

The native CS can not guarantee this (or any other complicated ones for that matter), as the native-CS is automatically disabling CS after a transfer (or at least after some longer delay). This results that even if cs_change is set on the last transfer (which means keep cs selected) the CS gets disabled eventually. But this is not guaranteed to happen and that is why the issue showed up with GPIO-CS (which honored the API correctly).

This is also one of the reasons why Native-CS is not supported upstreams and why I also recommend that it gets removed downstreams as well and we use GPIO-CS directly in the device-tree.

[129 more replies]