raspberrypi/linux #1549: SATA BRIDGE -  Super Top M6116 doesn't work

klak2, 2016-06-29:
SATA BRIDGE -  Super Top M6116 doesn't work:

kkadmin@rpi1:~ $ sudo lsusb -vd  14cd:6116
Bus 001 Device 007: ID 14cd:6116 Super Top M6116 SATA Bridge
  idVendor           0x14cd Super Top
  idProduct          0x6116 M6116 SATA Bridge
  bcdDevice            2.20

kkadmin@rpi1:~ $ uname -a
Linux rpi1 4.4.14-v7+ #895 SMP Sun Jun 26 13:59:02 BST 2016 armv7l GNU/Linux

[   12.225102] scsi 0:0:0:0: Direct-Access        Mass  Storage Device        PQ: 0 ANSI: 0
[   12.227867] sd 0:0:0:0: [sda] 234441646 512-byte logical blocks: (120 GB/112 GiB)
[   12.228300] sd 0:0:0:0: [sda] Write Protect is off
[   12.228324] sd 0:0:0:0: [sda] Mode Sense: 03 00 00 00
[   12.228759] sd 0:0:0:0: [sda] No Caching mode page found
[   12.228786] sd 0:0:0:0: [sda] Assuming drive cache: write through
[   12.236333]  sda: sda1 sda2
[   12.239170] sd 0:0:0:0: [sda] Attached SCSI disk
[   12.252781] sd 0:0:0:0: Attached scsi generic sg0 type 0
[   12.473492] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
[   12.473507] sd 0:0:0:0: [sda] tag#0 Sense Key : 0x4 [current] 
[   12.473514] sd 0:0:0:0: [sda] tag#0 ASC=0x0 ASCQ=0x0 
[   12.473535] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x85 85 06 20 00 05 00 fe 00 00 00 00 00 00 40 ef 00
[   12.644988] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
[   12.645018] sd 0:0:0:0: [sda] tag#0 Sense Key : 0x4 [current] 
[   12.645030] sd 0:0:0:0: [sda] tag#0 ASC=0x0 ASCQ=0x0 
[   12.645048] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x85 85 06 20 00 05 00 fe 00 00 00 00 00 00 40 ef 00
[   12.683819] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
[   12.683842] sd 0:0:0:0: [sda] tag#0 Sense Key : 0x4 [current] 
[   12.683855] sd 0:0:0:0: [sda] tag#0 ASC=0x0 ASCQ=0x0 
[   12.683871] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x85 85 06 20 00 05 00 fe 00 00 00 00 00 00 40 ef 00

Thank you - klak2

fanleatherface, 2016-07-22:
I am a noob, but I got this  lsusb  14cd:6116 Super Top M6116 SATA Bridge today and can mount devices, copy and write the Laptop HDD on ubuntu 14.04 on linux kernel 4.2.0-22. Ah this is raspberry talk here, so I am sorry for disturbing you. Only thing is, I cannot really read the smartctl data, although I gave the option -d usbcypress.

Ruffio, 2016-08-29:
@klak2 I guess that this should be reported upstream and closed here. When there is a fix for your issue, then eventually it will implemented here. Have you reported it upstream?

popcornmix, 2016-08-29:
Yes, seems likely to be an upstream issue.

popcornmix, 2016-09-02:
Looks to be a duplicate of https://github.com/raspberrypi/linux/issues/1287

closed 2016-09-02 by popcornmix