raspberrypi/linux #1550: Add Support for BTRFS root

dsx724, 2016-06-30:
With BTRFS as the root partition, the Raspberry Pi 3 IO runs a lot faster and feels a lot more responsive. ext4 is slow as dogs on MicroSD cards. Please change config_btrfs=y by default on the kernel defconfig. Compiling the kernel over and over again sucks when I just want to pull the latest. F2FS is enabled so I don't see why BTRFS isn't as well since F2FS has a lot more issues.

Ferroin, 2016-06-30:
I think the reasoning here is that F2FS is relatively small, while BTRFS is a pretty significant amount of code.  Also, F2FS got added almost immediately after it came out, and at that point in time, it actually did appear to be more stable than BTRFS.

I would personally love to see this though, especially considering that:
1. The particular code that would be used by most users (single device, dup profile for metadata and single for data) is generally considered pretty solidly stable these days.
2. If people wanted to, they could set things up with two partitions and raid1 data and metadata profiles, and then the system would keep running if the SD card got corrupted (and you'd get logged errors instead of random crashes).
3. COW filesystems usually give flash storage better life expectancies, especially when dealing with stuff like SD cards where good wear leveling is often almost non-existent.

popcornmix, 2016-07-01:
F2FS was seriously being considered as the default for raspbian after some promising performance reports when it was added to allow wider testing. Unfortunately the performance improvements didn't really pan out (there seemed to be an initial performance improvement, but that disappeared over time), and there seemed to be reliability issues.
As such we'd prefer F2FS wasn't built in, but it's a lot harder to remove something that people may be already using.
BTRFS won't be built in. It is large and will be used by a very small percentage of users, and will encourage more requests for other filesystems to be built in.

Ferroin, 2016-07-06:
Alternatively, it's trivially possible to build an initramfs including it which allows you to boot (at least, on Gentoo and anything that uses Dracut natively it is).

Ruffio, 2016-08-29:
@popcornmix as this feature request is not going to be implemented, this issue should be closed.

closed 2016-08-29 by popcornmix

ben423423n32j14e, 2017-07-10:
Please implement BTRFS.

stormeyes, 2018-11-23:
Please implement BTRFS. + 1