raspberrypi/linux #1552: RPi3: Bluetooth impedes wifi

giddyhup, 2016-07-01:
`Linux berry 4.4.13-v7+ #894 SMP Mon Jun 13 13:13:27 BST 2016 armv7l GNU/Linux`

Wifi gets extremely sluggish when Bluetooth is active. See [forum topic](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=140471) for reference. Try pinging your next hop and resolve a (nonexistant) Bluetooth device (`hcitool name 11:11:11:11:11:11`) simultaneously.

mweal-ed, 2016-07-04:
see also bug 1444

Ruffio, 2016-07-28:
@giddyhup did you read the https://github.com/raspberrypi/linux/issues/1444 ? Did it solve your issue?

giddyhup, 2016-07-29:
@Ruffio I read #1444 . There is no solution there just some more observations. I gave up using BT on a RPi3.

Molodax, 2017-01-25:
Is there any update regarding the issue?

pelwell, 2017-01-26:
I don't think there is anything we can do without Broadcom/Cypress. They have expressed an interest and we have pointed them at the relevant threads, prodding occasionally, but so far I've heard no feedback.

Ruffio, 2017-01-26:
@pelwell Any reference at Cypress that I can ping? I would like to help if I can...

pelwell, 2017-01-26:
They will see your comment, but I'd rather not overwhelm them.

Ruffio, 2017-01-26:
Ok, if there is anything that 'we' can help with regarding test etc., please let us know

Ruffio, 2017-01-26:
A little bit related to this topic of Bluetooth/Wifi: https://github.com/raspberrypi/linux/issues/1410

Ruffio, 2017-01-26:
#1410 
#1402

Molodax, 2017-01-26:
I would add this issue https://github.com/raspberrypi/linux/issues/1342 in bunch of the issues with WiFi and Bluetooth in RPi3, probably, they are interconnected.
At the moment, WiFi is very unreliable by itself and using of Bluetooth and WiFi together is not possible at all in RPi3.

thomasf, 2017-01-26:
I already referenced this issue from #1342 .. The only reliable solution for a
project I was working on with ~40rpis was to use an external usb wifi dongle.
We wanted both bt and wifi and after many hours of systematic testing the only
stable solution was to use an usb wifi.. Doing the opposite (usb bluetooth and
internal wifi) caused problems less often but in the end it still failed from
time to time.

IIRC we created a modprobe.d-file to blacklist a couple of kernel modules
related to the wifi functionality, looking quickly at our provisioning scrips
these ones seems to been have added to our prod setup:

`no-rpi-wifi.conf`:

```
blacklist brcmfmac
blacklist mrcmutil
```

Molodax, 2017-01-26:
Perhaps, using an external usb wifi dongle could be a workaround for some projects but it is not a solution for users of RPi3 project as such.

thomasf, 2017-01-26:
I would also have preferred things to just work and not having to spend some weeks on testing scenarios.. In the end the usb wifi path was a practical solution/workaround about 6-8 months ago (probably still is)  if you do need it to work right now and not maybe work at a unknown future date.

Ruffio, 2017-01-26:
Hmmm, considering that Cypress have expressed an interest but these issues are now more than half a year old, it doesn't look like it has been that that high prioritised :-( It is a little bit sad as IMHO the wifi/bluetooth combo is really a key selling point for RPi3. Hopefully Cypress will give it some (more) attention.

mweal-ed, 2017-01-26:
For reference this problem is not isolated to just the Pi.  I have encountered the same problems on other platforms that have shared hardware for wifi/bluetooth  (Intel Edison and Next Thing CHIP).   I know the Edison also uses a Cypress/Broadcom chipset, but I'm not sure about the CHIP.

mweal-ed, 2018-03-16:
Does anyone know if the pi3 B+ has the same problem?

jason-a69, 2018-04-24:
The Pi3 B+ has exactly the same problem, I can confirm that as I just did some testing.

JamesH65, 2018-04-24:
We did have improved firmware from Cypress that was supposed to help with BT/Wireless coexistence. I believe things did get better, but presumably, not much.

Note that the chip on the 3B+ is different to that on the 3B, with different firmware (we presume), so its a shame that they both exhibit the same issue. Clearly there are hurdles to overcome with coexistence that is still a work in progress.   

I guess we need to prompt Cypress again, unless @pelwell has any comments?

mweal-ed, 2018-04-24:
Out of curiosity, did you test with a 2.4G or 5G wifi connection?  I thought that  with 5G it might be using different circuitry so there would not be a coexistence problem.

jason-a69, 2018-04-24:
I used a 2.4G interface, I do not have a 5G router but you have just given me an excuse to go out and purchase an ASUS RT-AC86U.
I will double check that rpi-config has been run as well although I am sure I did before I started my testing.

JamesH65, 2019-01-18:
@pelwell

pelwell, 2019-01-18:
In August 2018 Cypress delivered some updated nvram-file parameters that have noticeably improved coexistence behaviour. I would suggest downloading a fresh Raspbian to a spare SD card and retesting. Alternatively you can grab the individual files:
* 43438 for Pi 3B and Pi Zero W: https://raw.githubusercontent.com/RPi-Distro/firmware-nonfree/master/brcm/brcmfmac43430-sdio.txt
* 43455 for Pi3 3B+ and Pi 3A+: https://raw.githubusercontent.com/RPi-Distro/firmware-nonfree/master/brcm/brcmfmac43455-sdio.txt

Copy them into the /lib/firmware/brcm folder (you will need to `sudo cp ...`).

jason-a69, 2019-01-26:
> In August 2018 Cypress delivered some updated nvram-file parameters that have noticeably improved coexistence behaviour. I would suggest downloading a fresh Raspbian to a spare SD card and [...]

I can confirm that Bluetooth and WiFi work much better together. I can stream from my phone via Bluetooth to a WiFi connected Pi without problems (using Moodeaudio).

longo92-vrmedia, 2021-01-30:
Any news on how resolve coexistence of bluetooth and wifi on pi0-w?