raspberrypi/linux #1554: [raspberrypi-kernel package] kernel version doesn't match /lib/modules

jorenvo, 2016-07-04:
Not sure if this is the right place to report packaging issues. But the current stable raspberrypi-kernel package for Raspbian Jessie [1] seems to contain a kernel with version 4.4.11+ [2], but the modules for /lib/modules seem to be for 4.4.13+ [3]. This is causing some issues for us [4], eg. lsmod doesn't work anymore, and there is no /dev/input/by-id present.

[1] # apt-cache show raspberrypi-kernel
Package: raspberrypi-kernel
Source: raspberrypi-firmware
Version: 1.20160620-1
Architecture: armhf
Maintainer: Serge Schneider <serge@raspberrypi.org>
Installed-Size: 104203
Breaks: raspberrypi-bootloader (<< 1.20160324-1)
Replaces: raspberrypi-bootloader (<< 1.20160324-1)
Provides: linux-image
Homepage: https://github.com/raspberrypi/firmware
Priority: extra
Section: kernel
Filename: pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20160620-1_armhf.deb
Size: 29878122
SHA256: 76c06968532426610bd1300b375ca3cd68daded2345ab06ec30670c3193f9be1
SHA1: 9325585053f2d77ae1781804e863f8c90c51dff4
MD5sum: 89b81c6d12cbda37457dc44bdd423fe9
Description: Raspberry Pi bootloader
 This package contains the Raspberry Pi Linux kernel.
Description-md5: 7536ab227cf13dcddacdeb86a3d06624

[2] # pi@raspberrypi:/boot $ strings kernel.img | grep '4\.4'
DTOKLinux version 4.4.11+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #888 Mon May 23 20:02:58 BST 2016
\# uname -a
Linux raspberrypi 4.4.11+ #888 Mon May 23 20:02:58 BST 2016 armv6l GNU/Linux

[3] # dpkg-query -L raspberrypi-kernel
...
/lib/modules/4.4.13+/kernel/lib/lz4/lz4_compress.ko
/lib/modules/4.4.13+/kernel/lib/zlib_deflate
/lib/modules/4.4.13+/kernel/lib/zlib_deflate/zlib_deflate.ko
/lib/modules/4.4.13+/kernel/lib/crc-ccitt.ko
/lib/modules/4.4.13+/kernel/lib/lru_cache.ko
/lib/modules/4.4.13+/kernel/drivers
/lib/modules/4.4.13+/kernel/drivers/hid
/lib/modules/4.4.13+/kernel/drivers/hid/hid-microsoft.ko
/lib/modules/4.4.13+/kernel/drivers/hid/hid-cherry.ko
/lib/modules/4.4.13+/kernel/drivers/hid/hid-topseed.ko
/lib/modules/4.4.13+/kernel/drivers/hid/hid-kensington.ko
/lib/modules/4.4.13+/kernel/drivers/hid/hid-magicmouse.ko
/lib/modules/4.4.13+/kernel/drivers/hid/hid-holtek-mouse.ko
/lib/modules/4.4.13+/kernel/drivers/hid/hid-waltop.ko
/lib/modules/4.4.13+/kernel/drivers/hid/hid-samsung.ko
/lib/modules/4.4.13+/kernel/drivers/hid/hid-zydacron.ko
/lib/modules/4.4.13+/kernel/drivers/hid/hid-dr.ko
/lib/modules/4.4.13+/kernel/drivers/hid/hid-holtek-kbd.ko
...

[4] https://github.com/odoo/odoo/issues/12650

popcornmix, 2016-07-04:
ping @XECDesign

XECDesign, 2016-07-05:
Looks okay on this end:
```
pi@raspberrypi:~ $ strings /boot/kernel.img | grep 4.4
4K4|v
n,*,4C4
4@4z
\v464
)4R4#
^s4L4la
4s4@T
DTOKLinux version 4.4.13+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #894 Mon Jun 13 12:43:26 BST 2016
pi@raspberrypi:~ $ ls /lib/modules/
4.4.13+  4.4.13-v7+
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.4.13+ #894 Mon Jun 13 12:43:26 BST 2016 armv6l GNU/Linux
```
Also checked by extracting kernel.img directly from the .deb file.

jorenvo, 2016-07-05:
@XECDesign you're right, not sure what's going on then. Will report back when we figured out the issue. Thanks for the quick response!

closed 2016-07-05 by jorenvo

jimlynnjulian, 2016-08-23:
Hello,
   I've tried to compile the Exar UART driver 21B1411 and get the same error. Is there a solution other than finding 4.4.13+ and installing those headers? If not obvious already, I know a lot about technology, just not this specific area.

Thank you,
Jim Julian