raspberrypi/linux #1555: How to enable Apparmor when build kernel?

zfyseu, 2016-07-06:
Hello, everyone!

I want to use Apparmor on raspbian.
Is there anyone know how to build the kernel so that I can use Apparmor?

Thanks!

popcornmix, 2016-07-06:
Some info here: https://github.com/raspberrypi/firmware/issues/468
But the options are not going to be added to the default raspbian kernel.

For info on how to build the kernel or whether other distributions may support this, it is best to ask on the [forum](www.raspberrypi.org/forums).

zfyseu, 2016-07-06:
Thanks , what does kernel options mean?

popcornmix, 2016-07-06:
Ask in the forum. This isn't the right place to discuss this as ~500 devs gets email notifications on every message here.

closed 2016-07-06 by popcornmix

zfyseu, 2016-07-28:
Hello popcornmix!

I have followed the link to cross compile my raspbian kernel.
But I don't know how and where to eidt kernel option to support apparmor?

Do you know how to do it ?
Thanks!

clivem, 2016-11-14:
> But the options are not going to be added to the default raspbian kernel.

I must be having a bad dream, having just looked at what got added to rpi-4.8.y tree configs? ;)

zfyseu, 2016-11-14:
edit  "/linux/arch/arm/configs/bcm2709_defconfig" by adding the following kernel options:

CONFIG_KEYS=y
CONFIG_SECURITY=y
CONFIG_SECURITYFS=y
CONFIG_SECURITY_NETWORK=y
CONFIG_SECURITY_PATH=y
CONFIG_SECURITY_APPARMOR=y
CONFIG_SECURITY_APPARMOR_BOOTPARAM_VALUE=1
CONFIG_SECURITY_APPARMOR_HASH=y
CONFIG_INTEGRITY=y
CONFIG_INTEGRITY_AUDIT=y

and then run the command "make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2709_defconfig"

combs, 2017-02-06:
i was able to get apparmor running on a Pi3 with these flags!