raspberrypi/linux #1557: ttyAMA0 returns eof.

rewolff, 2016-07-10:
I have a smartmeter connected to my serial port. The serial port returns EOF when I try to read from it using a unix-standard program like "cat". 

I expected 
```
TTY=/dev/ttyAMA0 
cat $TTY >> thelogfile &
stty 115200 < $TTY
```
to work.

pelwell, 2016-07-10:
I'm going to assume that you are on a Pi3 - helpfully you don't say - and that you have connected your Smart Meter to pins 8 and 10 (and a GND). On a Pi3 those pins are usually connected to /dev/ttyS0 because ttyAMA0 is used for Bluetooth. So that your application doesn't have to choose, /dev/serial0 is a link to the default UART on the Pi.

If you don't want to use Bluetooth then add `dtoverlay=pi3-disable-bt` to your config.txt and reboot; this will restore ttyAMA0 is the primary UART, and serial0 will point to it instead.

rewolff, 2016-07-10:
Sorry about not mentioning what pi I have (Well... I /HAVE/ all of them... :-) ). The pi in question is a pi zero. It's running raspbian, updated today. /proc/version says:  Linux version 4.1.13+ )

If I had been using the wrong serial port, I would not have gotten the meter's output correctly. It works fine with "kermit". It also works fine with a little program I wrote that instead of "n = read ... ; if (n == 0) exit ();" does a "sleep (0.1 seconds)" when EOF is returned.....

The "return 0 bytes" is "normal" for a serial port in "NONBLOCK" configuration. Programs like "kermit" (or for more modern people: minicom) expect that to happen and handle it sensibly. So that could explain why it wasn't seen by others. 

It could be that I did not reboot since using kermit. So maybe the "nonblock" flag sticks improperly in the device driver.

pelwell, 2016-07-10:
> It could be that I did not reboot since using kermit. So maybe the "nonblock" flag sticks improperly in the device driver.

Why don't you check that and report back.

rewolff, 2016-07-10:
I usually don't reboot Unix-machines on a whim. It's in an awkward place, and difficult to revive if it doesn't reboot properly... 

Rebooted. Indeed it is a leftover from the previous "kermit" session. Now "cat" works normally....

Sigh... I have rerun "kermit" stopped it... cat still works normally. I have rerun kermit, killed kermit with a HUP signal... cat still works normally.

PeterPablo, 2016-07-11:
Is it possible that you didn't reboot after updating your Raspberry Pi Zero? Currently 4.4 kernels are distributed over `apt`, though above you reported `4.1.13+`. Could you again check your output of `uname -a`, since you did a reboot in the meanwhile.

rewolff, 2016-07-11:
No, after the reboot, I seem to be still running 4.1.13. 

The package: raspberrypi-bootloader seems to bring in both /boot/kernel.img and /lib/modules/<a random ko file>

That package has a version number: 1.20160620-1 that I read as containing a reasonably recent date (less than three weeks ago)..... 
My raspbian reports " Raspbian GNU/Linux 8 " in /etc/issue.net.

rewolff, 2016-07-14:
Lets close this: cannot reproduce... :-(

closed 2016-07-14 by rewolff