raspberrypi/linux #1558: Opengles and opengl

SuperWangCC, 2016-07-14:
Can it upgrade to the newer version?

popcornmix, 2016-07-14:
Can you be more specific?

Ruffio, 2016-07-28:
@SuperWangCC have you more details as asked for by @popcornmix ?

popcornmix, 2016-07-28:
The firmware based driver supports OpenGLES 1.1 and 2.0.
It will never support desktop GL or openGLES 3.0 - the hardware wasn't designed for that.

It is possible the open source GL driver will support newer standards, as it supports software fallback for features the hardware doesn't support. Depending on how critical these features are, the performance may or may not be usable. Some info here:
https://www.raspberrypi.org/blog/another-new-raspbian-release/

But the new driver is currently experimental and not supported here.

Use the [forum](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=135790) or the [git repo](https://github.com/anholt/linux/issues) where the driver comes from for support.

closed 2016-07-28 by popcornmix