raspberrypi/linux #1560: Game causes NULL pointer dereference in kernel

softwarefailure, 2016-07-14:
I have a game that runs fine on x86 Linux but when compiling it for ARM on my Raspberry Pi 2, it often crashes the whole operating system. The issue seems to be related to the ALSA driver because it only occurs when sound is enabled in the game. No sound, no crashes.  

Here is what `var/log/syslog` looks like after a crash. The interesting thing is probably at `13:29:32` because it is reporting `Unable to handle kernel NULL pointer dereference at virtual address 00000000` at this point which could be related to #621.

```
Jul 10 13:26:37 raspberrypi kernel: [   58.723345] snd_bcm2835_pcm_prepare:329  error setting hw params
Jul 10 13:26:37 raspberrypi kernel: [   58.723388] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 0
Jul 10 13:26:47 raspberrypi kernel: [   68.566732] bcm2835_audio_set_params:583  Alsa controls not supported
Jul 10 13:26:47 raspberrypi kernel: [   68.566756] snd_bcm2835_pcm_prepare:329  error setting hw params
Jul 10 13:26:47 raspberrypi kernel: [   68.568377] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 0
Jul 10 13:27:04 raspberrypi kernel: [   84.908452] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 0
Jul 10 13:27:05 raspberrypi kernel: [   85.988429] bcm2835_audio_set_params:583  Alsa controls not supported
Jul 10 13:27:05 raspberrypi kernel: [   85.988455] snd_bcm2835_pcm_prepare:329  error setting hw params
Jul 10 13:27:05 raspberrypi kernel: [   85.988490] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 0
Jul 10 13:27:08 raspberrypi kernel: [   88.852333] snd_bcm2835_pcm_prepare:329  error setting hw params
Jul 10 13:27:21 raspberrypi kernel: [  102.173166] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 0
Jul 10 13:27:23 raspberrypi kernel: [  104.296299] snd_bcm2835_pcm_prepare:329  error setting hw params
Jul 10 13:27:23 raspberrypi kernel: [  104.296777] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 1
Jul 10 13:27:24 raspberrypi kernel: [  105.071317] bcm2835_audio_set_params:583  Alsa controls not supported
Jul 10 13:27:24 raspberrypi kernel: [  105.071345] snd_bcm2835_pcm_prepare:329  error setting hw params
Jul 10 13:27:24 raspberrypi kernel: [  105.071782] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 0
Jul 10 13:27:24 raspberrypi kernel: [  105.071813] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 1
Jul 10 13:27:24 raspberrypi kernel: [  105.226809] snd_bcm2835_pcm_prepare:329  error setting hw params
Jul 10 13:27:24 raspberrypi kernel: [  105.227236] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 0
Jul 10 13:27:24 raspberrypi kernel: [  105.557247] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 1
Jul 10 13:27:24 raspberrypi kernel: [  105.557448] snd_bcm2835_pcm_prepare:329  error setting hw params
Jul 10 13:27:24 raspberrypi kernel: [  105.557478] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 0
Jul 10 13:27:24 raspberrypi kernel: [  105.557501] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 1
Jul 10 13:27:24 raspberrypi kernel: [  105.701905] snd_bcm2835_pcm_prepare:329  error setting hw params
Jul 10 13:27:25 raspberrypi kernel: [  105.845243] bcm2835_audio_set_params:583  Alsa controls not supported
Jul 10 13:27:25 raspberrypi kernel: [  105.845268] snd_bcm2835_pcm_prepare:329  error setting hw params
Jul 10 13:27:25 raspberrypi kernel: [  105.845339] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 0
Jul 10 13:27:25 raspberrypi kernel: [  105.845369] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 1
Jul 10 13:27:25 raspberrypi kernel: [  105.845462] snd_bcm2835_pcm_prepare:329  error setting hw params
Jul 10 13:27:25 raspberrypi kernel: [  105.845491] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 0
Jul 10 13:27:25 raspberrypi kernel: [  105.845514] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 1
Jul 10 13:27:25 raspberrypi kernel: [  105.954211] snd_bcm2835_pcm_prepare:329  error setting hw params
Jul 10 13:27:25 raspberrypi kernel: [  105.954827] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 1
Jul 10 13:27:25 raspberrypi kernel: [  106.058414] bcm2835_audio_set_params:583  Alsa controls not supported
Jul 10 13:27:25 raspberrypi kernel: [  106.058440] snd_bcm2835_pcm_prepare:329  error setting hw params
Jul 10 13:27:25 raspberrypi kernel: [  106.058474] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 0
Jul 10 13:27:25 raspberrypi kernel: [  106.058498] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 1
Jul 10 13:27:25 raspberrypi kernel: [  106.426792] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 0
Jul 10 13:27:25 raspberrypi kernel: [  106.426838] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 1
Jul 10 13:27:25 raspberrypi kernel: [  106.571810] snd_bcm2835_pcm_prepare:329  error setting hw params
Jul 10 13:27:42 raspberrypi kernel: [  123.087638] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 1
Jul 10 13:27:43 raspberrypi kernel: [  124.494882] snd_bcm2835_pcm_prepare:329  error setting hw params
Jul 10 13:27:43 raspberrypi kernel: [  124.494935] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 0
Jul 10 13:27:43 raspberrypi kernel: [  124.494962] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 1
Jul 10 13:27:43 raspberrypi kernel: [  124.651874] snd_bcm2835_pcm_prepare:329  error setting hw params
Jul 10 13:27:43 raspberrypi kernel: [  124.651913] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 0
Jul 10 13:27:43 raspberrypi kernel: [  124.794293] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 0
Jul 10 13:27:44 raspberrypi kernel: [  124.954301] snd_bcm2835_pcm_prepare:329  error setting hw params
Jul 10 13:27:44 raspberrypi kernel: [  124.954904] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 0
Jul 10 13:27:44 raspberrypi kernel: [  124.954945] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 1
Jul 10 13:27:44 raspberrypi kernel: [  125.104276] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 1
Jul 10 13:27:44 raspberrypi kernel: [  125.261245] bcm2835_audio_set_params:583  Alsa controls not supported
Jul 10 13:27:44 raspberrypi kernel: [  125.261272] snd_bcm2835_pcm_prepare:329  error setting hw params
Jul 10 13:27:44 raspberrypi kernel: [  125.390072] snd_bcm2835_pcm_prepare:329  error setting hw params
Jul 10 13:27:45 raspberrypi kernel: [  125.992627] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 0
Jul 10 13:27:45 raspberrypi kernel: [  125.992673] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 1
Jul 10 13:27:45 raspberrypi kernel: [  126.582654] bcm2835_audio_set_params:583  Alsa controls not supported
Jul 10 13:27:45 raspberrypi kernel: [  126.582682] snd_bcm2835_pcm_prepare:329  error setting hw params
Jul 10 13:27:58 raspberrypi kernel: [  139.324682] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 0
Jul 10 13:27:58 raspberrypi kernel: [  139.534663] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 1
Jul 10 13:28:11 raspberrypi kernel: [  152.097032] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 1
Jul 10 13:28:17 raspberrypi kernel: [  158.224712] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 0
Jul 10 13:28:17 raspberrypi kernel: [  158.224755] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 1
Jul 10 13:28:19 raspberrypi kernel: [  159.963772] snd_bcm2835_pcm_prepare:329  error setting hw params
Jul 10 13:28:19 raspberrypi kernel: [  159.963817] bcm2835_audio_set_ctls:556 Couldn't set the controls for stream 0
[... 69 lines cut ...]
Jul 10 13:29:32 raspberrypi kernel: [  233.327958] 9f20: fff9fee9 ffffffff 00000000 00000000 00000000 00000000 b906b8c0 8095ea64
Jul 10 13:29:32 raspberrypi kernel: [  233.327970] 9f40: 803cb8f8 00000000 00000000 00000000 b9069fac b9069f60 80042678 803cb904
Jul 10 13:29:32 raspberrypi kernel: [  233.327981] 9f60: 39718000 00000000 00000000 8095ea64 00000000 00000000 b9069f78 b9069f78
Jul 10 13:29:32 raspberrypi kernel: [  233.327992] 9f80: 00000000 00000000 b9069f88 b9069f88 b906b8c0 8004258c 00000000 00000000
Jul 10 13:29:32 raspberrypi kernel: [  233.328002] 9fa0: 00000000 b9069fb0 8000fbc8 80042598 00000000 00000000 00000000 00000000
[... cut ...]
When the error occurs it usually renders the system unusable. Even rebooting is often not possible any more after the error has occurred making this quite a nightmare to debug.

It only seems to happen when the sound driver is in a stress situation. Unfortunately, I wasn't able to isolate the problem so the only test program I can provide is the full game. You can grab it here: http://www.softwarefailure.de/tmp/demo.tar.gz (I'm sorry but it's a 70 MB file)

Just try to play the game and the kernel should crash pretty soon. Sometimes it takes a few minutes but normally it happens earlier.

popcornmix, 2016-07-14:
I can run the game. Is it enough just to leave it on intro screen playing music to get the failure?
It ran for me until the game exited due to time limited demo (5 minutes I think?)

softwarefailure, 2016-07-14:
No, you need to play the game because it seems to occur when the sound driver is stressed, e.g. lots of sound fx are played and stopped, played and stopped, played and stopped again etc. There are no sound fx during the intro. So please play the game and it should show up.

softwarefailure, 2016-07-18:
Just out of curiosity, have you managed to reproduce the kernel crash on your system?

popcornmix, 2016-07-18:
No. I've just played the game for two 5 minute sessions without seeing the hang.
Are you seeing the issue on a clean jessie install?
Are you using hdmi or analogue audio?

You mention using Pi2 and running at 700Mhz - are you explicitly underclocking?

softwarefailure, 2016-07-18:
Yes, my test system has a clean Jessie install. I've now tested analogue audio and actually it doesn't seem to occur with analogue audio. Only HDMI audio seems to trigger the crash. So please try with HDMI audio and it should be reproducable on your system too...

I'm running the Pi2 at 700mhz because that is the only speed where raspi-config says "None", i.e. no overclocking. So is raspi-config wrong here? What speed should I use for no overclocking instead?

popcornmix, 2016-07-18:
I think you are looking at the Pi1 overclock settings (possibly raspi-config isn't very clear here).
Stock is 900MHz on Pi2.

softwarefailure, 2016-07-20:
Ok, I've now reverted it to 900Mhz. The crash, however, is still there but only when using HDMI. The kernel usually crashes pretty quickly when using HDMI. Also, there are some other audio glitches. When using analogue audio there are no glitches at all but with HDMI there is often a lag of 1 second before the audio can actually be heard. But of course the most annoying thing is that kernel crash because it takes down the whole machine.

softwarefailure, 2016-08-11:
Have you been able to reproduce the crash using HDMI audio?

popcornmix, 2016-08-12:
Yes, I have reproduced.
Are you using any signal handlers in your code? What signals can occur?

softwarefailure, 2016-08-12:
Yes, I'm using two signal handlers: One for `SIGRTMIN` and one for `SIGRTMIN+1`. The code goes like this:

	sigemptyset(&sigact.sa_mask);
	sigact.sa_flags = SA_SIGINFO;
	sigact.sa_sigaction = Custom_SignalHandler;
	sigaction(SIGRTMIN, &sigact, NULL);

Same for `SIGRTMIN+1`...

popcornmix, 2016-08-12:
Can you try: https://dl.dropboxusercontent.com/u/3669512/temp/kernel7.img

softwarefailure, 2016-08-12:
After this update my keyboard and mouse don't work any longer in Raspbian and there is an error from libkmod at startup, "could not open moddep file" or something like this. How can I fix this? I did some googling but I can't really try out any of the suggested fixes because mouse and keyboard don't work at all so I can't really do anything. The only thing I can still access is the NOOBS recovery mode but it doesn't allow me to do anything except doing a complete reinstall :/

popcornmix, 2016-08-12:
Okay, your modules are older than this kernel.
Running ``sudo rpi-update`` then copying the kernel would likely have worked.

Can you ssh in to Pi? I suspect ethernet will still be working as that doesn't require kernel modules?

softwarefailure, 2016-08-12:
Ok, ssh saved the day. After doing `sudo rpi-update` I was now able to try out the new kernel image. However, the problem is still there. Even worse, it seems to occur almost immediately now. With the old image I could often play for a few minutes before it happened, now it's just a few seconds before the kernel says goodbye...

popcornmix, 2016-08-12:
Ah you probably need this too:
https://dl.dropboxusercontent.com/u/3669512/temp/snd-bcm2835.ko
Put it in ``/lib/modules/4.4.16-v7+/kernel/sound/arm/snd-bcm2835.ko`` and reboot.
I've played the game with hdmi output and hit the 5 minute timeout a number of times with this kernel.

softwarefailure, 2016-08-12:
Thanks, this really seems to do the trick. I've played for 15 minutes now without any lockups so I think this solves the problem. Thanks!

On another note, when using HDMI audio there is always a clearly audible lag before sounds start. For example, try the start screen where you have to select a language. The game plays a short sound whenever the mouse cursor is over a flag. However, when using HDMI you won't hear any sound at all when the mouse cursor is over a flag. With analogue audio, however, the sound is played correctly. Also, when using HDMI and a new background music is started the first ~500 milliseconds or so of audio are always cut off. When using analogue audio, however, the music is played perfectly. Do you have any clue why this happens when using HDMI?

popcornmix, 2016-08-12:
HDMI devices usually take a noticeable amount of time to lock on to the audio signal.
If you are closing the audio connection and opening it again you will probably get this silence lag.
Obviously opening the connection just once and keeping it open will avoid this.

As a hack you can play continuous silence from another process and see if that helps.
You can also try adding ``force_hdmi_open=1`` to config.txt which may work (it's really only a debugging flag, so I can't promise it works).

popcornmix, 2016-08-12:
You should get the fixed kernel from latest ``rpi-update`` and it will appear in subsequent raspbian updates.

softwarefailure, 2016-08-12:
I think the problem is that I'm using `snd_pcm_open` for every sound that I play. Every sound effect that is played will result in a call to `snd_pcm_open`, then the device parameters will be set, then `snd_pcm_prepare` is called and then the PCM frames are sent to the device. This works fine with analogue audio but it seems to have its problem with HDMI. It's not only that playback starts with a lag, it also seems that playback is often cut off before all PCM frames have been played. You can clearly hear those cuts when trying the game using HDMI audio. Hmm.

popcornmix, 2016-08-13:
Well, obviously if you call ``snd_pcm_close`` we don't know you are likely to open it again imminently, so we stop outputting audio to HDMI. Can't you just call the open once when launching and the close when exiting?

softwarefailure, 2016-08-13:
Well, I was thinking of `snd_pcm_open` and `snd_pcm_close` using the concept of audio channels. If I used `snd_pcm_open` just once, I'd have to do the PCM mixing (and possibly resampling if frequencies of the sounds to play don't match) on my own. That's quite some work. If I use `snd_pcm_open` for every sound I want to play, ALSA will do everything for me. I just set the device parameters to the audio format of the PCM data and then feed the data using `snd_pcm_writei`. Finished.

I think your audio driver needs to check if there are still open PCM handles before stopping audio output to HDMI. For example, the test program I provided first uses `snd_pcm_open` to open a channel for the background music which is playing almost all of the time. Then, for every sound effect that is played, the program calls `snd_pcm_open` and then `snd_pcm_close` as soon as the sound effect has finished playing. At that time, however, the background music is *still* playing of course, just on another "channel"! And there might also be other sound effects playing of course. 

So in a nutshell, if my program calls `snd_pcm_close` this doesn't mean to stop *all* sound but just that particular PCM stream! When my program calls `snd_pcm_close` there are most probably still other PCM handles still being fed with audio data which must not be affected by calls to `snd_pcm_close` on a different PCM handle.

popcornmix, 2016-08-13:
I don't believe that snd_pcm_close will stop anything if another channel is open.
Can you test playing continuous music, then do a second snd_pcm_open, play a sound, snd_pcm_close - I don't believe the music will stop or the sound will be truncated.
If both the music and sound stop then you will likely get a gap when they both restart.

Ruffio, 2016-08-29:
@softwarefailure how is it going with the test, that @popcornmix suggested? I can see that there have several commits that have referenced to this issue. Has updating fixed your issue?

softwarefailure, 2016-08-29:
Sorry for the delay. As I said, the kernel crash is gone with the latest kernel. However, I think there is still an issue with the HDMI audio driver. I've created a small ALSA demo program which triggers the error. It's available here: http://www.softwarefailure.de/tmp/pisound.7z Please run it like this:

> ./alsastress bg.wav fx.wav
> or
> ./alsastress bg.wav fx2.wav

And then listen very carefully. Here there are clear audio glitches but only when using HDMI audio. There are no glitches when using analogue audio. 

Note that my demo program is extremely sloppy and there is no clean exit, just hit CTRL-C. Its only purpose is to demonstrate that the HDMI audio driver produces glitchy audio under stress. So please don't complain about the style :)

Also note that the glitches can only be heard if you listen carefully. I can hear the following glitches: The background music sometimes interrupts for a very short time and the fx samples sometimes don't play correctly. It sounds like something is cut off sometimes. But you need to listen carefully in order to notice it. And don't forget to compare it with analogue audio to see what it should sound like.

popcornmix, 2016-08-29:
This issue doesn't seem related to the original issue.
If the original problem is solved please close this.
Feel free to create a new issue with the new problem.

closed 2016-08-29 by softwarefailure

softwarefailure, 2016-08-30:
NB: I opened the above issue as #1623.