raspberrypi/linux #1561: mini uart (ttyS0) does not support parity.

rewolff, 2016-07-14:
This is clearly documented in the "peripherals" datasheet. 
However, when an application requests parity (it is required for STM32 bootloader communication) the driver simply writes the data to the register and does not notice that it doesn't stick.... 

I've looked into the tty/serial/8250 driver, and the LCR gets written quite often. IMHO it would be proper if the driver returns "can't do that sir" when you request a feature that isn't available on this implementation. We could detect the "doesn't exist" feature when detecting the port (write a 1 to the PE bit, read it back. If it comes back zero, the feature isn't there) Once detected at initialization time we need to check for the flag on "setting termios" time... Or we could try to do this "live" at setting termios time. 

As far as my research has reached so far, this both runs into the trouble that all the set_termios functions cannot return error as they either return void or simply "return 0;".

grigorig, 2016-07-22:
The right way to handle missing features is to unset the requested features in the termios structure.

JamesH65, 2017-05-18:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.

rewolff, 2017-05-18:
I understand that it looks as if nobody cares. But IMHO it is still a "should be fixed one day", so I vote for leaving it open. I don't know if you can tag it in such a way that it doesn't bother your "overview"...

pelwell, 2017-05-18:
I agree - it shouldn't be hard to fix this one.

pelwell, 2017-05-19:
There's a patch in the rpi-4.9.y kernel tree that implements this feature. It will be in the next rpi-update firmware release.

popcornmix, 2017-05-21:
Should be in latest rpi-update kernel. Please test and close if happy.

JamesH65, 2017-09-13:
Please update to the latest kernel which may contain a fix for this issue.

closed 2017-09-13 by JamesH65

pelwell, 2017-09-13:
Yes - the patch was even accepted upstream.