raspberrypi/linux #1567: Function tracing causes kernel panic with linux 4.4.11-v7+

giraldeau, 2016-07-22:
Starting function tracing with ftrace running 2016-05-27-raspbian-jessie-lite.img and linux 4.4.11-v7+ causes a kernel crash (see below). The computer is the raspberrypi2. 

To reproduce the bug, start a trace with trace-cmd: 

```
sudo apt-get install trace-cmd
sudo trace-cmd record -p function_graph -- sleep 1
```

This triggers the following Oops, which leads to a kernel panic right after. 

```
[   38.023281] Bad mode in data abort handler detected
[   38.045338] Internal error: Oops - bad mode: 0 [#1] SMP ARM
[   38.064659] Modules linked in: cfg80211 rfkill evdev snd_bcm2835 snd_pcm snd_timer snd bcm2835_gpi
omem bcm2835_wdt uio_pdrv_genirq uio ipv6
[   38.087060] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 4.4.11-v7+ #888
[   38.102831] Hardware name: BCM2709
[   38.115506] task: b990b3c0 ti: b993a000 task.ti: b993a000
[   38.130223] PC is at ftrace_ops_no_ops+0x38/0x124
[   38.144175] LR is at arch_cpu_idle+0x48/0x4c
[   38.157698] pc : [<800c6cfc>]    lr : [<80010b4c>]    psr: 200001d1
[   38.157698] sp : b993bf50  ip : 80010b18  fp : b993bfa4
[   38.187769] r10: 00000000  r9 : 808c4758  r8 : 805bde5c
[   38.202252] r7 : 8044546c  r6 : 00000368  r5 : 8044638c  r4 : 00000002
[   38.218082] r3 : 00073700  r2 : b99a8000  r1 : 00000000  r0 : b9073780
[   38.233949] Flags: nzCv  IRQs off  FIQs off  Mode FIQ_32  ISA ARM  Segment kernel
[   38.250788] Control: 10c5387d  Table: 38a3c06a  DAC: 00000055
[   38.265874] Process swapper/1 (pid: 0, stack limit = 0xb993a210)
[   38.281296] Stack: (0xb993bf50 to 0xb993c000)
[   38.295070] bf40:                                     b9073780 00000000 b99a8000 00073700
[   38.312742] bf60: 00000002 8044638c 00000368 8044546c 805bde5c 808c4758 00000000 b993bfa4
[   38.330424] bf80: 80010b18 b993bf50 80010b4c 800c6cfc 200001d1 ffffffff b993bfb4 b993bfa8
[   38.348160] bfa0: 80063da8 80010b10 b993bfdc b993bfb8 80063fd0 80063d80 80063b40 808c3c2b
[   38.365992] bfc0: b993bfb8 808583a4 808cd348 808cd348 b993bff4 b993bfe0 80015c88 80063dc8
[   38.383821] bfe0: 3992006a 00000055 00000000 b993bff8 000095ac 80015b68 55555555 55555555
[   38.401810] [<800c6cfc>] (ftrace_ops_no_ops) from [<80063da8>] (default_idle_call+0x34/0x48)
[   38.420105] [<80063da8>] (default_idle_call) from [<80063fd0>] (cpu_startup_entry+0x214/0x2b0)
[   38.438590] [<80063fd0>] (cpu_startup_entry) from [<80015c88>] (secondary_start_kernel+0x12c/0x138)
[   38.457519] [<80015c88>] (secondary_start_kernel) from [<000095ac>] (0x95ac)
[   38.474452] Code: e340301f e592100c e0033000 e3530000 (e5912514)
[   38.490377] ---[ end trace 3f310fff5227bfc1 ]---
[   38.504835] Kernel panic - not syncing: Attempted to kill the idle task!
[   38.521447] CPU2: stopping
[   38.534123] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G      D         4.4.11-v7+ #888
[   38.551884] Hardware name: BCM2709
[   38.565395] [<80018724>] (unwind_backtrace) from [<80014058>] (show_stack+0x20/0x24)
[   38.583305] [<80014058>] (show_stack) from [<803205a4>] (dump_stack+0xd4/0x118)
[   38.600813] [<803205a4>] (dump_stack) from [<80016188>] (handle_IPI+0x2a4/0x2c4)
[   38.618422] [<80016188>] (handle_IPI) from [<800094e4>] (bcm2836_arm_irqchip_handle_irq+0x80/0xbc)
[   38.637702] [<800094e4>] (bcm2836_arm_irqchip_handle_irq) from [<805b8e84>] (__irq_svc+0x44/0x5c)
[   38.656876] Exception stack(0xb993df48 to 0xb993df90)
[   38.672290] df40:                   00000000 b9f7d3c8 00000000 00000000 b993c000 808605dc
[   38.690986] df60: 10c0387d 80860580 805bde5c 808c4758 00000000 b993dfa4 808614f8 b993df98
[   38.709680] df80: 80010b34 80010b38 60000013 ffffffff
[   38.725315] [<805b8e84>] (__irq_svc) from [<80010b38>] (arch_cpu_idle+0x34/0x4c)
[   38.743304] [<80010b38>] (arch_cpu_idle) from [<80063da8>] (default_idle_call+0x34/0x48)
[   38.762051] [<80063da8>] (default_idle_call) from [<80063fd0>] (cpu_startup_entry+0x214/0x2b0)
[   38.781368] [<80063fd0>] (cpu_startup_entry) from [<80015c88>] (secondary_start_kernel+0x12c/0x138)
[   38.801148] [<80015c88>] (secondary_start_kernel) from [<000095ac>] (0x95ac)
[   38.818933] CPU0: stopping
[   38.832347] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G      D         4.4.11-v7+ #888
[   38.850823] Hardware name: BCM2709
[   38.865044] [<80018724>] (unwind_backtrace) from [<80014058>] (show_stack+0x20/0x24)
[   38.883653] [<80014058>] (show_stack) from [<803205a4>] (dump_stack+0xd4/0x118)
[   38.901827] [<803205a4>] (dump_stack) from [<80016188>] (handle_IPI+0x2a4/0x2c4)
[   38.920100] [<80016188>] (handle_IPI) from [<800094e4>] (bcm2836_arm_irqchip_handle_irq+0x80/0xbc)
[   38.939992] [<800094e4>] (bcm2836_arm_irqchip_handle_irq) from [<805b8e84>] (__irq_svc+0x44/0x5c)
[   38.959818] Exception stack(0x8085ff08 to 0x8085ff50)
[   38.975892] ff00:                   00000000 b9f633c8 00000000 00000000 8085e000 808605dc
[   38.995191] ff20: 80860500 80860580 805bde5c 808c4758 ba7ff6c0 8085ff64 808614f8 8085ff58
[   39.014590] ff40: 80010b34 80010b38 60000013 ffffffff
[   39.030851] [<805b8e84>] (__irq_svc) from [<80010b38>] (arch_cpu_idle+0x34/0x4c)
[   39.049442] [<80010b38>] (arch_cpu_idle) from [<80063da8>] (default_idle_call+0x34/0x48)
[   39.068769] [<80063da8>] (default_idle_call) from [<80063fd0>] (cpu_startup_entry+0x214/0x2b0)
[   39.088665] [<80063fd0>] (cpu_startup_entry) from [<805b3784>] (rest_init+0x88/0x8c)
[   39.107731] [<805b3784>] (rest_init) from [<807e7d34>] (start_kernel+0x3dc/0x3e8)
[   40.504774] SMP: failed to stop secondary CPUs
[   40.520463] ---[ end Kernel panic - not syncing: Attempted to kill the idle task!
```

P33M, 2016-07-22:
```[   38.233949] Flags: nzCv  IRQs off  FIQs off  Mode FIQ_32  ISA ARM  Segment kernel```

The clue is this line. The CPU in question is in FIQ mode.

We had to add `notrace` peppered around FIQ functions in the past to prevent crashes similar to these. This stacktrace looks a bit different since the PC is in a kernel function and not in any of the FIQ functions.

Is this a regression (i.e. have you run profiling successfully on a previous kernel version)?

JamesH65, 2017-05-18:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.
Please update to the latest kernel which may contain a fix for this issue.

closed 2017-05-18 by JamesH65