raspberrypi/linux #1569: rpi-4.4.y doesn't boot on raspberry pi 1

FandM-Tele, 2016-07-23:
Hi
i git clone the rpi-4.4.y on my system and cross compiled it via exist cross-toolchain in https://github.com/raspberrypi/tools, but it doesn't boot on my raspberry pi 1. i must notice that i'v cross compiled the kernel version 4.1.y later and that worked.
the following is the procedure that i used :
git clone https://github.com/raspberrypi/linux
cd linux
get the config.gz from running raspbian image :
gunzip -c config.gz > linux/.config
ARCH=arm CROSS_COMPILE=${CCPREFIX} make oldconfig
ARCH=arm CROSS_COMPILE=${CCPREFIX} make -j2
ARCH=arm CROSS_COMPILE=${CCPREFIX} INSTALL_MOD_PATH=../linux_modules make modules_install

howardqiao, 2016-07-23:
do not use old config
follow this:
https://www.raspberrypi.org/documentation/linux/kernel/building.md
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcmrpi_defconfig

Ruffio, 2016-07-28:
@FandM-Tele have you try @howardqiao solution? Did it work? If yes, then please close this issue.

Ruffio, 2016-08-29:
@popcornmix consider to close this issue, as there haven't been any response/feedback in a month

closed 2016-08-29 by popcornmix