raspberrypi/linux #1570: cpupower utils fails to set governor

asavah, 2016-07-24:
Board: rpi2
Kernel: rpi-4.7.y built with bcm2709_defconfig, no changes to kernel .config were made.
cpupower: built from rpi-4.7.y tree https://github.com/raspberrypi/linux/tree/rpi-4.7.y/tools/power/cpupower

Currently cpupower util from 4.7.y tree fails to set governor via
```
cpupower frequency-set --governor $whatever
```
No error messages whatsoever.
But the governor does not change as can be seen by reading /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor .

strace comparison may shed some light (on rpi2):
http://pastebin.com/spgUVMf

vs strace on odroid-c2:
http://pastebin.com/UbE0X1ts

Frequency and governor however can be read:
```
cpupower frequency-info
analyzing CPU 0:
  driver: BCM2835 CPUFreq
  CPUs which run at the same hardware frequency: 0 1 2 3
  CPUs which need to have their frequency coordinated by software: 0 1 2 3
  maximum transition latency: 355 us
  hardware limits: 600 MHz - 1000 MHz
  available frequency steps:  600 MHz, 1000 MHz
  available cpufreq governors: conservative ondemand userspace schedutil powersave performance
  current policy: frequency should be within 600 MHz and 1000 MHz.
                  The governor "ondemand" may decide which speed to use
                  within this range.
  current CPU frequency: 600 MHz (asserted by call to hardware)
```

Cpufrequtils (AFAIK cpufrequtils are deprecated) and "classic" ```echo ondemand >/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor``` do work as expected.

Cpupower does work as expected on all the hardware I own (intel x86_64, odroid-c2, etc) except pi's.
It's not a big deal, but I thought I'd report it.

Saren-Arterius, 2016-07-31:
I am having the same issue despite I am using a dual socket haswell-e machine, linux-ck 4.6.4. Downgrading cpupower resolves this.

popcornmix, 2016-07-31:
when you run make in tools directory you get this output:
> cpupower               - a tool for all things x86 CPU power

and trying to build it (on Pi) results in:
```
/usr/bin/ld: cannot find -lpci
collect2: error: ld returned 1 exit status
Makefile:222: recipe for target 'cpupower' failed
```
It doesn't look like it's meant to be cross-platform.

asavah, 2016-07-31:
> It doesn't look like it's meant to be cross-platform.

It is cross-platfortm if built properly.
Yes, libpci (pciutils) can be built for pi too, 
of course there is no use it, I've added it just to satisfy dependencies.
```
 ldd `which cpupower`
        linux-vdso.so.1 (0x7effc000)
        libarmmem.so => /usr/lib/libarmmem.so (0x76f1d000)
        libcpupower.so.0 => /usr/lib/libcpupower.so.0 (0x76f09000)
        librt.so.1 => /usr/lib/librt.so.1 (0x76ef2000)
        libpci.so.3 => /usr/lib/libpci.so.3 (0x76ed9000)
        libc.so.6 => /usr/lib/libc.so.6 (0x76da0000)
        libpthread.so.0 => /usr/lib/libpthread.so.0 (0x76d78000)
        /usr/lib/ld-linux-armhf.so.3 (0x76f32000)
        libz.so.1 => /usr/lib/libz.so.1 (0x76d51000)
        libresolv.so.2 => /usr/lib/libresolv.so.2 (0x76d2d000)
```

It works properly on odroid-c2.
It can read the current freq and governor on the pi's.
But it can't set governor.

asavah, 2017-01-12:
Currently it works as it should on kernel 4.9.2 (bcm2709_defconfig)
```
 cpupower frequency-info
analyzing CPU 0:
  driver: BCM2835 CPUFreq
  CPUs which run at the same hardware frequency: 0 1 2 3
  CPUs which need to have their frequency coordinated by software: 0 1 2 3
  maximum transition latency: 355 us
  hardware limits: 600 MHz - 1.20 GHz
  available frequency steps:  600 MHz, 1.20 GHz
  available cpufreq governors: conservative ondemand userspace powersave performance schedutil
  current policy: frequency should be within 600 MHz and 1.20 GHz.
                  The governor "ondemand" may decide which speed to use
                  within this range.
  current CPU frequency: 600 MHz (asserted by call to hardware)
```

```
cpupower frequency-set -g performance
Setting cpu: 0
Setting cpu: 1
Setting cpu: 2
Setting cpu: 3
```

```
cpupower frequency-info
analyzing CPU 0:
  driver: BCM2835 CPUFreq
  CPUs which run at the same hardware frequency: 0 1 2 3
  CPUs which need to have their frequency coordinated by software: 0 1 2 3
  maximum transition latency: 355 us
  hardware limits: 600 MHz - 1.20 GHz
  available frequency steps:  600 MHz, 1.20 GHz
  available cpufreq governors: conservative ondemand userspace powersave performance schedutil
  current policy: frequency should be within 600 MHz and 1.20 GHz.
                  The governor "performance" may decide which speed to use
                  within this range.
  current CPU frequency: 1.20 GHz (asserted by call to hardware)
```

Closed.

closed 2017-01-12 by asavah