raspberrypi/linux #1571: gcc crashes with -mtune=native on RPi3 only

staceycampbell, 2016-07-24:
gcc 4.9.2-10 is core dumping if the command line contains -mtune=native. This is on the new RPi3 64 bit platform only as far as I can tell.

```
pi@emu:~/tmp $ cat hello.c
#include <stdio.h>

int
main(void)
{
        printf("hello world\n");
        return 0;
}
pi@emu:~/tmp $ gcc -o hello -mtune=native hello.c
*** Error in `gcc': double free or corruption (top): 0x001c6cb0 ***
Aborted
```

popcornmix, 2016-07-25:
Sounds like a 64-bit debian issue, rather than anything related to raspberry pi kernel.

noloader, 2016-07-28:
@popcornmix - You seem to have to do a lot of *... NOT related to raspberry pi kernel"*.

I noticed some GitHubs have custom messages. For example, look at the [developer agreement prompt added in the Kovri GitHub](http://github.com/monero-project/kovri/pull/275):

[image]

Maybe you could add similar to help manage the flow of non-relevant bug reports.

Also, you should change the link at [Raspbian Bug Reports](https://www.raspbian.org/RaspbianBugs). The link to ***[filed with Raspberry Pi]()*** points here.

Finally, do you know where the link is to file Raspian Bug Reports?

Ruffio, 2016-08-29:
@staceycampbell has this issue been resolved? Please close it if it has. Thanks.

feilong, 2016-12-18:
A temporary solution is to install gcc from Stretch instead of Jessie, which worked for me when I tried to make linuxbrew work.

I follow most of the steps from this link (https://solarianprogrammer.com/2016/06/24/raspberry-pi-raspbian-install-gcc-compile-cpp-14-and-cpp-17-programs/) except that I installed `gcc` instead of `gcc-6` (so that gcc won't crash when I run `brew install`).

staceycampbell, 2017-01-19:
This issue still exists. Is this the place to report the issue? If not, where?

popcornmix, 2017-01-19:
gcc is a part of debian so I'd suggest here: https://www.debian.org/Bugs/

I'll close this as no linux kernel fix will help this issue.

closed 2017-01-19 by popcornmix

staceycampbell, 2017-01-19:
OK, sure. Bug has been submitted: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=851903