raspberrypi/linux #1576: "Error in 'g++': double free or corruption" when attempting to determine features

noloader, 2016-07-27:
We have a C/C++ library. The library [does not use Autotools](http://cryptopp.com/wiki/External_Dependencies); rather, it configures itself to the platform on the fly.

The library enables (and removes) features based on the compiler's preprocessor macros. To inspect the *base* set of macros, we usually issue `g++ -dM -E - </dev/null`.  To inspect the macros tuned for the platform, we add `-march=native` (and variants).

Attempting to inspect the preprocessor macros on a Raspberry Pi 3, which is ARMv8, results in the following. The ARMv7 image on an ARMv8 device is an interesting configuration, and I suspect it has something to do with the problem.

```
$ g++ -march=native -dM -E - </dev/null
*** Error in `g++': double free or corruption (top): 0x00fc7d68 ***
Aborted

$ g++ -mcpu=native -dM -E - </dev/null
*** Error in `g++': double free or corruption (top): 0x0012fda0 ***
Aborted

$ g++ -mtune=native -dM -E - </dev/null
*** Error in `g++': double free or corruption (top): 0x00285e00 ***
Aborted
```

According to the [GCC docs on ARM options](http://gcc.gnu.org/onlinedocs/gcc/ARM-Options.html), the first command using `-march=native` should kinda work. I'm fairly sure a double-free crash is unexpected:

> `-march=native` causes the compiler to auto-detect the architecture of the build computer. At present, this feature is only supported on GNU/Linux, and not all architectures are recognized. If the auto-detect is unsuccessful the option has no effect.

staceycampbell, 2016-07-28:
This looks like a dup of the [issue](https://github.com/raspberrypi/linux/issues/1571) I logged a few days ago.

JamesH65, 2017-05-18:
Closing as this issue has been deemed unnecessary.

closed 2017-05-18 by JamesH65

popcornmix, 2017-05-18:
The bug has been fixed in gcc:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=70132
But I'm not sure if it has been picked by debian jessie.

staceycampbell, 2017-05-18:
It still occurs with the latest (as of today) jessie.

```
$ uname -a
Linux emu 4.9.24-v7+ #993 SMP Wed Apr 26 18:01:23 BST 2017 armv7l GNU/Linux
```

popcornmix, 2017-05-18:
Not something we can fix here (this repo is for linux kernel).
Really it needs reporting to debian. If they bump to a newer gcc we will automatically get it.
You could try reporting here: https://github.com/RPi-Distro/repo
for a raspberry pi specific patch, but generally it is better if these are fixed upstream.