raspberrypi/linux #1577: GCC can't compile a program using PMULL, PMULL2, AESENC, AESDEC and friends

noloader, 2016-07-27:
Attempting to enable ARMv8 features on a Raspberry Pi 3, which is ARMv8, results in the following. The ARMv7 image on an ARMv8 device is an interesting configuration. Though its an odd configuration, I expect `crc32`, `pmull`, `pmull2` and `aes` to be available to me, even if I have to jump through some hoops to get it.

```
$ g++ --version
g++ (Raspbian 4.9.2-10) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
```

And

```
$ export CXXFLAGS="-DNDEBUG -g2 -O3 -march=armv8-a+crc -mfpu=neon -mfloat-abi=hard -D__ARM_FEATURE_CRYPTO -D__ARM_FEATURE_PMULL"

$ make -j 2
g++ -DNDEBUG -g2 -O3 -march=armv8-a+crc -mfpu=neon -mfloat-abi=hard -D__ARM_FEATURE_CRYPTO -D__ARM_FEATURE_PMULL -fPIC -c cryptlib.cpp
g++ -DNDEBUG -g2 -O3 -march=armv8-a+crc -mfpu=neon -mfloat-abi=hard -D__ARM_FEATURE_CRYPTO -D__ARM_FEATURE_PMULL -fPIC -c cpu.cpp
In file included from cpu.h:32:0,
                 from cpu.cpp:12:
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h: In function ‘uint8x16_t vaeseq_u8(uint8x16_t, uint8x16_t)’:
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h:13707:50: error: ‘__builtin_arm_crypto_aese’ was not declared in this scope
   return __builtin_arm_crypto_aese (__data, __key);
                                                  ^
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h: In function ‘uint8x16_t vaesdq_u8(uint8x16_t, uint8x16_t)’:
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h:13713:50: error: ‘__builtin_arm_crypto_aesd’ was not declared in this scope
   return __builtin_arm_crypto_aesd (__data, __key);
                                                  ^
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h: In function ‘uint8x16_t vaesmcq_u8(uint8x16_t)’:
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h:13719:44: error: ‘__builtin_arm_crypto_aesmc’ was not declared in this scope
   return __builtin_arm_crypto_aesmc (__data);
                                            ^
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h: In function ‘uint8x16_t vaesimcq_u8(uint8x16_t)’:
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h:13725:45: error: ‘__builtin_arm_crypto_aesimc’ was not declared in this scope
   return __builtin_arm_crypto_aesimc (__data);
                                             ^
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h: In function ‘uint32_t vsha1h_u32(uint32_t)’:
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h:13733:40: error: ‘__builtin_arm_crypto_sha1h’ was not declared in this scope
   __t = __builtin_arm_crypto_sha1h (__t);
                                        ^
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h: In function ‘uint32x4_t vsha1cq_u32(uint32x4_t, uint32_t, uint32x4_t)’:
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h:13742:60: error: ‘__builtin_arm_crypto_sha1c’ was not declared in this scope
   return __builtin_arm_crypto_sha1c (__hash_abcd, __t, __wk);
                                                            ^
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h: In function ‘uint32x4_t vsha1pq_u32(uint32x4_t, uint32_t, uint32x4_t)’:
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h:13750:60: error: ‘__builtin_arm_crypto_sha1p’ was not declared in this scope
   return __builtin_arm_crypto_sha1p (__hash_abcd, __t, __wk);
                                                            ^
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h: In function ‘uint32x4_t vsha1mq_u32(uint32x4_t, uint32_t, uint32x4_t)’:
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h:13758:60: error: ‘__builtin_arm_crypto_sha1m’ was not declared in this scope
   return __builtin_arm_crypto_sha1m (__hash_abcd, __t, __wk);
                                                            ^
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h: In function ‘uint32x4_t vsha1su0q_u32(uint32x4_t, uint32x4_t, uint32x4_t)’:
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h:13764:63: error: ‘__builtin_arm_crypto_sha1su0’ was not declared in this scope
   return __builtin_arm_crypto_sha1su0 (__w0_3, __w4_7, __w8_11);
                                                               ^
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h: In function ‘uint32x4_t vsha1su1q_u32(uint32x4_t, uint32x4_t)’:
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h:13770:57: error: ‘__builtin_arm_crypto_sha1su1’ was not declared in this scope
   return __builtin_arm_crypto_sha1su1 (__tw0_3, __w12_15);
                                                         ^
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h: In function ‘uint32x4_t vsha256hq_u32(uint32x4_t, uint32x4_t, uint32x4_t)’:
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h:13776:70: error: ‘__builtin_arm_crypto_sha256h’ was not declared in this scope
   return __builtin_arm_crypto_sha256h (__hash_abcd, __hash_efgh, __wk);
                                                                      ^
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h: In function ‘uint32x4_t vsha256h2q_u32(uint32x4_t, uint32x4_t, uint32x4_t)’:
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h:13782:71: error: ‘__builtin_arm_crypto_sha256h2’ was not declared in this scope
   return __builtin_arm_crypto_sha256h2 (__hash_abcd, __hash_efgh, __wk);
                                                                       ^
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h: In function ‘uint32x4_t vsha256su0q_u32(uint32x4_t, uint32x4_t)’:
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h:13788:56: error: ‘__builtin_arm_crypto_sha256su0’ was not declared in this scope
   return __builtin_arm_crypto_sha256su0 (__w0_3, __w4_7);
                                                        ^
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h: In function ‘uint32x4_t vsha256su1q_u32(uint32x4_t, uint32x4_t, uint32x4_t)’:
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h:13794:68: error: ‘__builtin_arm_crypto_sha256su1’ was not declared in this scope
   return __builtin_arm_crypto_sha256su1 (__tw0_3, __w8_11, __w12_15);
                                                                    ^
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h: In function ‘poly128_t vmull_p64(poly64_t, poly64_t)’:
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h:13800:83: error: ‘__builtin_arm_crypto_vmullp64’ was not declared in this scope
   return (poly128_t) __builtin_arm_crypto_vmullp64 ((uint64_t) __a, (uint64_t) __b);
                                                                                   ^
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h: In function ‘poly128_t vmull_high_p64(poly64x2_t, poly64x2_t)’:
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/arm_neon.h:13809:85: error: ‘__builtin_arm_crypto_vmullp64’ was not declared in this scope
   return (poly128_t) __builtin_arm_crypto_vmullp64 ((uint64_t) __t1, (uint64_t) __t2);
                                                                                     ^
GNUmakefile:757: recipe for target 'cpu.o' failed
make: *** [cpu.o] Error 1
make: *** Waiting for unfinished jobs....
```

noloader, 2016-07-28:
Andrew Roberts mentioned similar at [GCC Issue 70132: -mcpu=native can cause a double free abort.](http://gcc.gnu.org/bugzilla/show_bug.cgi?id=70132#c5). Unfortunately, the trail seems to end there.

popcornmix, 2016-07-28:
This repo is for the linux kernel. Your issue is a debian packaging issue.
Jessie uses gcc 4.9.2-2. I suspect armv8 support requires a newer version of gcc.

You may be better building a newer gcc yourself, or finding a pre-built package (perhaps from debian stretch). But this repo is not the right place for this.

closed 2016-07-28 by popcornmix