raspberrypi/linux #1579: How to edit kernel option when building the raspbian kernel?

zfyseu, 2016-07-28:
Hello popcornmix!

I have followed the link "https://www.raspberrypi.org/documentation/linux/kernel/building.md" to cross compile my raspbian kernel.
But I don't know how and where to eidt kernel option to support apparmor?

Do you know how to do it ?
Thanks!

howardqiao, 2016-07-28:
After 
make bcmrpi_defconfig or bck2709_defconfig
run 
make menuconfig

zfyseu, 2016-07-28:
Hello,howardqiao!

I have tried what you said.
however when I run "make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- zImage modules dtbs"
it will returns: 
"scripts/kconfig/conf  --silentoldconfig Kconfig
*
* Restart config...
*
*
* Configure standard kernel features (expert users)
*
Configure standard kernel features (expert users) (EXPERT) [Y/?] y
  Enable 16-bit UID system calls (UID16) [Y/n/?] (NEW) 
"
Do you know how to deal with it?

Thanks!

lurch, 2016-08-25:
Did you also read https://www.raspberrypi.org/documentation/linux/kernel/configuring.md ?

IMHO this issue should be closed, and you should ask your question on the Raspberry Pi Forums https://www.raspberrypi.org/forums/

closed 2016-08-26 by popcornmix