raspberrypi/linux #1582: w1-therm behavior

dkirkby, 2016-07-31:
I am reading 5 DS18B20 temperature sensors in 4.4.9-v7+ and noticed two features of `drivers/w1/slaves/w1-therm.c` that I have not seen documented:
* The values returned when reading the sensor device are often (always?) the result of the conversion triggered by the previous read, even when this was quite a long time ago.
* The time taken to read the sensor device is usually about 70ms but sometimes closer to 800ms. The datasheet reports a maximum conversion time of 750ms and the driver is indeed calling `msleep_interruptible(750)`, but apparently usually being interrupted.

I have an application where accurate timestamps for temperature measurements are important, so any advice is welcome. I am happy to contribute documentation once I understand better how this is working.

x29a, 2016-08-10:
I think this is not an issue per se with the raspberry version of the linux kernel, right? Id try to get an answer on [a linux mailing list](http://vger.kernel.org/vger-lists.html) or [stackoverflow](https://stackoverflow.com/) instead.

popcornmix, 2016-08-10:
Agreed, I don't see this as being anything Pi specific, so I don't think this is the best place to report this.

JamesH65, 2017-05-18:
Closing as this issue has been deemed unnecessary.

closed 2017-05-18 by JamesH65