raspberrypi/linux #1585: Problem with cross compile kernel

miltmobley, 2016-08-04:
Build of Rpi kernel from RPi source fails:

See the RPi forums conversation with the same subject as the subject of this report.
On a Linux MInt 17 system, I started with:
mkdir ~/raspberry
2. cd ~/raspberry
3. git clone https://github.com/raspberrypi/tools.git
4. git clone --depth 1 https://github.com/raspberrypi/linux.git
5. export CCPREFIX=~/raspberry/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/arm-linux-gnueabihf-
6. cd linux
7.  make ARCH=arm CROSS_COMPILE=${CCPREFIX} bcm2835_defconfig
8. make ARCH=arm CROSS_COMPILE=${CCPREFIX} 

The build proceeded normally until it tried to compile the dts files, where it failed with:
...
Kernel: arch/arm/boot/zImage is ready
DTC     arch/arm/boot/dts/bcm2835-rpi-b-rev2.dtb
Error: arch/arm/boot/dts/bcm2835-rpi.dtsi:31.20-21 syntax error
FATAL ERROR: Unable to parse input tree
make[1]: *** [arch/arm/boot/dts/bcm2835-rpi-b-rev2.dtb] Error 1
make: *** [dtbs] Error 2

On line 31, bcm2835-rpi.dtsi references BCM2835_FSEL_GPIO_OUT, which is defined
in dt-bindings/pinctrl/bcm3835.h, which is not included in bcm2708_common.dtsi.

bcm2835-rpi-b-rev2.dts includes bcm2835-rpi.dtsi
bcm2835-rpi.dtsi includes bcm2835.dtsi
bcm2835.dtsi includes bcm2708_common.dtsi

I compared the dts files to those in the kernel.org 4.4 source. The organization of files
is slightly different there, but it appears the innermost dtsi file does include pinctrl/bcm2835.h

miltmobley, 2016-08-04:
After inserting the missing header file include in bcm2708_common.dtsi, the build failed
with another problem:

DTC     arch/arm/boot/dts/bcm2835-rpi-b-rev2.dtb
Error: arch/arm/boot/dts/bcm2835-rpi.dtsi:59.1-7 Label or path sdhci not found
FATAL ERROR: Syntax error parsing input tree
make[1]: *** [arch/arm/boot/dts/bcm2835-rpi-b-rev2.dtb] Error 1
make: *** [dtbs] Error 2

In bcm2835-rpi.dtsi, the alias "&sdhci" is referenced, but it is not defined in the alias table
contained in bcm2708_common.dtsi.

This appears to be part of a process to change the logic of sd card handling,
so I can't make a workaround for it.

clivem, 2016-08-04:
Is there a reason you are trying to use bcm2835_defconfig, rather than bcmrpi_defconfig?
bcmrpi_defconfig is the 2835 config for the downstream kernel.

popcornmix, 2016-08-04:
https://www.raspberrypi.org/documentation/linux/kernel/building.md

pelwell, 2016-08-04:
Irrespective of the need to use them, I'll fix up the bcm2835 builds.

notro, 2016-08-04:
@pelwell Yes bcm2835_defconfig is broken. I couldn't get it to work on 4.3, I believe it was. If I remember correctly it had to do with dma problems and the mmc/sdhost drivers . It worked before 4.3.

miltmobley, 2016-08-14:
Hi, I tried bcmrpi_defconfig, and the build completed successfully. Thanks
for your assistance, you can close the bug report.

closed 2016-08-14 by popcornmix