raspberrypi/linux #1586: get_rsts=20 regardless of reset reason

mattj0nes2, 2016-08-04:
Hi,

I'm using 'vcgencmd get_rsts' to get the last reset reason, which is what I'm led to believe I should be doing to query this, but the result is always "get_rsts=20". I've tried killing the process that's kicking the watchdog, rebooting from the command line, forcing a kernel panic, and simply pulling the power, but it's always reason 20.

Also, when trying to get the same info via an ioctl call using WDIOC_GETBOOTSTATUS it always returns zero. I read in all of one place that this is zeroed because some other process reads/clears it, and you need to then retrieve it via get_rsts, but that brings me back to the constant code 20 issue ...

I'm on hardware version BCM2709.

Can anyone offer any info on this?

Thanks.

popcornmix, 2016-08-06:
Is this raspbian jessie, or noobs or something else?

mattj0nes2, 2016-08-08:
Raspian Jessie, 4.4.8-v7. Thanks.

popcornmix, 2016-08-08:
From power on I get:
```
pi@domnfs:~ $ vcgencmd get_rsts
get_rsts=1000
```
After ''sudo reboot" I get:
```
pi@domnfs:~ $ vcgencmd get_rsts
get_rsts=20
```
Note you cannot distinguish a watchdog reboot from an explicit reboot, as the reboot mechanism uses the watchdog to reset the chip.

If you are in control of the reboot you can somewhat control what ends up in the rsts register:
```
echo 62 | sudo tee /sys/module/bcm270?/parameters/reboot_part
sudo reboot
```
```
pi@domnfs:~ $ vcgencmd get_rsts
get_rsts=574
```
reboot_part describes the partition to boot from after reset, but if you use an invalid partition it will boot from the default one (first) and you can detect the additional bits set in rsts. reboot_part=63 means don't reboot and is used by ``sudo halt``.

mattj0nes2, 2016-08-08:
We appear to have /sys/module/bcm2835_wdt, not bcm270*. Is that the issue?

popcornmix, 2016-08-08:
I believe ``/sys/module/bcm270?/parameters/reboot_part`` should have been present for quite a while, although your kernel is somewhat old.
Might be worth trying a ``sudo apt-get update && sudo apt-get upgrade raspberrypi-bootloader raspberrypi-kernel`` and a reboot.

closed 2017-05-18 by JamesH65