raspberrypi/linux #1587: Kernel crash using 4.4.16

ezar, 2016-08-07:
```
[24480.835708] INFO: task kworker/u8:2:20204 blocked for more than 120 seconds.
[24480.840274]       Not tainted 4.4.16-v7+ #899
[24480.844783] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[24480.849397] kworker/u8:2    D 805b527c     0 20204      2 0x00000000
[24480.853950] Workqueue: kmmcd mmc_rescan
[24480.858484] [<805b527c>] (__schedule) from [<805b57f4>] (schedule+0x50/0xa8)
[24480.863044] [<805b57f4>] (schedule) from [<8046bb38>] (__mmc_claim_host+0xb8/0x1cc)
[24480.867619] [<8046bb38>] (__mmc_claim_host) from [<8046bc7c>] (mmc_get_card+0x30/0x34)
[24480.872141] [<8046bc7c>] (mmc_get_card) from [<80473d50>] (mmc_sd_detect+0x2c/0x80)
[24480.876681] [<80473d50>] (mmc_sd_detect) from [<8046e28c>] (mmc_rescan+0xc8/0x324)
[24480.881210] [<8046e28c>] (mmc_rescan) from [<8003c644>] (process_one_work+0x154/0x458)
[24480.885715] [<8003c644>] (process_one_work) from [<8003c99c>] (worker_thread+0x54/0x500)
[24480.890224] [<8003c99c>] (worker_thread) from [<80042678>] (kthread+0xec/0x104)
[24480.894710] [<80042678>] (kthread) from [<8000fbc8>] (ret_from_fork+0x14/0x2c)
[24600.896680] INFO: task kworker/u8:2:20204 blocked for more than 120 seconds.
[24600.900070]       Not tainted 4.4.16-v7+ #899
[24600.903380] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[24600.906743] kworker/u8:2    D 805b527c     0 20204      2 0x00000000
[24600.910093] Workqueue: kmmcd mmc_rescan
[24600.913416] [<805b527c>] (__schedule) from [<805b57f4>] (schedule+0x50/0xa8)
[24600.916767] [<805b57f4>] (schedule) from [<8046bb38>] (__mmc_claim_host+0xb8/0x1cc)
[24600.920101] [<8046bb38>] (__mmc_claim_host) from [<8046bc7c>] (mmc_get_card+0x30/0x34)
[24600.923416] [<8046bc7c>] (mmc_get_card) from [<80473d50>] (mmc_sd_detect+0x2c/0x80)
[24600.926728] [<80473d50>] (mmc_sd_detect) from [<8046e28c>] (mmc_rescan+0xc8/0x324)
[24600.930040] [<8046e28c>] (mmc_rescan) from [<8003c644>] (process_one_work+0x154/0x458)
[24600.933328] [<8003c644>] (process_one_work) from [<8003c99c>] (worker_thread+0x54/0x500)
[24600.936636] [<8003c99c>] (worker_thread) from [<80042678>] (kthread+0xec/0x104)
[24600.939926] [<80042678>] (kthread) from [<8000fbc8>] (ret_from_fork+0x14/0x2c)
```

ezar, 2016-08-09:
Nothing?

Ruffio, 2016-08-09:
I think you must explain in details what happened, what did you/program do when it occurred? Can you reproduce it? how? have you done any configuration? which OS? Raspbian?

ezar, 2016-08-09:
I try to answer all this questions:

I can´t reproduce it. It occurs sporadically once or twice a day. 
I think it started when I update the kernel.

I'm using raspbian lite.

x29a, 2016-08-10:
Which raspberry is this on and which storage card do you use?

ezar, 2016-08-10:
Pi3

Samsung Evo MB-MP16DA/EU
man:0x00001b oem:0x534d name:00000 hwrev:0x1 fwrev:0x0

bcutter, 2016-11-01:
I also saw this a few days ago according /var/log/kern.log(.1):

```
Oct 29 12:37:07 RPi1Raspbian kernel: [4461281.508821] INFO: task kworker/u8:1:1675 blocked for more than 120 seconds.
Oct 29 12:37:10 RPi1Raspbian kernel: [4461281.516071]       Not tainted 4.4.13-v7+ #894
Oct 29 12:37:10 RPi1Raspbian kernel: [4461281.520720] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Oct 29 12:37:10 RPi1Raspbian kernel: [4461281.528830] kworker/u8:1    D 805b4bbc     0  1675      2 0x00000000
Oct 29 12:37:10 RPi1Raspbian kernel: [4461281.535471] Workqueue: kmmcd mmc_rescan
Oct 29 12:37:10 RPi1Raspbian kernel: [4461281.539571] [<805b4bbc>] (__schedule) from [<805b5134>] (schedule+0x50/0xa8)
Oct 29 12:37:10 RPi1Raspbian kernel: [4461281.546892] [<805b5134>] (schedule) from [<8046ba94>] (__mmc_claim_host+0xb8/0x1cc)
Oct 29 12:37:10 RPi1Raspbian kernel: [4461281.554828] [<8046ba94>] (__mmc_claim_host) from [<8046bbd8>] (mmc_get_card+0x30/0x34)
Oct 29 12:37:10 RPi1Raspbian kernel: [4461281.563056] [<8046bbd8>] (mmc_get_card) from [<80473cac>] (mmc_sd_detect+0x2c/0x80)
Oct 29 12:37:10 RPi1Raspbian kernel: [4461281.571018] [<80473cac>] (mmc_sd_detect) from [<8046e1e8>] (mmc_rescan+0xc8/0x324)
Oct 29 12:37:10 RPi1Raspbian kernel: [4461281.578888] [<8046e1e8>] (mmc_rescan) from [<8003c644>] (process_one_work+0x154/0x458)
Oct 29 12:37:10 RPi1Raspbian kernel: [4461281.587076] [<8003c644>] (process_one_work) from [<8003c99c>] (worker_thread+0x54/0x500)
Oct 29 12:37:10 RPi1Raspbian kernel: [4461281.595448] [<8003c99c>] (worker_thread) from [<80042678>] (kthread+0xec/0x104)
Oct 29 12:37:10 RPi1Raspbian kernel: [4461281.603042] [<80042678>] (kthread) from [<8000fbc8>] (ret_from_fork+0x14/0x2c)
```
It only happened ONCE - and I´m not sure (and can´t reverse check with "/var/log/apt/history.log" (cause it´s empty)) if it is/was related to package updates; I think there recently were ones kernel related, weren´t they?

Ruffio, 2017-02-05:
@ezar is this still an issue?

JamesH65, 2017-05-18:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.

closed 2017-05-18 by JamesH65

pelwell, 2018-05-17:
What's the connection? The symptom - a hung_task - can have many different causes.