raspberrypi/linux #1590: pps dtoverlay missing parameters

tlhackque, 2016-08-09:
Upgrading to a device tree kernel (latest raspbian via apt-get), I am attempting to convert from a custom GPIO driver for PPS to the built-in one.

My hardware requires assert_falling_edge=true in the PPS configuration.

Linux redbox 4.1.19+ #858 Tue Mar 15 15:52:03 GMT 2016 armv6l GNU/Linux

Reviewing the pps-gpio,c (in drivers/clients) source, there is a device tree parameter for this.
However, it is not documented, and the syntax is not obvious.  The property name passed to of_get_property() is "assert-falling-edge".

I've tried variations of:
dtoverlay=pps-gpio,gpiopin=24,assert-falling-edge

(I've tried assert-falling-edge=on, as well as omitting the hyphens, replacing them with underscores)

2>&1 vcdbg log msg | grep 'dtpar\|dtover'

always reports:
001252.102: dtparam: gpiopin=24
0001252.754: dtparam: assert-falling-edge=true
001254.536: Unknown dtparam 'assert-falling-edge' - ignored

Looking further, it doesn't appear to be in the tree definition file - arch/arm/boot/dts/overlays/pps-gpio-overlay.dts.

Issues:
- The assert-falling-edge parameter should be in the tree definition file.
- The pps-gpio references in the documentation should mention this parameter.
- The device tree docs mention Boolean parameters, but the dtoverlay description  doesn't say what values are accepted by dtoverlay/dtparam.  E.g. "On"? "True"? Just the param name (like the tree).
- The docs don't say anything about errors in config.txt.  Buried in a readme is "vcdbg log msg"; this should be referenced in the docs along with dtoverlay/dtparam.  And advice to check the log after editing!
- For completeness, the "capture_clear" parameter to PPS doesn't have a corresponding device tree parameter, but should.

I'd appreciate it if someone who knows the gory details of devicetree could provide a corrected overlay file that recognizes assert-falling-edge.

Thanks!

pelwell, 2016-08-09:
* I have added the assert-falling-edge parameter to the overlay and the README.

* Booleans are implemented as integers, but with a different storage format; a non-zero value is true, and zero is false. The strings "y", "yes", "on" and "true" (and "down", for GPIO pulls) evaluate to 1, "n", "no", "off" and "false" are 0 (and "up" is 2). Something that looks like a number is parsed as one, so "1" and "0xffff" are both true. Finally, a parameter that is listed in a dtoverlay directive but without an "=" assignment is implicitly assigned the value of 1. This area could be documented better, but our guides tend to use "on" and "off" as the recommend values for true and false.

* Summary DT documentation and descriptions of the overlays and their parameters can be found in /boot/overlays/README. However, the canonical documentation, as mentioned at the end of the README, is [here](https://www.raspberrypi.org/documentation/configuration/device-tree.md). The troubleshooting section does explain how to debug overlays.

* The pps-gpio driver is an unmodified upstream driver so I'm reluctant to add a new DT parameter to it. You could make your request to the maintainer - Rodolfo Giometti <giometti@enneenne.com>.

tlhackque, 2016-08-09:
Thanks for the helpful quick response.  My immediate problem seems to be solved!

I installed **device-tree-compiler** and ran

dtc -@ -o pps-gpio-overlay.dtb -O dtb -I dts pps-gpio-overlay.dts

on the new file, and copied the new dtb to /boot/overlays.

I changed config.txt to 

dtoverlay=pps-gpio,gpiopin=24,assert_falling_edge=true

With this, the PPS source is accepted by ntpd.

I've documented this for the linuxpps discussions list.

I've interacted with Rodolfo before; I just sent him a note suggesting that he add capture-clear.

Thanks again for the quick reply.

I'll close this issue in a day or two - it takes that long to completely validate ntpd.

closed 2017-05-18 by JamesH65