raspberrypi/linux #1594: Feature request - Add a module for sc16is752 to the standard configurations

MasterWuff, 2016-08-11:
Hi,

Recently I had a problem when I was switching from RPi2 to RPi3, because of only one UART interface. Never the less, I decided to buy a SC16IS752 from NXP. It can connected it to the RPi via I2C and SPI. The external interfaces are two UART ports and 8 GPIOs.

But here is the problem, the module isn't in the standard configurations to work with that.
There is the solution to complile the kernel, but I don't feel really confident with that.

Why to add the module?
-The code already exists: https://github.com/raspberrypi/linux/blob/rpi-4.1.y/drivers/tty/serial/sc16is7xx.c
-The overlay already exists (/boot/overlays/sc16is752-spi1.dtbo)
-It is only 24kB
-i'm not the only one who is dealing with one UART interface

Thanks in advance and on spending so much time to keep the system up to date.

pelwell, 2016-08-11:
It's fine with me. @popcornmix?

popcornmix, 2016-08-12:
Okay.

popcornmix, 2016-08-12:
This is now in latest ``rpi-update`` kernel.

MasterWuff, 2016-08-13:
Thanks a lot, this is what I call a support. Less then 24 hours.
Short Feedback: it works GREAT

I will write a short how to do in the forum: https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=146908

popcornmix, 2016-08-13:
Cool. Feel free to close the issue.

DimionDR, 2016-08-18:
Hello!
 Why I2C drivers implementation?!!
Overlay file only for SPI - /boot/overlays/sc16is752-spi1.dtbo

MasterWuff, 2016-08-18:
Hi DimionDR,
the kernel add on is for I2C and SPI. By default there is one overlay for a SPI connection, but you can write your own for I2C. Or have a look here:  https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=146908&start=25
You can use both

DimionDR, 2016-08-18:
Hello,
I am new in DTS, is it possible to change working interface for the sc16is7xx module without recompiling? My problem is what I want to use the module with SPI...
modinfo output:
pi@raspberrypi:~ $ modinfo sc16is7xx
filename:       /lib/modules/4.4.17-v7+/kernel/drivers/tty/serial/sc16is7xx.ko
description:    SC16IS7XX serial driver
author:         Jon Ringle <jringle@gridpoint.com>
license:        GPL
srcversion:     2021DCD9C98A4797C2CB85D
alias:          of:N*T*Cnxp,sc16is762*
alias:          of:N*T*Cnxp,sc16is760*
alias:          of:N*T*Cnxp,sc16is752*
alias:          of:N*T*Cnxp,sc16is750*
alias:          of:N*T*Cnxp,sc16is741*
alias:          of:N*T*Cnxp,sc16is740*
alias:          i2c:sc16is762
alias:          i2c:sc16is760
alias:          i2c:sc16is752
alias:          i2c:sc16is750
alias:          i2c:sc16is741
alias:          i2c:sc16is740
alias:          i2c:sc16is74x
depends:        regmap-i2c
intree:         Y
vermagic:       4.4.17-v7+ SMP mod_unload modversions ARMv7

MasterWuff, 2016-08-18:
I'm sorry, I didn't get the file working with SPI. With i2c it works, but to be honest, the commends to controll the chip are missing (like setting bau drate or gpios). I know how to write to the registers with I2C or SPI, but the overlay was too much for my knowledge. Perhaps someone else has more experience and will complete the work and thread in the forum. To spend nearly two weeks on such a small issue is too long.

pelwell, 2016-08-18:
What @DimionDR has spotted is that the config change only enables support for the I2C interface, not the SPI mode, which explains the behaviour you are seeing:
```
$ grep SC16 .config
CONFIG_SERIAL_SC16IS7XX_CORE=m
CONFIG_SERIAL_SC16IS7XX=m
CONFIG_SERIAL_SC16IS7XX_I2C=y
# CONFIG_SERIAL_SC16IS7XX_SPI is not set
```
I'll reopen this until SERIAL_SC16IS7XX_SPI is also defined.

MasterWuff, 2016-08-18:
Okay, my fault. I will edit the thread in the forum

pelwell, 2016-08-18:
With the SERIAL_SC16IS7XX_SPI option specified, modinfo now includes:
```
alias:          spi:sc16is7xx
...
depends:        regmap-spi,regmap-i2c
```
I've merged a patch that adds this option to the standard configurations. It will get picked up by the next kernel build.

popcornmix, 2016-08-19:
latest ``rpi-update`` firmware has this in

Ruffio, 2016-08-29:
@MasterWuff have you tried the new firmware? Feel free to close the issue, if the issue has been fixed.

MasterWuff, 2016-08-30:
Just checked it with SPI - it works

closed 2016-08-30 by MasterWuff

paulenuta, 2016-10-27:
Hi there,
I try to add a sc16is752-i2c but command `dtc -@ -I dts –O dtb –o sc16is752-i2c.dtbo sc16is752-i2c.dts` give me `Error: missing files`.  If I load sc16is752-spi1 overlay in /boot/config.txt I have no option to put'it on i2c and is not worling. If I put  sc16is750-i2c in /boot/config.txt is working but I only get ttySC0 :(.
Also datasheet said 1843200 crystal.
Please advise.
Thanks,
Paul

pelwell, 2016-10-27:
Please start a topic on the [Device Tree forum](https://www.raspberrypi.org/forums/viewforum.php?f=107) with your problem - please include full details of what you have tried, and post or link to your overlay file(s).

paulenuta, 2016-10-28:
OK, I was hopping i get some help here. my interest and issues are wit existing overlays that have been discussed here, why there is available only overlays for single uart sc16is750 on i2c and dual sc16is752 on spi? There is come config where I can tell that the sc16is752-spi1 overlay will be used on i2c?
Thanks a lot.

paulenuta, 2016-10-28:
Problem was solved, thanks to PhilE, it was caused by copy/paste and "EN Dash" mixed with "Hyphen-Minus"
[https://www.raspberrypi.org/forums/viewtopic.php?p=1059652#p1059652](url).

pelwell, 2016-10-28:
Cool. Yes, that PhilE can be quite helpful. ;-)