raspberrypi/linux #1598: Feature Request - Add IPv6 6RD module to standard configuration

rickard2, 2016-08-12:
Hello

I'm following this guide trying to enable IPv6 using 6RD on my Raspberry Pi: http://www.litech.org/6rd/

It seems that the current kernel `4.4.17+ #109` does not support this: 

```
$ zgrep 6RD /proc/config.gz
# CONFIG_IPV6_SIT_6RD is not set
```

Would it be possible to add this to the standard configuration? 

Thank you for all your great work, keep it up!

JamesH65, 2017-05-18:
@popcornmix Still not in 4.9 kernel - worth adding? It doesn't seem to be included in very many architectures.

popcornmix, 2017-05-22:
The code size increase is low and it is enabled in Ubuntu so I've let this in: https://github.com/raspberrypi/linux/commit/8dad3ff88215827e4a5ada04a717ea8fda01c267
Will be in next rpi-update kernel.

popcornmix, 2017-05-26:
Should be in latest rpi-update kernel. Please test and close if happy.

rickard2, 2017-05-27:
Works like a charm, thank you very much!

closed 2017-05-27 by rickard2