raspberrypi/linux #1599: RPi A + Raspicam: periodically system freeze after playing audio

crashmaster1, 2016-08-13:
System: RPi model A with Raspicam, 256 MB, Archlinux, Kernel 4.4.17

After playing an audio file (device-tree-param=audio=on, snd-bcm2835 module is used) the system freezes for a period up to 10-15 secs until it reads out input from key buffer. This freeze period is repeated again and again.

'rmmod snd-bcm2835' helps and stops the freezes, everything seems to work fine again.
Any ideas or is it a bug?
Thanks and greetings, CM.

popcornmix, 2016-08-15:
How is raspicam involved with this bug report?
Can you report ``dmesg`` output after a freeze?

crashmaster1, 2016-08-15:
The raspicam hatdware is only connected,  no cam specific modules are loaded and nothing accesses the cam after bootup. 'dmesg' output doesn't make sense because there is no change after the freeze.
It also doesn't matter what kind of audio player is used. mpg123, soxplayer, mpv - all end up with this freeze.
I have another RPi (same model) without cam (identical software versions and configs) and everything works fine there.

popcornmix, 2016-08-15:
So does disconnecting the camera cable without changing anything else avoid the freeze?

crashmaster1, 2016-08-15:
I had also tried that and audio works fine then. I also built the cam into my other RPi - same effect.
Would be helpful now to hear from somebody else using the same equipment.

popcornmix, 2016-08-15:
It sounds very like a power supply issue. The camera requires more power when connected and you get a voltage drop as the supply can't handle it.
Measuring the 5V power on the Pi itself under load would be useful.
You would also try a different power supply and/or power supply USB cable.

crashmaster1, 2016-08-15:
That's an idea although i use only external power supplies 5V/3A (not the cheap crap for smartphones). Will try it later and come back to report. Thanks so far!

crashmaster1, 2016-08-15:
I connected it to my professional power supply normally used for electronical developments and regulated it to 5.1 V but it's not the reason - freezes like before.

popcornmix, 2016-08-15:
Have you measured the voltage at the Pi end with a voltmeter when under load?
The USB cable used for power supply is frequently the cause of the voltage drop.

LoyalNine1487, 2016-08-15:
Here's my idea to check
1. Do you have a way to check the video output of the pi? If so right before it freezes it their a rainbow icon in the top right corner?

popcornmix, 2016-08-15:
Unfortunately no rainbows on model A. That needs a Pi2/Pi3.

crashmaster1, 2016-08-15:
I know about these problems and the power supply for testing is a professional equipment for engeneers. It has no problem with voltage regulation, always stable 5.1 V even if it needs up to 5 A - i ialso have measured that.
The RPi including cam and wlan dongle needs round about  3.5 Watts (= 700mA).
I also use only double shielded cables for USB and not these cheap thin ones.

I should try it with an older kernel but i need f2fs at boottime for rootfs - the Arch Kernel includes it.
Have to look to find a kernel some versions back. On the other side: A 'modprobe snd-bcm2835; play foo.mp3; rmmod snd-bcm2835' is a workaround - ugly but it works. :)

closed 2016-08-20 by crashmaster1