raspberrypi/linux #1600: gpio-poweroff dt overlay also switches pin when "halt --halt" is executed

MaxValue, 2016-08-16:
In #1031 an optional feature (`gpio-poweroff` device tree overlay) was introduced to mimic the poweroff behaviour of standard desktop computers: If the system is being powered off, also the power supply will be "detached" - the LEDs are off. This is like making an ACPI call to the power supply unit.

So first of all, in the documentation of gpio-poweroff (overlays README file) it is written:
> Info:   Drives a GPIO high or low on reboot

This is clearly not correct, because this feature only activates on poweroff (not reboot), right?

**And now for the real issue:** The feature also activates on halt - eg. only the software is completely stopped, but the hardware still has power. Test it yourself with `shutdown --halt now` or `halt --halt` or `systemctl halt`.
Is this maybe intended?

I searched for similar issues and only found the citation of [this bugfix](https://github.com/raspberrypi/linux/wiki/Upstreaming#poweroff), but i am not sure if this is related.

Maybe i am just understanding the behaviour of "halting a system" differently here, if so please tell me too.

Thanks a lot in advantage!

pelwell, 2016-08-22:
The reference to "reboot" in the README is sloppy use of language - we'll correct it.

The standard halt behaviour on 32-bit ARMs is to disable interrupts and leave the CPUs spinning. The RPi branches have a patch (3ce05d6e2e4994e06ec73ff95969ae189681a8e8) that makes halt equivalent to poweroff. This was to reduce power consumption and heat in what should be a low-power mode, so the change was intentional.

If you can explain why would you want to halt without powering off then we can consider whether it is worth reinvestigating.

MaxValue, 2016-08-22:
Thank you for reading my issue!
I wanted to point out the halt-behaviour, because i learned "If you see a bug, report it.". Also it kind of shook my beliefs about shutdown-behaviour in general.

For me it is no real problem, i just thought that this information might be useful to this project. I dunno.

pelwell, 2016-08-22:
OK, for now I will say "This behaviour is by design" (at least for RPi kernels). I've added a note to the README to this effect. If this proves to be a problem for users then we can revisit the issue.

closed 2016-08-22 by pelwell