raspberrypi/linux #1601: UART1 mapped to GPIO 14 15 is not stable

Cheong2K, 2016-08-17:
I have a Bluetooth module (BCM43438) connected to GPIO 14 15 of my Pi Zero, I also make use of the UART1 to test the module, and found that it sometimes fails to load the driver. If it loads successfully, the BlueZ will work fine.

pelwell, 2016-08-17:
1) UART1 is inferior to UART0, with smaller FIFOs and no flow control. Also, its source clock is the same as the VPU/GPU clock, which means that unless you force core_freq=250 the baud rate will change as the Pi enters and leaves turbo mode, meaning you will get garbage out.

2) Try connecting a terminal emulator (e.g. putty) to port 14&15 and verify that no data is transmitted until hciattach is called. I've found the BCM43438 is sensitive to receiving random data before the HCI traffic.

Cheong2K, 2016-08-18:
Thank you for the information and you are right, we got garbage from the UART1. I think we will change to use UART0.

BTW, the Pi3 has the UART clock = 48000000 by default, is it possible to have it on the Pi Zero without adding init_uart_clock in /boot/config.txt?

pelwell, 2016-08-18:
I don't see why we couldn't standardise on a 48MHz clock - any increase in power usage would be tiny, and with a 16.6 divisor the minimum data rate with the higher clock is still about 45baud, which is is too low for anybody to want to use.

Any thoughts, @popcornmix?

pelwell, 2016-08-22:
I'm going to close this issue and track the clock rate change progress in [your issue in the firmware repo](https://github.com/raspberrypi/firmware/issues/643) instead.

closed 2016-08-22 by pelwell

Cheong2K, 2016-08-23:
Excellent! Thank you.