raspberrypi/linux #1602: asm/types.h issue?

arrmo, 2016-08-18:
Hi,

Sorry if this is a dumb question, but struggling with a module build. I have downloaded the source, but when I try to build I get an error message as follows,

```
In file included from include/linux/types.h:5:0,
                 from /home/pi/Documents/EW-7811UTC_Linux_Driver_4.3.19/driver/rtl8821AU_linux_v4.3.19.5_17672.20160506_BTCOEX20150921-58/include/basic_types.h:80,
                 from /home/pi/Documents/EW-7811UTC_Linux_Driver_4.3.19/driver/rtl8821AU_linux_v4.3.19.5_17672.20160506_BTCOEX20150921-58/include/drv_types.h:31,
                 from /home/pi/Documents/EW-7811UTC_Linux_Driver_4.3.19/driver/rtl8821AU_linux_v4.3.19.5_17672.20160506_BTCOEX20150921-58/core/rtw_cmd.c:22:
include/uapi/linux/types.h:4:23: fatal error: asm/types.h: No such file or directory
 #include <asm/types.h>
```

And looking at https://github.com/raspberrypi/linux/blob/rpi-4.4.y/include/uapi/linux/types.h, it does seem to point to asm/types.h ... which doesn't exist.

Thoughts?

popcornmix, 2016-08-23:
Are you following the kernel building documentation?
https://www.raspberrypi.org/documentation/linux/kernel/building.md

arrmo, 2016-08-24:
Yep! I can build the kernel, no issue at all. Seems to be an issue specific to the rtl8812au driver ... :(.

Will close this, not a kernel issue. Thanks!

closed 2016-08-24 by arrmo