raspberrypi/linux #1603: Add device EFT30 Sagem descriptor to USB driver stucture

krajcovic, 2016-08-18:
USB Device - idVendor=079b, idProduct=0028 hasn't record in USB driver structure in cdc-acm.c

I need only add this 3. lines to drivers/usb/class/cdc-acm.c
{ USB_DEVICE(0x079b, 0x0028), /* Ingenico EFT30 Sagem Terminal */
        .driver_info = NO_UNION_NORMAL, /* has no union descriptor */
},

dmesg: from my compiled kernel with changed cdc-acm.c
[ 1380.309578] usb 1-1.3: new full-speed USB device number 5 using dwc_otg
[ 1380.412944] usb 1-1.3: New USB device found, idVendor=079b, idProduct=0028

6by9, 2016-08-18:
That is an upstream file, so any changes need to go in upstream first.
From MAINTAINERS
USB ACM DRIVER
M:      Oliver Neukum <oliver@neukum.org>
L:      linux-usb@vger.kernel.org
S:      Maintained
F:      Documentation/usb/acm.txt
F:      drivers/usb/class/cdc-acm.*

So you'll need to post on linux-usb making the request (ideally with a suitable patch), and report back here when it has been reviewed and accepted to the upstream kernel.

krajcovic, 2016-08-18:
Thanks for answer. I'll attempt a contribute to linux-usb

JamesH65, 2017-05-18:
Closing as this issue has been deemed unnecessary.

closed 2017-05-18 by JamesH65