raspberrypi/linux #1604: SPI Slave

TheJokiv, 2016-08-18:
Dear friend
Is it possible to realise SPI slave mode.
The "slave mode" need when we communicate with NCN5120.
NCN5120 - it is a smart transceiver for KNX-TP and it can work only in SPI master. So for communicate with him, we need SPI slave in Raspberry.
Thank you very much

pelwell, 2016-08-18:
10 seconds of googling turned up [this](http://raspberrypi.stackexchange.com/questions/36169/can-raspberry-pi-function-as-spi-slave) , which links to [this](https://www.raspberrypi.org/forums/viewtopic.php?p=810290#p810290) and others. So yes it may be possible, but you would need to write (or find) a Linux driver.

msperl, 2016-09-09:
Geert Uytterhoeven recently started an effort in the linux kernel to support spi-slave, but the last I remember it has not gone very far (yet).

See the first RFC patch: https://lkml.org/lkml/2016/6/22/423

Note that as the BSC device does not support DMA (and has small FIFOS) there also kernel-interrupt latency issues that would force you to use a low spi clock frequency work stable.

This can be a special problem when running "READ" (master sends: <command byte> <READ> <READ> <READ>....) operations where the kenrel has to react within one SPI clock cycle deciding what to send for the next byte...

pelwell, 2018-04-29:
It doesn't make sense for us to undertake this work - one comment in a year hardly constitutes an outcry - but if someone writes a driver we'll take it.

surge9000, 2018-05-02:
yeah, many people have, including: https://github.com/matthiasbock/Raspberry-Pi-SPI-slave

the problem seems to be you 
Pi folks need to 'help' the SoC designer realise that they arent MASTERS OF THE UNIVERSE!, i.e., connect the pins properly, so we can do this in hardware.

nobody 'complains' on github, because those people are the type to just write it themselves... if thats what you base your metrics on, well... you're not getting very accurate results; is all I can say.

JamesH65, 2018-05-02:
@surge9000 If you have suggestions for other metrics to use on which features to implement, we are all ears, but I'm not sure what is wrong with the number people who ask us for it via this github issue. We certainly are not going to waste time trawling round the internet looking for things to do - we have plenty already on our plates here!

surge9000, 2018-05-21:
:) case in point

mahatma1, 2018-07-13:
> Geert Uytterhoeven recently started an effort in the linux kernel to support spi-slave, but the last I remember it has not gone very far (yet).

I just researched this because I'd need spi slave functionality on my raspberry as well. seems like he succeeded and spi slave support was added to the linux kernel in v4.13 (https://lkml.org/lkml/2017/7/4/435).

I'm wondering if that changes anything? futhermore I found out that the linux repository also contains some files with an spi-driver for the BCM2835 (https://github.com/torvalds/linux/blob/master/drivers/spi/spi-bcm2835.c). since we got kernel support and we've already a driver for the BCM-chip, could you point out what's still missing to get it actually working?

msperl, 2018-07-13:
It is not as simple - it is a totally distinct HW-block that would require a separate driver just for Spi-slave functionality (which also would support i2c slave).
So feel free to develop it.

mahatma1, 2018-07-13:
> It is not as simple - it is a totally distinct HW-block that would require a separate driver just for Spi-slave functionality (which also would support i2c slave).

thanks for the quick reply.
I just noticed that you actually helped to develop the BCM2835 driver I linked in my previous message :)
the reason I thought that this driver might help is because as far as i known the raspberry itself (atleast v1) is based on a BCM2835 SoC. I've also looked into the BCM2835 datasheet which states that there is spi slave functionality, that can be enabled by setting certain control register. I wasn't really aware that there are still some essential parts missing to get it actually working on the raspberry.

since I'm very new to this kind of work and I still don't fully understand which problems need to be solved to get it working I don't think I'm suited to actually develop it myself. for my current needs I'll probably go with a workarround by adding an arduino and outsourcing the spi slave communication parts to it (I'm working on a university project that needed to be finished within several weeks).

however, thanks again for your reply - keep up the good work! :)

ropg, 2019-12-22:
I too would love to have working SPI slave functionality to make Pi Zeroes talk to one another... I'll happily pay a bounty if someone feels like doing the work...

goshhhy, 2020-01-03:
i would also really appreciate having this functionality available.