raspberrypi/linux #1606: Net boot vs SD Card Boot Framebuffer Initialization

Synoia, 2016-08-19:
We are booting a R-Pi with uboot download the kernel and dtb.

When booting for the sd card, framebuffer is set correctly and fbset has values consistent with the parameters from tvservice -s.

When booting everything, including kernel and dtbs from the file server, and starting uboot from the last line in config.txt (the initial boot is from the sd card)  tvservice -s shows the same values as sdcard boot, however fbset shows 800x400 resolution which is not consistent with service.

We can see the kernel start at the correct size, and then after about 10 lines of long the scree changes resolution. 

How do we correct this miscommunication between firmware and kernel?

Synoia, 2016-08-19:
> "start at the correct size" should be "start at the correct screen resolution"

Synoia, 2016-08-20:
Putting raspbian Lite on a new sdcard reveals further inconsistencies between the hardware's view and the setup of the frame bufffer:
tvsercice -state 0x12000a [HDMI CEA (16) RGB 11m 16:9], 1920x1080

fset
mode "1824x984"
   geometry 1824 984 1824 984 16
   timings 0 0 0 0 0 0 0
   rgba 5/11,6/5,5/0,016
emdmode    

fbset -match results in the loss of the 4 raspberries at the top of the screen and no change in resolution on the screen, which appears inconsistent with the definition of the -match parameter with fbset, which is: "make the physical resolution match the virtual resolution"

That did not happen.

fsbset 1920 1080 1920 1080 32 appears to set the display to the same resolution as tvservice -s, and simultaneously eliminates the 4 raspberrys at the top of the screen.

Entering fsbset 1920 1080 1920 1080 32 through a terminal window makes the screen go black, and there is no further display. One suspects x is using a pointer to the frame buffer and not a method, which is poor practice. 

Possibly the x command xrandr could drive fbset, but we have done no testingn.

popcornmix, 2016-08-23:
By default firmware passes the framebuffer size (based on the detected display size) to the kernel through the command line. e.g:
```
cat /proc/cmdline 
8250.nr_uarts=0 dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2709.boardrev=0x2a02082 bcm2709.serial=0xed4e363 smsc95xx.macaddr=B8:27:EB:D4:E3:63 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  console=ttyS0,115200 kgdboc=ttyS0,115200 console=tty1 nfsroot=10.3.14.63:/home/dc4/rootfs ip=dhcp rootwait ```
```
Are you seeing that in the uboot and non-uboot cases? I suspect not.
uboot isn't something we use ourselves or support, but I suspect it doesn't preserve the cmdline the firmware creates. If that is the case then you will have to supply a suitable cmdline to uboot yourself.

Synoia, 2016-08-23:
Thanks. That's the information I needed. The u-boot information I have does not discuss preserving the kernel boot parameters. It appears the contents of commandline.txt are appended to the parameters supplied by the firmware.

Synoia, 2016-08-23:
PC

Interesting, u-boot appears to completely swallow the firmware built kernel parameters. Can you tell me where is the kernel parameter string built by the firmware?  Anywhere accessible?Thanks 
Duncan Hare

714 931 7952

Synoia, 2016-08-23:
Interesting, u-boot appears to completely ignore the firmware built kernel parameters. Can you tell me where is the kernel parameter string built by the firmware?  Anywhere accessible?
Thanks

popcornmix, 2016-08-24:
The firmware is not open source.
The command line parameters are passed to linux using the standard device tree mechanism (/chosen/bootargs).

Synoia, 2016-08-24:
It appears we need hot-plug of monitor. The mechanism of providing video parameters in the kernel command line does tends to force reboot on hot-plug of monitor. 

Changing framebuffer size with on-the-fly from x resulted blank screen.

lurch, 2016-08-25:
Perhaps hot-plugging the monitor causes some kind of power-glitch, which is enough to cause the Pi to reboot?

Synoia, 2016-08-25:
The problem is the Video config parameters are kernel parameters. If a different monitor is plugged in the parameters can be wrong. If the monitor is unplugged on boot the video parameter can be wrong.
Monitor video parameters need to be dynamic, and calculated when the monitor is detected. A static configuration such as config.txt forces a reboot.
 Duncan Hare

714 931 7952

lurch, 2016-08-26:
Ahh, I guess you mean attempting to hot-plug a monitor with a different resolution *requires* a reboot?

I dunno if it meets your needs, but there were some recent changes to the firmware to allow the hdmi resolution to be changed without requiring a reboot https://github.com/raspberrypi/firmware/issues/637
(but you probably need to close X first, then start it again after the resize)

Synoia, 2016-08-26:
(but you probably need to close X first, then start it again after the resize)
Oh, with the GUI login, you mean a reboot?
I suggest we look at PC's as the model to follow. They do not have this design. The OS can still be Debian.
Duncan Hare

714 931 7952

lurch, 2016-08-26:
> Oh, with the GUI login, you mean a reboot?

Nope, no reboot necessary e.g. https://www.google.co.uk/search?q=close+xserver

> I suggest we look at PC's as the model to follow. They do not have this design.

As you may have noticed, the Raspberry Pi is very different to a PC!!
AIUI (which may be wrong), the way Raspbian works currently is that the firmware sets up the framebuffer device, and then X simply uses whatever framebuffer it's given. I believe the open-source GPU driver work being done by Eric Anholt will allow a more traditional "X controls the display directly", which appears to be what you're asking for.

Synoia, 2016-08-26:
Yes! Correct, the user controls X which controls all else. The solution we want is user driven. We guarantee controlling the user is outside any scope of our work :-)

A combination of the command xrandr, or equivalent, a test pattern generator, combined with a cache of the current monitors settings seems  a requirement..

Change the monitor, purge the cache and go through a user driven simple process to make the system work as the user desires.

Thank you, I believe we agree. How complete is Eric's work? Duncan Hare

714 931 7952

lurch, 2016-08-30:
> How complete is Eric's work?

I've not been following it myself, but you can read more about it at http://anholt.livejournal.com/

@popcornmix I suggest this issue be closed, as it seems to have gone off-topic somewhat.

closed 2016-08-30 by popcornmix